olikraus/u8g2 #276: slow to draw on dogxl160 button problems

hunter1000, 2017-05-28:
Hi,

I got a problem:

I got a dogxl160 screen on a 3,3 volt zero board with 4 buttons and use #include "U8glib.h". But the buttons are realy slow to respond. I did some investigation and found out that de draw cycle in my sketch takes about 331 milliseconds to draw. The buttoncontrol takes 0 milliseconds to cycle.

Question: Am I doing something wrong or is this inherant of the U8glib.h library?

I initialize the display using:
U8GLIB_DOGXL160_2X_BW u8g(13, 11, 10, 9, 4);  // SPI Com: SCK = 13, MOSI = 11, CS = 10, A0 = 9, reset = 4

Buttons are controled by a copy the arduino debounce example.

Any help would be nice!

olikraus, 2017-05-28:
This is the issue tracker for U8g2, which is the successor of u8glib. So maybe you also want to test U8g2. Both libs support hardware SPI, however board support is limited for u8glib. At the moment your constructor activates software emulated SPI, which is indeed very slow. Better use HW SPI.

hunter1000, 2017-05-29:
Hi,

Thanks for the tip!!!

I got it working on a samd21 with fullframe HW SPI: Sparkfun ATSAMD21G R3 breakout on the native SPI ports.

I made my own zero breakout, but when I selected the zero in board manager, it didnt work. I think the native spi ports are not activated for a zero. So I had to use the SW SPI. that is so slow page, a page draw takes about 300ms.

With the Sparkfun ATSAMD21G R3 breakout on the native SPI ports it takes about 30 ms to draw a page.

Setup:
Install the samd21 boards (zero) in arduino.
Install the sparkfun boards in arduino (follow instructions: https://learn.sparkfun.com/tutorials/samd21-minidev-breakout-hookup-guide/setting-up-arduino)
Connect DOGXL160 via ICSP lagacy header.
Use constructor: U8G2_UC1610_EA_DOGXL160_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9,3);

Kind regards,

Albert-Jan Campagne

hunter1000, 2017-05-29:
I think you can use the sparkfun board in arduino for any zero breakout clone, and make it work.

olikraus, 2017-05-29:
Great to see that this works for you.

closed 2017-07-01 by olikraus