olikraus/u8g2 #277: SW I²C works but HW I²C doesn't on 0.91 OLED

miko007, 2017-05-28:
Hello,

i have the following diplay connected to a *Funduino UNO* board. If i use the constructor

```c++
U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
```
it is working fine, but very slow. So i opted to use the hardware I²C to "pump it up" by using:

```c++
U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);
```

The `SCL` and `SDA` pins of the display are connected to the corresponding pins on the Arduino board. I also tried to use the pins `A4` and `A5` and also using external resistors as pullup, but neither of it worked. *(power is completely supplied by the Arduino board, therefore common ground is present)*

Sadly, the display stays black. It seems that it also does not get found by the Arduino I²C scanner sketch. Has anyone some advise for sorting this out? 

[image]

olikraus, 2017-05-28:
The constructor call is wrong. The HW I2c constructor does not expect the clock and data pins. The i2c pins are hardwired.

olikraus, 2017-05-28:
I have updated the FAQ for this topic:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

miko007, 2017-05-28:
okay thanke @olikraus, but still no luck. so i have to assume a hardware failure on either the funduino board or the display *(i read smth about some displays not sending `ACK` correctly)*.

My wiring seems appropriate and i now took the constructor you mentioned, but the display stays dark. Also the fact that the I²C scanner doesn't find the displays address, sadly seems to confirm my theory :/

olikraus, 2017-05-28:
Well, in such a case you have to live with the SW I2C constructor.

closed 2017-07-01 by olikraus