olikraus/u8g2 #278: Add pointer to user data in u8x8_t

devsaurus, 2017-05-28:
There is currently no way to pass complex information to the callbacks for gpio&delay and byte oriented communication other than by global variables.
It would be beneficial to have a member like `void *user` as part of `u8x8_t` for that purpose.

olikraus, 2017-05-28:
I agree, but i never expected, that someone else wants to create more complex callback functions. Most people are quite happy with global data here. The question is, is it really required to add another pointer variable here. It would be useless and wasted space for 99.9% of the users.

devsaurus, 2017-05-28:
> The question is, is it really required to add another pointer variable here.

It would be helpful, there are ways to work around it. Consider the case of more than one display and the callbacks have to maintain driver info per display connection. I could use a global array for sure, but even for this approach I need to abuse e.g. `U8X8_PIN_D7` to store the array index.

> It would be useless and wasted space for 99.9% of the users.

It could be conditioned by a macro like `U8X8_USE_PINS` if you're concerned about the memory footprint.

olikraus, 2017-05-28:
ok... 

- [ ] Add user pointer
- [ ] conditional code, based on U8X8_WITH_USER_POINTER
- [ ] u8x8_GetUserPointer, u8x8_SetUserPointer
- [ ] Write documentation

devsaurus, 2017-05-29:
Thanks a lot for considering this topic, @olikraus!

olikraus, 2017-08-12:
u8g2:
      void *getUserPtr() { u8g2_GetUserPtr(&u8g2); }
      void *setUserPtr(void *p) { u8g2_SetUserPtr(&u8g2, p); }

closed 2017-08-12 by olikraus