olikraus/u8g2 #281: Offset artifacts w/SSD1306 128x64 4SPI display using NONAME, works with VCOMH0

DLehenbauer, 2017-05-29:
Using the [HiLetgo SSD1306 128x64](https://www.amazon.com/gp/product/B01N1LZQA8/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1) with an Arduino Uno/4W_HW_SPI, I found that on each call to sendBuffer(), a few scan lines were shifted +1px to the right.

- This repros with NONAME_F and NONAME_2, but not NONAME_1 or VCOMH0_*.
- Which scan lines are shifted appears to be random on each call to sendBuffer().
- The framebuffer artifacts are persistent until the next clear/draw.
- The repro frequency seems to increase with the number of pixels lit.

Below is a short repro program.  For me, the display looks ~like this:

```
## ## ## ## ## ## ## ##
## ## ## ## ## ## ## ##
 ## ## ## ## ## ## ## ##
## ## ## ## ## ## ## ##
## ## ## ## ## ## ## ##
## ## ## ## ## ## ## ##
 ## ## ## ## ## ## ## ##
## ## ## ## ## ## ## ##
```

```cpp
#include <U8g2lib.h>

// With NONAME, some scanlines are shifted +1px right on each sendBuffer().
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI u8g2(U8G2_R0, /* cs */ 10, /* dc */  8, /* reset */ 7);

// With VCOMH0, display output is correct
//U8G2_SSD1306_128X64_VCOMH0_F_4W_HW_SPI u8g2(U8G2_R0, /* cs */ 10, /* dc */  8, /* reset */ 7);

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();
  // Problem begins to repro at 7+ iterations.
  for (int i = 0; i < 8; i++) {
    u8g2.drawBox(i * 16, 0, 15, 64);
  }
  u8g2.sendBuffer();
}
```

olikraus, 2017-05-29:
Due to a user request I have increased HW SPI speed for the ssd1306 to it's maximum. May you need to shorten your wires.

DLehenbauer, 2017-05-29:
You were right, my problem was wiring related, but more embarrassing.  My +5v power rail was disconnected.  The SSD1306 was leeching power from the data lines, and working amazingly well, considering.  (Thanks for the hint to recheck my wiring.)

closed 2017-05-29 by DLehenbauer