olikraus/u8g2 #282: Half display not working with library

charlielito, 2017-05-31:
Hi!
Im trying your library with a SSD1306 128x64 oled I2C display with an arduino mini pro. I run the example "Hello World" (see code at the end of this issue), but the display looks as follow:

[image]

I tried your former library and got the same results. But I guarantee that it is not a display problem because I run and used the Adafruit library for the Oled and it worked well in the whole display.

Do you know whats happening?

```
#include <Wire.h> 
#include <U8g2lib.h>
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
void setup(void) {
  u8g2.begin();
}
void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

olikraus, 2017-05-31:
Will the u8x8 graphics test work?

charlielito, 2017-05-31:
The problem solved itself! No clue how but thanks for the prompt response anyway!

closed 2017-05-31 by charlielito

olikraus, 2017-05-31:
ok, great :-)