olikraus/u8g2 #283: u8g2_ExportBufferXBMP

imarinl, 2017-06-01:
I'd to share with you this function. It is so useful if you want to write documentation and you need take a screen shot.

This function exports the buffer to an XBMP file.

You are free to change this as you want:
```
/*
 * Export u8g2 Buffer to a XBMP file
 *
 * @param u8g2	a pointer to u8g2_t
 * @param name	Name to export, for example test then you'll get test.xbm
 *
 * return 0 if all was ok, otherwise an error
 */

int32_t u8g2_ExportBufferXBMP (u8g2_t *u8g2,const char *name)
{
	char buff [256];
	uint8_t *buff_to_export;
	int height_bytes=u8g2->pixel_buf_height/8;
	int width_bytes=u8g2->pixel_buf_width/8;
	FILE *file_to_export;
	int i,j,z;

	snprintf (buff,sizeof(buff),"%s%s",name,".xbm");
	file_to_export=fopen(buff,"w");

	if (!file_to_export)
	{
		return -EFAULT;
	}

	buff_to_export = malloc ((height_bytes*8)*width_bytes);
	if (!buff_to_export)
		return -ENOMEM;

	fprintf (file_to_export, "#define %s%s %d\n",name,"_width",width_bytes*8);

	fprintf (file_to_export, "#define %s%s %d\n",name,"_height",height_bytes*8);

	fprintf (file_to_export, "static unsigned char %s%s[] = {\n",name,"_bits");

	// Clear buffer
	for (i=0;i<height_bytes*8;i++)
		for (j=0;j<width_bytes;j++)
			*(buff_to_export+i*width_bytes+j)=0;

	// Fill buffer
	for (i=0;i<height_bytes;i++)
	{
		for (j=0;j<width_bytes*8;j++)
		{
			for (z=0;z<8;z++)
				{
				if ((u8g2->tile_buf_ptr[i*128+j])&(1<<z))
					{
					*(buff_to_export + (i*8+z)*width_bytes + (j/8))|=(1<<(j%8));
					}
			}
		}

	}

	// Write buffer
	j=0;
	for (i=0;i<height_bytes*8;i++)
	{
		for (z=0;z<width_bytes;z++)
		{
			fprintf (file_to_export,"0x%02x, ",*(buff_to_export+i*width_bytes+z));
			j++;
			if (j==12)
			{
				fprintf (file_to_export,"\n");
				j=0;
			}
		}
	}

	fseek(file_to_export,-2,SEEK_CUR);
	fprintf (file_to_export,"}");

	free (buff_to_export);
	fclose(file_to_export);

	return 0;
}
```

olikraus, 2017-06-01:
Looks good, however note that the content of "u8g2->tile_buf_ptr" depends on the target display.
Actually this could be tested by looking at "u8g2->ll_hvline" which is one of
    u8g2_ll_hvline_vertical_top_lsb
    u8g2_ll_hvline_horizontal_right_lsb
So altogether there are two basic types of memory mapping in the displays.
  u8g2_ll_hvline_vertical_top_lsb used by SSD13xx, UCxxxx, ST7565 displays
  u8g2_ll_hvline_horizontal_right_lsb used by more classic displays like KS0108 or ST7920

Which type of display did you use for testing?

imarinl, 2017-06-01:
SSSD1309

olikraus, 2017-06-01:
yes, then this code applies only for u8g2_ll_hvline_vertical_top_lsb

closed 2017-07-01 by olikraus