olikraus/u8g2 #284: add extended fonts, -m '32-701'

olikraus, 2017-06-01:
this will also include glyphs hex 170/171 (see #105)

olikraus, 2017-06-01:
update wiki for "e" range (32-701)

olikraus, 2017-06-01:
The fonts are included in the .1 prerelease:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

tuccerHU, 2017-06-02:
Can you give me a hint how can i get the font what i need??? I am not an expert in this..... :D

olikraus, 2017-06-02:
1. look out for the font name (ends in "te"). Fonts can be found 
here: https://github.com/olikraus/u8g2/wiki/fntgrpadobex11
and here: https://github.com/olikraus/u8g2/wiki/fntgrpx11

2. install the developer snapshot of u8g2: download 
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
then inside Arduino IDE, do add library zip file installation.

3. Have fun with the new fonts. Use u8g2.drawUTF8() !!!

closed 2017-07-01 by olikraus