olikraus/u8g2 #285: SSD1606 Deep Sleep Mode Control

njsharkracer, 2017-06-03:
I see that when your initilizing this display, you disable the Deep Sleep Mode as needed.  Is there any way to re-enable the Deep Sleep Mode after it finishes writing to the display?  I'm not sure if the deep sleep mode also disables the charge pump.  I am trying to get the smart prototyping veriant of this display down to almost 0 current once it's done refreshing so I can run it with a atmega328 on a coin cell for 6+ months.  As of right now, with the LED disbaled on the back of the display module, the unit idles at 0.8mA, so clearly something is still switched on.

olikraus, 2017-06-03:
It can be added. Do you know the command sequence for deep sleep mode?

njsharkracer, 2017-06-03:
[GDE021A1.pdf](https://github.com/olikraus/u8g2/files/1049859/GDE021A1.pdf)

After looking at your u8x8_d_ssd1606_172x72.c file, line 72 is your disable the deep sleep mode.  In the attached PDF, the control registers for the deep sleep control are found on page 20.  I am not good at decoding register entries but it does show in the PDF that there is an enable and disable option.  On page 35 of the PDF, they show a flow chart for typical startup and shutdown procedures.  It does not list putting the unit in deep sleep but rather to shut off the charge pumps with the included registers in the flow chart.  I'm not sure which option is better but I am willing to test this on the hardware side of things if you are able to make sense of the coding portion.

olikraus, 2017-06-03:
The shutdown sequence is already done, see line 169 in u8x8_d_ssd1606_172x72.c

We could add
  U8X8_CA(0x10, 0x01),	/* deep sleep mode */
after line 171. will this help?

njsharkracer, 2017-06-03:
I did not see line 169, that is good.  We can try adding your addition after line 171 and see if it helps.  The only problem is I just finished breaking my display completely (broken glass), so I have already ordered a replacement from Smart Prototypes and it should be in my hand in 2 weeks for testing.  Hopefully with that code it will reduce the idle current.

olikraus, 2017-06-03:
Regarding the current: Smart Prototypes might have their own charge pump on the breakout board. This is obviously not affected by these commands.

njsharkracer, 2017-06-03:
After consulting the module from Smart prototypes that I have and the reference circuit schematic located in the above PDF, they are following the reference circuit in regards to the external components required for the charge pump which is mosfet controlled from the screen IC itself.  the additional code may or may not help.  If you want we can close this issue for the time being and I will manually add the additional line of code for testing in two weeks when I get my replacement screen in.  Then i'll report my finding to you in this issue.

olikraus, 2017-06-03:
No problem, I leave this issue open for some time.

njsharkracer, 2017-06-21:
I got my new screen in.  Putting the new code after line 171 just gives me a black screen. If I put it after the end_transfer and end, it doesn't get the command correctly and has no effect over the original version.  Is there a way to put this line in the powersave mode part of the code?

olikraus, 2017-06-21:
After display refres, u8g2 always puts the display back to power sleep. line 171 should be the correct position.

njsharkracer, 2017-06-22:
Ok, i played with the code some more and reviewed the datasheet again.  The flow chart on page 35 of the datasheet shows it sending command 20H again after closing the charge pump.  Without that command, it must not have been actually shutting off the charge pump.  I copied line 163:

  U8X8_C(0x20),	/* execute sequence */

And put it after line 169:

 U8X8_CA(0x22, 0x03),	/* disable clock and charge pump */

And now the idle current dropped from 200uA to 24uA.  This should resolve the idle current.  It also has no issue with waking back up for each refresh in my sketch every minute and have not had any black screen issues.

olikraus, 2017-06-23:
Wow, nice finding. Can you attach the modified c code to this ticket here?
I will then update the repository.

njsharkracer, 2017-06-28:
`/*

  u8x8_d_ssd1606_172x72.c

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  
  
  SSD1606: 128x180x2 
  two-bit, four graylevels
  command 
    0x22: assign actions
    0x20: execute actions
  
  action for command 0x022 are (more or less guessed)
    bit 7:	Enable Clock
    bit 6:	Enable Charge Pump
    bit 5:	Load Temparture Value (???)
    bit 4:	Load LUT (???)
    bit 3:	Initial Display (???)
    bit 2:	Pattern Display --> Requires about 945ms with the LUT from below
    bit 1:	Disable Charge Pump
    bit 0:	Disable Clock
    
    Disable Charge Pump and Clock require about 267ms
    Enable Charge Pump and Clock require about 10ms

  Notes:
    - Introduced a refresh display message, which copies RAM to display
    - Charge pump and clock are only enabled for the transfer RAM to display
    - U8x8 will not really work because of the two buffers in the SSD1606, however U8g2 should be ok.

*/

#include "u8x8.h"

#define L(a,b,c,d) (((a)<<6)|((b)<<4)|((c)<<2)|(d))

/* GDE021A1, 2.1" EPD */
static const uint8_t u8x8_d_ssd1606_172x72_gde021a1_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_CA(0x10, 0x00),	/* Deep Sleep mode Control: Disable */
  U8X8_CA(0x11, 0x03),	/* Define data entry mode, x&y inc, x first */
[... 291 lines cut ...]
  /* sck_pulse_width_ns = */ 100,	/* SSD1606: 100ns */
  /* sck_clock_hz = */ 8000000UL,	/* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
  /* spi_mode = */ 0,		/* active high, rising edge */
  /* i2c_bus_clock_100kHz = */ 4,
  /* data_setup_time_ns = */ 40,
  /* write_pulse_width_ns = */ 150,	
  /* tile_width = */ 22,		/* 22*8 = 176 */
  /* tile_hight = */ 9,		/* 9*8 = 72 */
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
  /* pixel_width = */ 172,
  /* pixel_height = */ 72		
};

uint8_t u8x8_d_ssd1606_172x72(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    if ( msg == U8X8_MSG_DISPLAY_SETUP_MEMORY )
    {
      u8x8_d_helper_display_setup_memory(u8x8, &u8x8_ssd1606_172x72_display_info);
      return 1;
    }
    return u8x8_d_ssd1606_172x72_generic(u8x8, msg, arg_int, arg_ptr);
}

`

olikraus, 2017-06-28:
thanks, i will keep it open until i finde time for repo updated

njsharkracer, 2017-06-28:
ok

olikraus, 2017-08-12:
hmm.. this is not working for my display.

closed 2017-08-27 by olikraus