olikraus/u8g2 #286: reduce data size

schmurfy, 2017-06-03:
is there a way to reduce this library size by dropping fonts I don't use ?

olikraus, 2017-06-03:
This happens automatically for gcc with enable linker garbage collector (which is default for the Arduino IDE).

schmurfy, 2017-06-03:
I missed that but it means my problem might not be there, I am running my code on an atmega328 and when including this library my data usage (SRAM I think ?) jumps 1476 byes up which brings me above 100% and my program does not even start, any idea what might take that much ?

If you compile the GraphicTest example you already have 71% memory used :(

olikraus, 2017-06-03:
Suggestions differ depending on the location of the data (RAM or FLASH).

Flash: 
Use only a few fonts or better only one font. Try to use the "tr" fonts if possible. "tr" (transparent + reduced) will contain lesser glyphs, but will also occupy lesser flash.

RAM:
Use the _1_ page buffer display constructor. This will minimze the RAM consumption.

schmurfy, 2017-06-04:
Thanks a lot, much better !
My problem was with the RAM usage and using only 1 buffer does the trick, I now realize it was documented but this library does a lot of things and I did not know where to look, I really like it though :)

closed 2017-06-04 by schmurfy

olikraus, 2017-06-04:
:)

maopiccoto, 2018-03-25:
This tread helped me a lot, thanks!