olikraus/u8g2 #290: u8g2_font_unifont_t_extended missing in build

gogela, 2017-06-14:
Not sure if intentionaly or not, but even it is declared and documented the **u8g2_font_unifont_t_extended** is missing in the library and the font  build dir too. Any plans to add it there? It is quite essential for latin based languages with accented chars.
thx

olikraus, 2017-06-14:
Yes, this is related to issue #284 which is assigned to release 2.16. Which means, it is not inside the current official release 2.15.

Actually the problem is this: I can do a release planing on code level, however the documentation always reflects the latest (beta) release, because there is no release/rollout for the documentation.

More specific: the u8g2_font_unifont_t_extended font is documented (because it is implemented) but it not rolled out.

As a workaround you can download the latest developer snapshot:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Install this zip file manually via the "Add Library zip" menu option in the Arduino IDE. After this, the font should be included and u8g2 should be on version 2.16.1.

Side question: I am interessted what makers do with the extended font. Is it useful? You are planing to develop within the Arduino IDE? Any issues with UTF8 here? Are there any alternatives beside u8g2?
This feature is quite new and probably also unique...

I would also be interessted about your project it self...

olikraus, 2017-06-14:
Also note that I have added extended (_te) fonts also in some other font styles:
Here
https://github.com/olikraus/u8g2/wiki/fntgrpadobex11
and here:
https://github.com/olikraus/u8g2/wiki/fntgrpx11

gogela, 2017-06-14:
Great, I'll use the dev snapshot.
thanks

closed 2017-06-14 by gogela