olikraus/u8g2 #291: Running the library from other than the root ino file.

Tulsa2008, 2017-06-19:
I have a question for you.  I am using your library in a relatively large program.  As such, I have the program broken down into several .h/.cpp file pairs.  

I have noticed that your library doesn't want to run except from where ever it is started.   

Is this a requirement of the library?? If so no problem, I will deal with that.  If not, how do I set the library up such that I can call and use it from anywhere in the program??

I am not necessarily looking for a change in the library just some in site.

Thanks for a great library and excellent support.

olikraus, 2017-06-19:
Hm... i can not say much to this topic.
1. U8g2 follows the advices for Arduino Libraries 
2. U8g2 has (almost) no specific path information, except for some subdires, which must exist. 

If you try to do this from within Arduino IDE, i would say it is difficult, but not because of U8g2: It is more difficult because of the Arduino IDE itself.

Tulsa2008, 2017-06-19:
You might be correct.  I had to setup the SD chip/ board drivers to load from the .h/.cpp file pair where all of my SD related functions exist to get it to work.  That was not that of a big deal as I included all of the utilities I use when reading, writing or working with the SD board in the same  .h/.cpp pair.  I setup a function that runs the initialization routines and call it from the "Setup" loop during the booting process.  I could do the same with U8g2.  

I  have a about fifteen functions that make calls to U8g2.  I very likely will put most of them in the same .h/,cpp file pair and initialize U8g2 from the  U8g2 file pair.    I could setup a shell function to make the actual calls to the U8g2 library and call it from wherever I need to.  

I'll contact the Arduino group to see what their thoughts are on the subject.  I'll let you know what if anything they tell me..

closed 2017-07-01 by olikraus

JamesGKent, 2017-07-04:
for the record, there is an easy way to use this library from any file in your project.
so to do this I've made a pair for files for the screen declaration:
screen.h:
```
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

extern U8G2_KS0108_128X64_F *u8g2;

void init_screen();
```
screen.cpp:
```
#include "screen.h"

U8G2_KS0108_128X64_F *u8g2;

//U8G2_KS0108_128X64_F u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 18, /*dc=*/ 17, /*cs0=*/ 14, /*cs1=*/ 15, /*cs2=*/ U8X8_PIN_NONE, /* reset=*/  U8X8_PIN_NONE); 	// Set R/W to low!
void init_screen(){
	u8g2 = new U8G2_KS0108_128X64_F(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 18, /*dc=*/ 17, /*cs0=*/ 14, /*cs1=*/ 15, /*cs2=*/ U8X8_PIN_NONE, /* reset=*/  U8X8_PIN_NONE); 	// Set R/W to low!
	u8g2->begin();
}
```
then by including "screen.h" in any file will give you access to the screen functions. but note that you need to put the type in three places when you change the constructor, once in screen.h, and two in screen.cpp

olikraus, 2017-07-04:
Thanks for the information.

Tulsa2008, 2017-07-05:
Thanks a bunch for the help.  The problem of accessing the U8G2 from other than the "ino" file has become a major issue in my in my project.  The "ini" file was getting to the point where the only functions in it where  screen handling functions.

I am not familiar with the "new" function but I will look it up.  I am familiar with the "->" tool though.

Again thanks for the hint.

JamesGKent, 2017-07-06:
that's ok, hope it helps, I've had to use this arrangement in a C++ windows dll before.
all that happens is that by using a pointer in the header rather than the normal constructor we avoid defining it in the header, but we do declare its type. by declaring it as `extern` we let the compiler know that we will actually properly declare it somewhere else, this avoids getting a multiple definitions error from the compiler for every file it gets included into.
as you are aware `->` is used when accessing methods of an object when we're using a pointer to the object instead of `.` being used when you have a proper reference to the object.

as for the `new` operator, this lets you create the object and assign it to the pointer during program flow. but be aware that the way it's written if you call the `init_screen` function multiple times it will cause a memory leak as the memory will never be freed from the previous calls to `new`

Tulsa2008, 2017-07-06:
Again, thanks a bunch.  This will certainly help.  The program I am developing is running on an Arduino Mega2560.  The program basically runs a dash board in my Off-Road car.  I am using a Rugged Circuits 512K memory board to provide the needed RAM for the project.  That is where I am using the "->" tool.  I am loading map images into to the external RAM from a SD chip using the "->" tool to control where the processor places the map images.  I am currently running two (2) GLCD Transflective displays and may go to three(3).  

I did look up the "new" function and now understand how it works.  I also see that if you want to recover the RAM allocated by the use of "new" you need to use "delete[]".  In this case I don't think that will be a problem.  I will also try to use this with the SD library as well.  I have the same problem with it but I am not using it anywhere as often as I am using the U8G2 library.        

Again, thanks for the help