olikraus/u8g2 #292: Port U8GLIB_64128N to U8G2

njsharkracer, 2017-06-28:
Could you please port over the settings for U8GLIB_64128N from U8Glib to U8G2?  That constructor and settings works with digole's 128x64 display.

olikraus, 2017-06-28:
can be done, hope you can do some testing...

njsharkracer, 2017-06-28:
yes i have a rig set up ready for testing.

olikraus, 2017-06-29:
I have added the display. Beta release is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Use the "Add Zip File Library" feature of the Arduino IDE.

Can you test the U8x8 FlipMode example? Will it work correctly?
Maybe send a screeshot if something is wrong.

Thanks

njsharkracer, 2017-06-29:
I have tried any test sketch you have for the 64128n but it will not work.  I can load up the graphicstest from the old u8glib and it runs perfectly, after I adjust the setcontrast to 40, but once I upload the u8u8 flipmode example from u8g2 and use your new port of 64128n to u8g2, all it does is clear the screen and pin 13 blinks once a second or so. setcontrast does not do anything for the flipmode example, so I can not tell if your ported code does not work or i'm having an issue with setcontrast with u8g2.

olikraus, 2017-06-30:
Strange. Can you test other ST7565 devices? Will you see something?

njsharkracer, 2017-06-30:
Ok, I got a chance to play with your code a little bit today.  I copied everything over for the init sequence from u8g to u8g2, removed the delay calls, and updated the u8g to u8g2 calls for each command (yeah I can't program).  The display is now working correctly with the code below, but everything is shifted to the right and starting on the 5th pixel instead of the 1st pixel with the flipmode test. 
[image]

/*================================================*/
/* Displaytech 64128n */

static const uint8_t u8x8_d_st7565_64128n_init_seq[] = {
    
  U8X8_START_TRANSFER(),            /* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            	   /* soft reset */
  U8X8_C(0x0A2),   				   /* 0x0a2: LCD bias 1/9 (according to Displaytech 64128N datasheet) */
  U8X8_C(0x0A0),  				   /* Normal ADC Select (according to Displaytech 64128N datasheet) */
  U8X8_C(0x0c8),                   /* common output mode: set scan direction normal operation/SHL Select, 0x0c0 --> SHL = 0, normal, 0x0c8 --> SHL = 1 */
  U8X8_C(0x040),		           /* Display start line for Displaytech 64128N */
  U8X8_C(0x028 | 0x04),            /* power control: turn on voltage converter */
  U8X8_C(0x028 | 0x06),            /* power control: turn on voltage regulator */
  U8X8_C(0x028 | 0x07),            /* power control: turn on voltage follower */
  U8X8_C(0x010),                   /* Set V0 voltage resistor ratio. Setting for controlling brightness of Displaytech 64128N */
  U8X8_C(0x0a6),                   /* display normal, bit val 0: LCD pixel off. */
  U8X8_C(0x081),      	           /* set contrast */
  U8X8_C(0x01e),        	       /* Contrast value. Setting for controlling brightness of Displaytech 64128N */
  U8X8_C(0x0af),		           /* display on */
  U8X8_C(0x0a5),		           /* display all points, ST7565 */
  U8X8_C(0x0a4),		           /* normal display */
  U8X8_END_TRANSFER(),             /* disable chip */
  U8X8_END()             		   /* end of sequence */
};

olikraus, 2017-06-30:
good work, but i do not understand what you mentioned about flip mode. It works with flip mode or not?
At least the picture seems to be correct.

There are some issues with the your init sequence. One point is, that the sequence should turn off display at the end. with the 0xae command.

njsharkracer, 2017-06-30:
If you look at the picture again you'll see there are 4 blank rows next to the F in flipmode and when it flips the other direction, part of the F is off the screen as the whole screen is shifted by 4 pixels.

I'll let you fix the details with the coding, I suck at it.  I am lucky I mashed that together enough for it to work.  You are undoubtedly much better then I am.

njsharkracer, 2017-06-30:
I just tried the graphics test example, everything is aligned correctly in that, so it must just be the sketch for flipmode.  Ignore my first comment on the last post.

olikraus, 2017-07-01:
ok, thanks for the feedback. I will create another beta release soon. Hope you do another test on this.

olikraus, 2017-07-01:
I have added another beta release.Maybe you can give it a try.
I did 3 changes:
1. i tried to fix the 4 pixel issue
2. i modified the end of the init sequence so that the display stays off
3. i rotated the screen by 180 degree in the init sequence.

njsharkracer, 2017-07-01:
The 4 pixel issue with flipmode is fixed.  
The screen rotation worked correctly.

Thank you so much!  I would call this done.

olikraus, 2017-07-01:
wow, thanks a lot for the feedback and the testing.

closed 2017-07-01 by olikraus