olikraus/u8g2 #293: Generic 128*64 1.3inch OLED (I2C) issues

philpugh51, 2017-06-29:
(originally incorrectly posted in the 'old' library section)

I  have a problem with generic 1.3" 128X64 display.
As soon as it initialised you get the annoying 1 or 2 columns of random pixels on RHS of display

the graphics example works but is shifted one or two pixels to the left - leaving the random RHS pixels on display,
 
I have working code using 0.96" 128X36 display that will 'work' on this display (in FRAME mode and only in top half obviously) using the construct 
U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C u8g2(U8G2_R0, /* clock=/ 5, / data=/ 4, / reset=*/ U8X8_PIN_NONE);  This shows the same issues as the graphics example.

Using the following construct I get a full FRAME display working with the annoying artifacts already mentioned.
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=/ 5, / data=/ 4, / reset=*/ U8X8_PIN_NONE);

Is there a simple fix do you think?

Phil

Sorry this is on ESP8266-12E device

olikraus, 2017-06-29:
Did you try the SH1106 constructor?

philpugh51, 2017-06-30:
Didn't spot the i2c version until now. Will try it and report back later

philpugh51, 2017-06-30:
Ooopppps..

Suprising what happens when you use the correct library!!

Many thanks - all working as expected

olikraus, 2017-06-30:
:)

closed 2017-06-30 by olikraus