olikraus/u8g2 #294: Feature Request: GDEP015OC1

PeterKietzmann, 2017-07-04:
Are there any plans  to port the 1.54 inch E-paper display 200x200 GDEP015OC1?

olikraus, 2017-07-04:
no plan at the moment.
I did a quick search for this display. The controller seems to be a IL3829, which might be similar/compatible to the SSD1607. Did you try the SSD1607 constructor?

PeterKietzmann, 2017-07-05:
@olikraus thanks for your quick response. Until now I tried nothing and I don't know if I will find the time. Support would have been great because we integrate your wonderful lib in our IoT OS [RIOT](https://github.com/RIOT-OS/RIOT) and I just got some [aconno dev boards](http://aconno.de/acd52832/) which are equipped with an GDEP015OC1 dispaly. No need to keep this issue open from my side.

olikraus, 2017-07-05:
I am quite sure that the ssd1607 constructor will work.

olikraus, 2017-07-15:
ok, please reopen if required

closed 2017-07-15 by olikraus