olikraus/u8g2 #295: 96x16 OLED support

whatnick, 2017-07-04:
I have a 96x16 oled mounted on an Adafruit feather. Hello World runs with scaling issues, OLED is set as ssd128x32. Support for this appreciated.

olikraus, 2017-07-04:
Can you share the code, which generates the content on your OLED?
The Adafruit Feather OLED comes with a 128x32 OLED. Did you mount or own 96x16 OLED on it?

whatnick, 2017-07-04:
Yes I managed to burn the 128x32 with a freak accident, wrongly ordered a 96x16 and mounted that. It works fine. With some glitches. The output is produced by the u8g2 "hello_world" sample.

```
void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB14_tr);	// choose a suitable font
  u8g2.drawStr(0,20,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}
```

olikraus, 2017-07-04:
Understand. ok, then please locate u8x8_d_ssd1306_128x32.c on your drive (might be somewhere in Arduino/libraries/u8g2/...
for reference:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x32.c

Then goto line 50 and change
U8X8_CA(0x0d3, 0x000), /* display offset */
to
U8X8_CA(0x0d3, 0x010), /* display offset */

Will this work?

Oliver

whatnick, 2017-07-08:
Should I copy this and make a custom class with this offset hardcoded ?

olikraus, 2017-07-08:
I will create a new class, but before that i would like to know what init sequence is required. So if you could modify the 132x32 constructor from above, so that the hello world appears correctly, then i will start to implement this.

Do you have a datasheet for your 96x16 display module?

whatnick, 2017-07-08:
This is the module I have -
http://www.buydisplay.com/default/graphic-i2c-0-69-inch-96x16-oled-display-mp3-module-white-on-black
. Datasheet is on a link on the page.

whatnick, 2017-07-12:
> I have created a new release. Please test the u8x8 "hello world" and "flip
> mode" examples.
> will they work. The constructor is:
>
> U8X8_SSD1306_96X16_ER_HW_I2C
>
> You can download the beta release here:
>
> https://github.com/olikraus/U8g2_Arduino/archive/master.zip
>
> Please manually install this release via the "add zip library" menu of
> your Arduino IDE.

I have tested the new release with the OLED and ESP32. Works great. Thanks
a lot for supporting this one. Hope others find it useful as well.

olikraus, 2017-07-12:
great, thanks for the feedback

closed 2017-07-15 by olikraus