olikraus/u8g2 #296: Digispark ATTiny85 U8x8 compile error

simap, 2017-07-05:
First of all, thank you for this library!

It looks like there should be support for Digispark on the [boards list](https://github.com/olikraus/u8g2/wiki/boardlist) and also as one of the commented out constructors in the [examples](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8x8/HelloWorld/HelloWorld.ino#L55) but when I try this, I get a compile error. I've tried with Arduino 1.6.5 and 1.8.2. 
```
In file included from /var/folders/t2/9c5014nr8xl_kw008k8bwpm80000gp/T/arduino_modified_sketch_373643/HelloWorld.ino:39:0:
/Users/benh/Documents/Arduino/libraries/U8g2/src/U8x8lib.h:230:12: error: conflicting return type specified for 'virtual size_t U8X8::write(const uint8_t*, size_t)'
     size_t write(const uint8_t *buffer, size_t size) {
            ^
In file included from /Users/benh/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/Stream.h:24:0,
                 from /Users/benh/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/TinyDebugSerial.h:31,
                 from /Users/benh/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/WProgram.h:18,
                 from /Users/benh/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/Arduino.h:4,
                 from /var/folders/t2/9c5014nr8xl_kw008k8bwpm80000gp/T/arduino_modified_sketch_373643/HelloWorld.ino:38:
/Users/benh/Library/Arduino15/packages/digistump/hardware/avr/1.6.7/cores/tiny/Print.h:75:18: error:   overriding 'virtual void Print::write(const uint8_t*, size_t)'
     virtual void write(const uint8_t *buffer, size_t size);
                  ^
```
So there is disagreement on the return value of `write`. In normal Arduino cores it would return the number of bytes written, but I think the Digistump core cut this to save code space.

I poked around in the source codes for [Digistump's arduino core](https://github.com/digistump/DigistumpArduino/blob/master/digistump-avr/cores/tiny/Print.h#L75), and as far as I can tell the signature for this function hasn't changed and has always returned void. 

Is there a particular version that I'll need to support 8x8 on the Digispark?

olikraus, 2017-07-05:
I had tested U8g2 once with attiny85. But this is long time back. I don't know whether there is an old version of U8g2 which returns void, but maybe you can just change U8g2.

simap, 2017-07-05:
BTW it compiles just fine if I target Adafruit Trinket, which I think uses the normal Arduino core and has the same chip. I'll see if I can hack something into a working state.

Thanks!

olikraus, 2017-07-15:
seems to be solved

closed 2017-07-15 by olikraus

ivkusan, 2017-08-17:
It still doesn't work on Digispark ATTiny85

Arduino: 1.8.2 (Windows 7), Board: "Digispark (Default - 16.5mhz)"
In file included from C:\Users\ikusan\Desktop\sketch_jul23a\sketch_aug17a\sketch_aug17a.ino:3:0:
C:\Users\ikusan\Documents\Arduino\libraries\U8g2\src/U8x8lib.h:237:12: error: conflicting return type specified for 'virtual size_t U8X8::write(const uint8_t*, size_t)'
     size_t write(const uint8_t *buffer, size_t size) {
            ^

In file included from C:\Users\ikusan\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/Stream.h:24:0,
                 from C:\Users\ikusan\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/TinyDebugSerial.h:31,
                 from C:\Users\ikusan\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/WProgram.h:18,
                 from C:\Users\ikusan\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/Arduino.h:4,
                 from sketch\sketch_aug17a.ino.cpp:1:
C:\Users\ikusan\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\cores\tiny/Print.h:75:18: error:   overriding 'virtual void Print::write(const uint8_t*, size_t)'
     virtual void write(const uint8_t *buffer, size_t size);
                  ^

exit status 1
Error compiling for board Digispark (Default - 16.5mhz).

ivkusan, 2017-08-17:
Now I found Your [comment](https://forum.arduino.cc/index.php?topic=441215.0) and it seems legit.

pbird2018, 2018-01-21:
I give up to use U8x8 library in my Digispark Attiny85, but I could use DigisparkOLED example to enable 0.96 OLED.
Only mark one of  "// oled.bitmap(0, 0, 128, 8, digistumplogo);" line and it's //#include "digistump_128x64c1.h", then the program could be compiled with suitable memory size.   
See below example:
[code]
#include <DigisparkOLED.h>
#include <Wire.h>
// ============================================================================

#include "img0_128x64c1.h"
//#include "digistump_128x64c1.h"

void setup() {
  // put your setup code here, to run once:

  oled.begin();

}

void loop() {
  
  // put your main code here, to run repeatedly:
  oled.fill(0xAA); //fill screen with color
  delay(1000);
  oled.fill(0x55); //fill screen with color
  delay(1000);
  oled.clear(); //all black
  delay(1000);
  //usage: oled.setCursor(X IN PIXELS, Y IN ROWS OF 8 PIXELS STARTING WITH 0);
  oled.setCursor(0, 0); //top left
  oled.setFont(FONT8X16);
  oled.print(F("DIGISTUMP")); //wrap strings in F() to save RAM!
  oled.setFont(FONT6X8);
  oled.print(F(" OLED!"));
  oled.setCursor(0, 2); //two rows down because the 8x16 font takes two rows of 8
  oled.println(F("test")); //println will move the cursor 8 or 16 pixels down (based on the front) and back to X=0
  oled.print(F("test test test test test")); //lines auto wrap
  
  delay(3000);
  //usage oled.bitmap(START X IN PIXELS, START Y IN ROWS OF 8 PIXELS, END X IN PIXELS, END Y IN ROWS OF 8 PIXELS, IMAGE ARRAY);
  oled.bitmap(0, 0, 128, 8, img0_128x64c1);
  delay(3000);
 // oled.bitmap(0, 0, 128, 8, digistumplogo);
  delay(3000);
}
[/code]
As right now, I am looking for a good "printf" for this OLED module.