olikraus/u8g2 #297: Porting u8x8 (sw i2c)

Mekol, 2017-07-05:
Hey, i have an issue with porting.
I want to use u8x8 mode with SSD1306, i have written simple gpio_and_delay, but i have issue 
with u8x8_Setup.
Wiki says that this function have 4 args: 
`u8x8_Setup(u8x8_d_ssd1306_128x64_noname, u8x8_cad_ssd13xx_i2c, u8x8_byte_sw_i2c, uC specific)`
(uC specific is gpio_and_delay?)
But in u8x8_setup.c i see more arguments:
`void u8x8_Setup(u8x8_t *u8x8, u8x8_msg_cb display_cb, u8x8_msg_cb cad_cb, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)`

Can you explain me what the rest of args mean? This is what i wrote:
`u8x8_Setup(&scr,u8x8_cad_ssd13xx_i2c,u8x8_byte_sw_i2c,u8x8_gpio_and_delay_template);`
(with too few arguments error) 
Do i have to write something more than gpio/delay functions?

Also, i have second (maybe stupid) question:
I want to use simple swi2c and this library has "built in" this interface, all i have to do is set up gpio_and_delay (with pins) correctly?

olikraus, 2017-07-05:
Short answer via my mobile: 
Yes, last arg must be the gpio and delay proc
And yes, for SW I2C only the gpio and delay proc must be written.
Other functions for SW I2C are listed in the porting guide.

olikraus, 2017-07-05:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

olikraus, 2017-07-05:
My own code which implements SW I2C for u8x8 on an STM32L0:
https://github.com/olikraus/u8g2/tree/master/sys/arm/stm32l031x6/u8x8_test

Mekol, 2017-07-05:
thanks, solved :)

Mekol, 2017-07-06:
Now im having new problems.
1. My output is glitched (probably something wrong with gpio/delay)
2. It runs really slow (it takes about 7s from clicking reset to drawing "this")

Can you look at this?

Tested with 16mhz arduino Uno and atmega8 8mhz

[image]

my Main.cc:
```
#include "u8g2.h"
#include  "u8x8.h"
#include <util/delay.h>

// u8x8_gpio_and_delay.cc
uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
		void *arg_ptr);

u8x8_t scr;
int main() {
	u8x8_Setup(&scr, u8x8_d_ssd1306_128x64_noname, u8x8_cad_ssd13xx_i2c,
			u8x8_byte_sw_i2c, u8x8_gpio_and_delay_template);

	u8x8_InitDisplay(&scr);
	u8x8_SetPowerSave(&scr, 0);
	u8x8_ClearDisplay(&scr);
while (1) {
u8x8_DrawString(&scr, 0, 0, "abcd");
}
	return 0;
}

```
my gpio/delay:
```
#include "u8g2.h"
#include "u8x8.h"
#include <avr/io.h>

#include <util/delay.h>

// PC5 - I2C CLOCK PIN
// PC4 - I2C DATA PIN

uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
		void *arg_ptr) {
	switch (msg) {
	case U8X8_MSG_GPIO_AND_DELAY_INIT: // called once during init phase of u8g2/u8x8
		//DDRC &= ~(_BV(PC5)) & ~(_BV(PC4)); //inputs
		//PORTC |= _BV(PC5) | _BV(PC4);
//do i need this^?
		break;
	case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli seconds
		_delay_us(arg_int*1000UL);
		break;
	case U8X8_MSG_DELAY_I2C:// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
		_delay_us(arg_int <= 2 ? 5 : 1);
		break;			// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
	case U8X8_MSG_GPIO_I2C_CLOCK:
		if (arg_int == 0) {			// arg_int=0: Output low at I2C clock pin
			DDRC |= _BV(PC5);
			PORTC &= ~(_BV(PC5));
		} else {	// arg_int=1: Input dir with pullup high for I2C clock pin
			DDRC &= ~(_BV(PC5));
			PORTC |= _BV(PC5);
		}
		break;
	case U8X8_MSG_GPIO_I2C_DATA:
		if (arg_int == 0) {			// arg_int=0: Output low at I2C data pin
			DDRC |= _BV(PC4);
			PORTC &= ~(_BV(PC4));
		} else {	// arg_int=1: Input dir with pullup high for I2C data pin
			DDRC &= ~(_BV(PC4));
			PORTC |= _BV(PC4);
		}
		break;
	default:
		u8x8_SetGPIOResult(u8x8, 1);			// default return value
		break;
	}
	return 1;
}

```

olikraus, 2017-07-06:
Did you assign a proper font?

Mekol, 2017-07-06:
ok, font works, but very slow, like letter by letter

olikraus, 2017-07-06:
I think you can remove the DELAY_I2C on any atmega architecture.

Mekol, 2017-07-06:
ok works, can you explain me where i need to use delay_i2c? and why?

closed 2017-07-06 by Mekol

olikraus, 2017-07-06:
in general this delay is needed, but atmega architecture is sooo slow, that the delay is already sufficient without any explicit delay. 1 us are 16 clocks for a 16 MHz ATmega. These 16 clocks are already eaten up by the function call itself, so you need this delay only for faster uC architectures like Arduino Due or other Cortex M? with 48 MHz or more.