olikraus/u8g2 #299: how to create arduino library from a fork?

zerog2k, 2017-07-06:
I see there is some tool `tools/release/arduino/create_release.sh` however this looks like part of some build pipeline which expects some other stuff to exist.

I also looked at the wiki, but no helpful information.

https://github.com/olikraus/u8g2/wiki/internal#release

Is there a special way to build a test version of the arduino lib so I can test prior to sending in PR to this upstream repo?

zerog2k, 2017-07-06:
```
jens@spire:~/build/nrf/rcxr01/u8g2/tools/release/arduino$ ./create_release.sh 
== copy files to work dir ==
cp: cannot create regular file './../../../../U8g2_Arduino/src/.': No such file or directory
cp: target './../../../../U8g2_Arduino/src/.' is not a directory
sed: can't read ./../../../../U8g2_Arduino/src/U8g2lib.h: No such file or directory
sed: can't read ./../../../../U8g2_Arduino/src/U8x8lib.h: No such file or directory
cp: target './../../../../U8g2_Arduino/src/clib/.' is not a directory
cp: target './../../../../U8g2_Arduino/src/clib/.' is not a directory
rm: cannot remove './../../../../U8g2_Arduino/src/clib/u8x8_d_stdio.c': No such file or directory
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/ScrollingText’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/ContrastTest’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/DirectAccess’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/A2Printer’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/ExtUTF8’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/FPS’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/HelloWorld’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/PrintHelloWorld’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/PrintUTF8’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/U8g2Logo’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/FlipMode’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/SelectionList’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/GraphicsTest’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/Clock’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/XORTest’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/XBM’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/page_buffer/PowerSaveTest’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/games/LittleRookChess’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/full_buffer/HelloWorld’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/full_buffer/U8g2Logo’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/full_buffer/PrintUTF8’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/full_buffer/SelectionList’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/full_buffer/GraphicsTest’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/full_buffer/FPS’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/u8x8/HelloWorld’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/u8x8/GraphicsTest’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/u8x8/FlipMode’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/u8x8/MessageBox’: File exists
mkdir: cannot create directory ‘../../../../U8g2_Arduino/examples/u8x8/ArduboyTest’: File exists
cp: cannot create regular file './../../../../U8g2_Arduino/extras/.': No such file or directory
~/build/nrf/rcxr01/u8g2/tools/release/arduino ~/build/nrf/rcxr01/u8g2/tools/release/arduino
== adding files to repo ==
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
[... cut ...]
/home/jens/build
rm: cannot remove 'u8g2_arduino_.zip': No such file or directory

zip error: Nothing to do! (u8g2_arduino_.zip)
cp: cannot stat 'u8g2_arduino_.zip': No such file or directory
unzip:  cannot find or open u8g2_arduino_.zip, u8g2_arduino_.zip.zip or u8g2_arduino_.zip.ZIP.
~/build/nrf/rcxr01/u8g2/tools/release/arduino
now create a release in gitub for U8glib_Arduino, tag/name =
no prefix required, release name can be empty
jens@spire:~/build/nrf/rcxr01/u8g2/tools/release/arduino$ 
```

zerog2k, 2017-07-06:
ok, I think it involves checking out the target repo along side this code repo, which I have forked and now seems to build, but still trying to figure out how to setup a special test tag of this:
https://github.com/olikraus/U8g2_Arduino

olikraus, 2017-07-06:
"create_release.sh" copies the required files from the master repo U8g2 to the Arduino repo. The Arduino repo has to be in a special format so that it is accepted by the Arduino Library Manager.

closed 2017-07-07 by zerog2k