olikraus/u8g2 #302: Increase speed for some SSD13xx controller

olikraus, 2017-07-06:
The code looks like this:
```
    case U8X8_MSG_DISPLAY_DRAW_TILE:
      ...
      do
      {
	c = ((u8x8_tile_t *)arg_ptr)->cnt;
	ptr = ((u8x8_tile_t *)arg_ptr)->tile_ptr;

	do
	{
	  u8x8_cad_SendCmd(u8x8, 0x015 );	/* set column address */
	  u8x8_cad_SendArg(u8x8, x );	/* start */
	  u8x8_cad_SendArg(u8x8, x+1 );	/* end */

	  u8x8_cad_SendCmd(u8x8, 0x075 );	/* set row address */
	  u8x8_cad_SendArg(u8x8, y);
	  u8x8_cad_SendArg(u8x8, y+7);
	  u8x8_cad_SendCmd(u8x8, 0x05c );	/* write to ram */
	  u8x8_cad_SendData(u8x8, 32, u8x8_ssd1322_8to32(u8x8, ptr));
	  ptr += 8;
	  x += 2;
	  c--;
	} while( c > 0 );
	arg_int--;
      } while( arg_int > 0 );
```

It might be optimized to:
```
    case U8X8_MSG_DISPLAY_DRAW_TILE:
      ...
      u8x8_cad_SendCmd(u8x8, 0x075 );	/* set row address */      
      u8x8_cad_SendArg(u8x8, y);
      u8x8_cad_SendArg(u8x8, y+7);
      do
      {
	c = ((u8x8_tile_t *)arg_ptr)->cnt;
	ptr = ((u8x8_tile_t *)arg_ptr)->tile_ptr;

	do
	{
	  u8x8_cad_SendCmd(u8x8, 0x015 );	/* set column address */
	  u8x8_cad_SendArg(u8x8, x );	/* start */
	  u8x8_cad_SendArg(u8x8, x+1 );	/* end */

	  u8x8_cad_SendCmd(u8x8, 0x05c );	/* write to ram */
	  u8x8_cad_SendData(u8x8, 32, u8x8_ssd1322_8to32(u8x8, ptr));
	  ptr += 8;
	  x += 2;
	  c--;
	} while( c > 0 );
	arg_int--;
      } while( arg_int > 0 );
```

check:
 * [ ] SSD1322
 * [ ] SSD1325
 * [ ] SSD1327
 * [ ] SSD1329

Rappalot, 2017-07-06:
I am doing  this on a ssd1322 256x64 display, and it works for me. About 25% speed increase, when the full screen buffer is transfered.

olikraus, 2017-07-06:
ok, cool. Thanks for reporting. I probably need to do some further testing and check whether all parts of u8g2/u8x8 still work.

olikraus, 2017-07-06:
FPS results:
```
  U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI		Uno	8MHz 16 Bit	Clip=9.0 Box=10.7  @=2.0 Pix=3.0		issue 302 before optimization
  U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI		Uno	8MHz 16 Bit	Clip=10.9 Box=13.5  @=2.0 Pix=3.2	issue 302 after optimization
```

olikraus, 2017-07-06:
SSD1322/25/27 modified and tested. 
This can be closed.

closed 2017-07-06 by olikraus