olikraus/u8g2 #304: Problem: Using Arduino DUE and SSD1306 with HW I2c not working

TKaluza, 2017-07-11:
Hi there! 
Wonderfull library, I love it and thank you for making it!

I have used my Arduino MEGA with your library and everything worked fine. 
Then switched for more computing power to the Arduino Due and I do net get the display working. 

The Arduino DUE ist connected with the Programming Port. Using Pins 20 (SDA) and 21 (SCL) as the I2C connection ports. Those are connected to the display with a level shifter which has at LV 3.3V and at HV 5.0V. The Display itself is connected with 5V from the 5V port. 

After every upload of the sketch I hit the reset "Button". Cables not broken. Display works on MEGA.

With this setup and using the [I2C-Scanner](https://playground.arduino.cc/Main/I2cScanner) I get:
```
Scanning...
I2C device found at address 0x3C  !
done
```

So I think the setup is valid? (It got power and the DUE seems to find it - nothing else is connected)

As a constructor I am using the same as with the MEGA: 
`U8G2_SSD1306_128X64_NONAME_F_2ND_HW_I2C mydisplay(U8G2_R0,/* reset=*/ U8X8_PIN_NONE); ` 

The Display stays black. I also tried to power it from different power sources all 5V or connecting the I2C pins directly. 

I saw, that you are using the DUE yourself, that's why I am asking here. 
If there's anything unclear -- I'm happy to provide further information. 
Thank you very much!

olikraus, 2017-07-11:
Pins 20 (SDA) and 21 (SCL) are the first I2C interface, istn't it?
Maybe try
`U8G2_SSD1306_128X64_NONAME_F_HW_I2C mydisplay(U8G2_R0,/* reset=*/ U8X8_PIN_NONE);
`

Just to ensure, that the display it self works, you can also try the more slower SW I2C constructor:
`U8G2_SSD1306_128X64_NONAME_F_SW_I2C mydisplay(U8G2_R0, 21, 20,/* reset=*/ U8X8_PIN_NONE);
`

TKaluza, 2017-07-11:
Thanks for the fast replay. 

I'm sorry I meant the first I2C. 

I also tried your **SW** you've suggested, but It does not work. 

What I'm curious about is, if I have fried the Pins, it shouldn't be able to detect a I2C device, should it?

It is my frist time with the DUE, so using the Programming Code for powering the System is okay, or do I need to power it differently? 
I also tried the sparkfun library and example for the SSD1306 but it also doesn't work. So I suppose your code is probably not the problem but something with my DUE. 
If you have any thought about that, I would be delighted to here them :). 

Thanks

olikraus, 2017-07-11:
Strange, yes if the scanner finds the OLED, then the interface should work. Did you try using the OLED directly, without levelshifter?  Also use 3.3 power when trying this

TKaluza, 2017-07-11:
Now come a  few pictures - sorry. 
[image]
**Pic 1**: 
Working with connection to _MEGA_

[image]
**Pic 2**: 
Just better overview of the cables

[image]
**Pic 3**: 
Now connceted to the _DUE_ trying both `HW` or `SW`

[image]
**Pic 4**: 
Checking level shifter - seems to output something. 

[image]
**Pic 5**:
Checking now on the _DUE_  with 3.3V and direct connection. The SDA and SCL are showing some voltage. 

[image]
**Pic 6**: 
Connecting the VCC just to 3.3V but the SCL and SDA to the _MEGA_. It works.

[image]
**Pic 7**: 
And the I2C Scanner running on the _MEGA_ finds it. 

[image]
**Pic 8**: 
Setting up the _DUE_ with the I2C Scanner

[image]
**Pic 9**: 
It can also find the display. 

---

Do you have any idea or thought what I'm doing wrong? 
If not, absolutly no problem. I don't have a clue. Trying some other code tomorrow. But I already testet the adafruit lib with the example. I'll also trying other display-modul tomorrow. 

Thank you very much.

olikraus, 2017-07-12:
So you tried all three constructors with Due (1st HW, 2nd HW and SW I2C). Each of them you tried with a proper I2C level shifter, correct?
With each kind of wiring, the OLED is detected by the I2C scanner, but none of the OLED libs works.

I once had a similar case, where an OLED works with one board, but did not work with another (altough I do not remember, whether Due was involved here).

Maybe you should try to use an external power supplier, but all in all, I do not know what else could be the reason.

closed 2017-08-12 by olikraus

YerBabiE, 2018-08-15:
I've just encountered the same issue, I was using one brand of display fine with a wemos D1 mini using the 5V rail etc. then I switched to this new screen, and it will not light up it says www.diymore.cc on the back. If I scan for 12c interfaces it finds it, but it just will not work when given commands. I have tried on a few screens and few D1's all they are do the same thing... very frustrating. Did you end up getting a solution to this?

olikraus, 2018-08-15:
No :(

YerBabiE, 2018-08-16:
hmm, I think it's just certain screens that have an issue. I'm now going to be a little more picky when ordering....