olikraus/u8g2 #305: Problem using u8g2.userInterfaceInputValue

jgrossen, 2017-07-14:
Thanks very much for this library....

However, I can not get the u8g2.userInterfaceInputValue to work correctly.  I have copied the "Select Voltage" example in your documentation.  The display is correct and the Right/Left buttons scrolls thru the numeric values correctly.  But when the Select button is used the function value is 1 as expected but the index number returned as the "value" parameter is always zero.  Also when the function is called a second time the number in the scroll list has been incremented by 3 or 4.

Here is my code:
 #include <Arduino.h>
#include <U8g2lib.h>
#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

int retVal;
int inputVal;
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);

void setup() {
  Serial.begin(9600);
  Serial.println("Test Input Value");
  u8g2.begin(8, 9, 10, U8X8_PIN_NONE, U8X8_PIN_NONE, U8X8_PIN_NONE);
  u8g2.setFont(u8g2_font_6x12_tr);
}

void loop() {
  retVal = u8g2.userInterfaceInputValue("Select Voltage", "DAC= ", inputVal, 0, 5, 1, " V");
  Serial.println(retVal);
  if (retVal == 1){
    Serial.print("Input value: ");
    Serial.println(inputVal);
  }
  delay(4000);
}

Can you help?

Also I get a compiler error if I write the function as:
int reVal = u8g2.userInterfaceInputValue("Select Voltage", "DAC= ", inputVal, 0, 5, 1, " V");

However this works for the SelectionList function:
int selIndex = u8g2.userInterfaceSelectionList("Command", 2, "Reset\nSelect Servo");

olikraus, 2017-07-14:
You have to provide a pointer to the variable:
retVal = u8g2.userInterfaceInputValue("Select Voltage", "DAC= ", &inputVal, 0, 5, 1, " V");

jgrossen, 2017-07-14:
Thanks for the help.  I am very much a C novice and pointers are new to me....

closed 2017-07-14 by jgrossen

olikraus, 2017-07-14:
no problem