olikraus/u8g2 #308: haxrcorp-4089-cyrillic-altgr-regular.font

olikraus, 2017-07-15:
@yksht I continue with the font topic here

pulled out font topic from #266 

> Hi. I converted cyrillic font "HaxrCorp 4089 Cyrillic AltGr Regular" to work with u8g2. I used it in my home project but didn't found any info about its license. I would be glad to share converted font with comunity. Could I commit it? And are there problems with license?

> I got this font from here: https://fonts2u.com/haxrcorp-4089-cyrillic-altgr-regular.font
Here are links to "fontshop.com" but all links are broken ((
Designer's nickname is sahwar. Also didn't found any info about designer.

> bdfconv -f 1 -m '32-255,1025,1040-1103,1105' -n haxrcorp_4089_cyr_altgr_reg -o haxrcorp_4089_cyr_altgr_reg.c haxrcorp_4089_cyr_altgr_reg.bdf

olikraus, 2017-08-12:
Original source is here:
https://fontstruct.com/fontstructions/show/192981/haxrcorp_4089

License:
https://creativecommons.org/licenses/by-sa/3.0/

olikraus, 2017-08-12:
done

https://github.com/olikraus/u8g2/wiki/fntgrpfontstruct#haxrcorp4089

closed 2017-08-12 by olikraus

yksht, 2017-08-14:
oh!! you did a great job!! thanks a lot!

olikraus, 2017-08-14:
:-)