olikraus/u8g2 #309: ST7565 invalid init command

nickvanderstel, 2017-07-18:
ST7565 init sequences are sending command 0x0f8 in an attempt to set the voltage booster ratio. The ST7565 booster ratio is set by external capacitors - there is no command to configure the booster other than turning it on/off. Command 0x0f8 translates to a 'test' command that you are not supposed to use. See the [datasheet](http://www.newhavendisplay.com/app_notes/ST7565.pdf) pages 40 & 41. The ST7567 has an internal booster circuit and command 0x0f8 is used to configure it.

olikraus, 2017-07-19:
This command is documented on page 48 of the controller datasheet

http://www.newhavendisplay.com/app_notes/ST7565R.pdf

It is also suggested in several display module datasheets (EA, NHD). It should be there.
Examples:
http://www.newhavendisplay.com/specs/NHD-C12864LZ-FSW-FBW-3V3.pdf
http://www.lcd-module.com/eng/pdf/grafik/dogm128e.pdf

olikraus, 2017-07-19:
I have updated the cmd comparison for this controller. 

https://github.com/olikraus/u8g2/blob/master/doc/controller_cmds.txt

nickvanderstel, 2017-07-19:
There appears to be at least three variants of the ST7565: ST7565, ST7565**P**, and ST7565**R**. The ST7565**R** is the only one with the boost command. And it looks like different types of the NHD C12864 boards have different controllers. The C12864LZ that you linked has the ST7565R. This one has the ST7656P:

http://www.newhavendisplay.com/specs/NHD-C12864A1Z-FSW-FBW-HTT.pdf

And, looking at the [product page](http://www.newhavendisplay.com/lcd-chiponglass-128-x-64-c-3_114.html), the current displays do have either the ST7565P or the ST7565R.

Should the st7565 code be split into st7565r and st7565p?

Link to the P variant for reference:

http://www.newhavendisplay.com/app_notes/ST7565P.pdf

olikraus, 2017-07-19:
> The ST7565R is the only one with the boost command.

Maybe I am wrong, but the booster command 0xf8 is mentioned in both datasheets, R and P.

> Should the st7565 code be split into st7565r and st7565p?
I always create the drivers for a specific display, so i think it is not required to split this (only exception had been the SSD1306_noname driver).

nickvanderstel, 2017-07-20:
Ok... yeah the boost command is there in the ST7565P datasheet, I just didn't look carefully enough. The datasheets make it look like 0x0f8 would be considered a test command, which is what caused my confusion:

[image]

What initially threw me off is the datasheet for the display I am using:
http://www.newhavendisplay.com/specs/NHD-C12864A1Z-FSW-FBW-HTT.pdf

It uses the ST7565P but the datasheet links to the ST7565 controller, which does not have the boost command:
http://www.newhavendisplay.com/app_notes/ST7565.pdf

Sorry about all this! I'll go ahead and close.

closed 2017-07-20 by nickvanderstel

olikraus, 2017-07-20:
Thanks for the interessting discussen. I have learned some new things about the ST7565.