olikraus/u8g2 #313: ubuntu installation

naomi87, 2017-07-20:
`test.ino:6:18: fatal error: u8x8.h: File o directory non esistente
compilation terminated.`

I cant figure out how to install this librarie on ubuntu, looks like the IDE cant recognize the subfolder structure. I'm struggling to find an answer in the forums.
thanks

olikraus, 2017-07-20:
It nicely works for my own ubuntu.
Did you follow this procedure?
https://github.com/olikraus/u8g2/wiki/u8g2install

naomi87, 2017-07-24:
I dont have the manage libraries option in my Arduino IDE on lubuntu :(
I put the extracted zip file in the libraries folder... I can see the librarie in the menu and I can open the example files.

olikraus, 2017-07-24:
Well, then, the question is, how can i support here? 
I have tested the two options in the official Arduino IDE (library manager and add library zip file menu). Both options work without problems. BTW manual extraction works, 
My suggestion is to use the official 1.8.x release or Arduino IDE together with the library manager.

JamesGKent, 2017-07-25:
just a guess, but i seem to remember you saying something about this repo being for the development, and you have another arduino repo that puts the files into the correct folder structure to be recognized by the IDE, is it possible that the wrong repo/download is being used here? is it worth trying this [zip](https://github.com/olikraus/U8g2_Arduino/archive/2.15.2.zip) which is release 2.15.2 (latest) from your arduino repo, and extracting the contents into the arduino libraries folder?

olikraus, 2017-07-25:
Possible... the correct download location is mentioned here:
https://github.com/olikraus/U8g2_Arduino
The lastest version of the zip is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

closed 2017-08-12 by olikraus