olikraus/u8g2 #315: setup, loop vs. main and while(1)

Fox82, 2017-07-23:
Dear u8g2 supporters,

Why is this code for me working,

`
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

/* Constructor */
//U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0);

//U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0);
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);

/* u8g2.begin() is required and will sent the setup/init sequence to the display */
void setup(void) {
	u8g2.begin();
}

/* draw something on the display with the `firstPage()`/`nextPage()` loop*/
void loop(void) {
		u8g2.firstPage();
	do {
		u8g2.setFont(u8g2_font_ncenB14_tr);
		u8g2.drawStr(0,20,"Hello World!");
	} while ( u8g2.nextPage() );
	delay(1000);
}

`

and this one is not working! Display is blank. This is a big issue for me because with the switch from u8g to u8g2lib Display is not woring any more for my project.

`
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

/* Constructor */
U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);

int main (void) {
	u8g2.begin();

while(1){
		u8g2.firstPage();
	do {
		u8g2.setFont(u8g2_font_ncenB14_tr);
		u8g2.drawStr(0,20,"Hello World1");
	} while ( u8g2.nextPage() );
	delay(1000);	
}
}
`

olikraus, 2017-07-23:
Hmmm.. without loop it will not work, but this is an Arduino point, not  a U8g2 issue.

Fox82, 2017-07-23:
I replaced the structure to setup and loop. now it works. But just strange, because u8g was woring with that and made no problems.

olikraus, 2017-07-23:
setup/loop are mandatory for any Arduino program. Maybe it had been just luck, that u8glib did work without setup/loop.

Fox82, 2017-07-23:
Something is wrong. I cannot place any display request on an other place than in the loop.

For example this leads also to an frozen OLED.

`ISR(INT1_vect) { 

	u8g2.firstPage();
	do {
		u8g2.setFont(u8g2_font_ncenB14_tr);
		u8g2.drawStr(0,20,"INT1 on");
	} while ( u8g2.nextPage() );

} // ISR INT1 END`

olikraus, 2017-07-23:
Display output takes some time. If isr is faster than the output to the display, then your program will not work. Again this is not an U8g2 problem. Maybe you should better ask in Arduino for such questions.

closed 2017-07-28 by Fox82

huseyinbb, 2022-03-13:
I want to ask a question about display fonts light. I want to use the U8g2 library in my Arduino flashing text. Adafruit has a backlight code but I couldn't find a code like a backlight for the U8g2 library. Can you help me?

olikraus, 2022-03-13:
If the backlight is controllable, then you can use normal Arduino commands.

huseyinbb, 2022-03-13:
We want use flashing text in the display with using U8g2 but we havent do
this yet.
Do you have any code like "backlight adafruit " this ?

13 Mar 2022 Paz 22:27 tarihinde olikraus ***@***.***> şunu
yazdı:

huseyinbb, 2022-03-13:
We want use flashing text in the display with using U8g2 but we havent do
this yet.
Do you have any code or function  like "backlight adafruit " this ?

olikraus, 2022-03-13:
You need to provide more information. I have no idea what "backlight adafruit" is. Please remember: This is u8g2 library. It has nothing todo with Adafruit.

olikraus, 2022-03-14:
I don't see a video