olikraus/u8g2 #318: WaveShare 2.9 inch e-paper support

bogus105, 2017-07-28:
E-paper displays are getting more and more popular for battery powered applications. WaveShare e=paper displays are nice and almost cheap. It would be fantastic to add WaveShare e-paper displays support in your excellent u8g2 library. 

Please add 2.9inch 296x128 pixels module to your list.
http://www.waveshare.com/w/upload/e/e6/2.9inch_e-Paper_Datasheet.pdf

Thank you!

olikraus, 2017-07-28:
Can you takeover testing?

bogus105, 2017-07-28:
Oliver,
can you explain me what is LUT? i've seen it before when looking at the waveshare code.
I found this:

`//Write LUT register  
#ifdef EPD02X13
  static unsigned char LUTDefault_full[] = {0x32,0x22,0x55,0xAA,0x55,0xAA,0x55,0xAA,0x11,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x1E,0x1E,0x1E,0x1E,0x1E,0x1E,0x1E,0x1E,0x01,0x00,0x00,0x00,0x00};
  static unsigned char LUTDefault_part[] = {0x32,0x18,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x0F,0x01,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};
#else
  static unsigned char LUTDefault_full[] = {0x32,0x02,0x02,0x01,0x11,0x12,0x12,0x22,0x22,0x66,0x69,0x69,0x59,0x58,0x99,0x99,0x88,0x00,0x00,0x00,0x00,0xF8,0xB4,0x13,0x51,0x35,0x51,0x51,0x19,0x01,0x00};
  static unsigned char LUTDefault_part[] = {0x32,0x10,0x18,0x18,0x08,0x18,0x18,0x08,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x13,0x14,0x44,0x12,0x00,0x00,0x00,0x00,0x00,0x00};
#endif
`

so it seems 2.13inch display has different LUT table and all other displays (1,54inch and 2,9inch) share the same LUT values:
`//Write LUT register  
  static unsigned char LUTDefault_full[] = {0x32,0x02,0x02,0x01,0x11,0x12,0x12,0x22,0x22,0x66,0x69,0x69,0x59,0x58,0x99,0x99,0x88,0x00,0x00,0x00,0x00,0xF8,0xB4,0x13,0x51,0x35,0x51,0x51,0x19,0x01,0x00};
  static unsigned char LUTDefault_part[] = {0x32,0x10,0x18,0x18,0x08,0x18,0x18,0x08,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x13,0x14,0x44,0x12,0x00,0x00,0x00,0x00,0x00,0x00};
#endif
`

Yes i can test is. Be aware the hardware i have on hand now: WaveShare 2.9inch 296x122 module and Arduino Nano converted to 3.3V, Arduino Uno (but 5V). I ordered ESP8266 in China and waiting for it (it may take some time).

olikraus, 2017-07-28:
The LUT contains the display specific waveform for the pixel programming of the display.
I also found this in the code.

```  
LUT for the 296x128 device:
static unsigned char LUTDefault_full[] = {0x32,0x02,0x02,0x01,0x11,0x12,0x12,0x22,0x22,0x66,0x69,0x69,0x59,0x58,0x99,0x99,0x88,0x00,0x00,0x00
,0x00,0xF8,0xB4,0x13,0x51,0x35,0x51,0x51,0x19,0x01,0x00};
  static unsigned char LUTDefault_part[] = {0x32,0x10,0x18,0x18,0x08,0x18,0x18,0x08,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00
,0x00,0x13,0x14,0x44,0x12,0x00,0x00,0x00,0x00,0x00,0x00};

	EPD_Write(GDOControl, sizeof(GDOControl));	// Pannel configuration, Gate selection
	EPD_Write(softstart, sizeof(softstart));	// X decrease, Y decrease
	EPD_Write(VCOMVol, sizeof(VCOMVol));		// VCOM setting
	EPD_Write(DummyLine, sizeof(DummyLine));	// dummy line per gate
	EPD_Write(Gatetime, sizeof(Gatetime));		// Gage time setting
	EPD_Write(RamDataEntryMode, sizeof(RamDataEntryMode));	// X increase, Y decrease
	EPD_SetRamArea(0x00,(xDot-1)/8,(yDot-1)%256,(yDot-1)/256,0x00,0x00);	// X-source area,Y-gage area
	EPD_SetRamPointer(0x00,(yDot-1)%256,(yDot-1)/256);	// set ram

```

bogus105, 2017-07-28:
It looks like the information you was looking for, right? i still can't understand display pixels orienation vs RAM - i mean how to organize memory (font libraries, drawing procedures, etc.) to properly display its content on the display.
Are you able to merge the part of the code above with your powerful u8g2 library? It is far beyond my skills. Hey, i'm an oil&gas engineer, not a programmer:)

olikraus, 2017-07-28:
Let's see. I just collected information first.

bogus105, 2017-07-28:
I found what i was trying to say:) This is Scan Mode. See IL3820 datasheet:

https://www.smart-prototyping.com/image/data/9_Modules/EinkDisplay/GDE029A1/IL3820.pdf

bobybobybob, 2017-08-08:
I have that lcd too. Would be perfect using u8g2 because it is easy and perfect good

olikraus, 2017-08-08:
I did not found time to work on this... From where did you buy this device?

bogus105, 2017-08-08:
Oliver we will wait:) Mine came from robotic etc stuff online shop from Poland: botland.com.pl but i've seen them in at least few online shops so far.

bobybobybob, 2017-08-08:
I got it from Ebay   Have a look at http://www.ebay.de/itm/Waveshare-2-9-296-128-ePaper-e-Ink-Display-Modul-mit-SPI-Raspberry-Pi-Arduino/122554080476

or the little one here http://www.ebay.de/itm/Waveshare-1-54-200-200-ePaper-e-Ink-Display-Modul-mit-SPI-Raspberry-Pi-Arduino/112447578775

bobybobybob, 2017-08-08:
Would it help if we sponsor you one of that device for developing get easy for you better have own ?

cimba007, 2017-08-09:
Got my waveshare 2.9" just recently and the demo code drives me crazy. The orientation feels just .. wrong. Anyone got this mystery solved?

bogus105, 2017-08-09:
i tried an example but commented out most of the demo code leaving 'prograss bar' only. After uploading progress bar showed up at the botton of the screen running across the whole screen. But any tries to modify its shape (width / lenght / segment count / speed) was unsuccesful. The original library seems sooo wierd!
Really looking forward for Oliver to have some spare time to look at this:) His u8g2 is the most reliable and powerfull lib for displays i've ever seen!

cimba007, 2017-08-09:
Just so anybody gets the ideo why this display is so strange ..

[image]

The display get initialized with ID = 01 

[image]

I modified the string example to print a little arrow .. which is printed from right to left .. where I would have expected it to be printed from left to right (the origin is "wrong" and the direction too ..

```  const int origin_x = 0;
  const int origin_y = 0;
  Serial.println(">> end of setup <<");

  byte pattern[16] = { 0xFF-0b00000000, 0xFF-0b00011000,0xFF-0b00011000,0xFF-0b00011000,0xFF-0b11111111,0xFF-0b01111110,0xFF-0b00111100,0xFF-0b00011000,
                      0xFF-0b00000000, 0xFF-0b00011000,0xFF-0b00011000,0xFF-0b00011000,0xFF-0b11111111,0xFF-0b01111110,0xFF-0b00111100,0xFF-0b00011000 };
  EPD.EPD_Dis_Part(origin_x+0,origin_x+7,origin_y+8+0,origin_y+8+16-1,(unsigned char *)pattern,1);
  //driver_delay_xms(DELAYTIME);
    
  EPD.EPD_Dis_Part(origin_x+16+0,origin_x+16+7,origin_y+8+0,origin_y+8+16-1,(unsigned char *)pattern,1);
  //driver_delay_xms(DELAYTIME);

  EPD.EPD_Dis_Part(origin_x+16+0,origin_x+16+7,origin_y+32+0,origin_y+32+16-1,(unsigned char *)pattern,1);
```

Since I have very little nubmers as x/y coordinates I expect my arrows to appear near the origin of the display (so rotating the display is no option).

EPD.EPD_Dis_Part(0,7,8+0,8+16-1,(unsigned char *)pattern,1);

olikraus, 2017-08-11:
> Would it help if we sponsor you one of that device for developing get easy for you better have own ?

Thanks for the kind offer, but this is not required.
I try to order one of these bigger displays. However I have to wait a little but longer for the order.

bogus105, 2017-08-11:
Oliver note among Waveshare e-paper displays the biggest one - 4.3 inch 800x600 pixels is controlled in different manner! it uses UART. The rest displays: 1.54inch, 2.13inch and 2.9inch uses that wierd, unfriendly protocol/library. So if you're going to order the biggest to create user-friendly library then the biggest would be 2.9 inch one.

olikraus, 2017-08-11:
Yes, my assumption is this at the moment:

Waveshare 2.9", 296×128, IL3820 (compatible to SSD1608?)
Waveshare 2.13", 250x122, IL3895 (compatible to SSD16xx??)
Waveshare 1.54", 200x200, controller unknown, could be compatible to SSD1607

My goal was to focus on the 2.9".

bogus105, 2017-08-11:
Good to hear that. I have the same one for tests. I can test it on the following platforms: arduino uno, mega, nano and esp8266-12e.

bobybobybob, 2017-08-11:
The Codemaker from GxEPD use 3 different Dingdongs for that 3 Display Formats. https://github.com/ZinggJM/GxEPD

GxGDEP015OC1 => Waveshare 1.54 SPI
GxGDE0213B1 => Waveshare 2.13 SPI
GxGDEH029A1 => Waveshare 2.9 SPI

olikraus, 2017-08-11:
> Good to hear that. I have the same one for tests. I can test it on the following platforms: arduino uno, mega, nano and esp8266-12e.

So you would be able to test the 200x200 device? There is already a u8g2 constructor for the SSD1607. Will this constructor work?

> The Codemaker from GxEPD use 3 different Dingdongs for that 3 Display Formats. https://github.com/ZinggJM/GxEPD

Somehow they do not reveal the controller. It should be something like ILxxxx or SSDxxxx.

bogus105, 2017-08-11:
Oliver, i have 2.9 inch 296x128 pixels only.

bobybobybob, 2017-08-11:
So you would be able to test the 200x200 device?
-------------
Dear Friend,
i will order now that display and when it is here i can test it for you. Only need then a arduino esp code blabla for Programm it like you want.

olikraus, 2017-08-11:
Please note that the esp32 is not yet supported by u8g2 (well actually the esp32 core library is not yet fully Arduino compatible)

bobybobybob, 2017-08-11:
I can test (pro mini, nano, uno, mega, ESP07, ESP12e, ESP12F, ESP32 and stm32F103)

cimba007, 2017-08-11:
I tried to visualize what the original "example" for the display is doing and came to the conclusion .. that it is fucked up.

Not only are the Y-Start and End parameters swapped and inconsistent in the example (function EPD_Dis_Part calling EPD_part_display) .. x and y are swapped all together .. and its a real mess .. I wonder why the fonts still come up the right way .. I guess minus and minus results in plus.

[image]

[IL3820.zip](https://github.com/olikraus/u8g2/files/1218693/IL3820.zip)

First of all .. how is the display oriented and what is x and y?

[image]

Suggests that the controller supports up to 320 rows which is 
[image]

Orange bars try to represent the actual bytes (8bit) that are written one by one to the ram. (remember golden pins from the first picture)

[image]

So my approch was .. that I want to write some data to the origin!!! x = 0 and y = 0

The pictures in _8.3 Data Entry Mode Setting (11h)_ are pretty confusing .. so reading the text will help.

When AM = “1”, the address counter is updated in the Y direction.

ID [1:0]="11”
X: increment
Y: increment

This should write the first byte in the position at bottom left, the next byte right next to it (right) as y is increased .. and if it should reach the end of the y-window it would increment x and write one row to the top .. beginning at left.

So what should be done to work with this concept?

Remember by arrow pattern?

```
  byte pattern[16] = { 0xFF-0b10000001, 0xFF-0b00011000,0xFF-0b00011000,0xFF-0b00011000,0xFF-0b11111111,0xFF-0b01111110,0xFF-0b00111100,0xFF-0b00011000,
                      0xFF-0b00000000, 0xFF-0b00011000,0xFF-0b00011000,0xFF-0b00011000,0xFF-0b11111111,0xFF-0b01111110,0xFF-0b00111100,0xFF-0b00011000 };
```

Visualisation:
[image]

To stay as close as possible to the datasheet and my thoughts from the beginning my source code looks like this.

  uint8_t xStart;
  uint8_t xEnd;
  uint8_t yStart;
  uint8_t yEnd;    // End must be greater than start
  
```
  xStart = 32+0;
  xEnd = 32+7;
  yStart = 0;
  yEnd = 7;    // End must be greater or equal? than start
  
  EPD.EPD_part_display(xStart/8,xEnd/8,yStart%256,yStart/256,yEnd%256,yEnd/256);    
  EPD.EPD_WriteDispRam(xEnd-xStart, yEnd-yStart+1,pattern);
  EPD.EPD_Update_Part();
  driver_delay_xms(500);
  EPD.EPD_part_display(xStart/8,xEnd/8,yStart%256,yStart/256,yEnd%256,yEnd/256);  
  EPD.EPD_WriteDispRam(xEnd-xStart, yEnd-yStart+1,pattern);
```

What would you expect from this?

[image]

If you have ever played with the SSD1306 OLED-Display this part from its datasheet might be familiar as the SSD1306 has a similar memory structure.

[image]

If anyone got any questions to my writeup freel free to ask.

olikraus, 2017-08-11:
wow, looks like i need to have this display in my lab also.

cimba007, 2017-08-11:
I will try to sum up my findings and strip down the code to provide a really easy to understand and use example. Maybe this will help you to impelment this display to u8g2

cimba007, 2017-08-11:
**Hardware:**
Arduino ProMini (1Mhz internal osc.) **@ 3.3Volt THE DISPLAY MIGHT BE DESTROYED ON 5VOLT**

**Software:** 
Arduino 1.8.2

**Connection:**
BUSY (purple) D7
RST (white) D8
DC (green) D9
CS (orange) D10
CLK (yellow) D13
DIN (blue) D11
GND (black) Ground
3.3V (red) 3.3Volt

As promised I stripped out unneded stuff and changed the font to a progmemfont inspired from https://github.com/greiman/SSD1306Ascii/blob/master/SSD1306Ascii/src/fonts/Adafruit5x7.h

This results in a very little working example. There is **NO SERIAL OUTPUT** as this would only bloat the size of the sketch.

Der Sketch verwendet **3462 Bytes (10%) des Programmspeicherplatzes**. Das Maximum sind 32256 Bytes.
Globale Variablen verwenden **118 Bytes (5%) des dynamischen Speichers**, 1930 Bytes für lokale Variablen verbleiben. Das Maximum sind 2048 Bytes.

[epd_2_9_easy.zip](https://github.com/olikraus/u8g2/files/1218860/epd_2_9_easy.zip)

**Bonus:** (not in the example) 
With the Arduino's voltage reulator removed, the atmega328p in powerDown mode and the display reset with a SWRESET (command 0x12) the whole "project" consumes ~50µA with the ram-data preserved.

bogus105, 2017-08-13:
what a great job cimba007. What do you mean by:

> After working some time with this display .. I get very disappointed .. just from removing the main power while having some capacitors in line (which will result in voltage < 2.4V) the memory can become corrupted. This corruption can not be reset by hardware-reset pin .. only be completly disconnecting the display from power ..<

cimba007, 2017-08-13:
I had some rare cases where the display would not erase black Pixels when writing new content over old one. I am not sure if it was caused by some old parts of the init Code of the example.

bogus105, 2017-08-13:
ok, got it.

olikraus, 2017-08-14:
Added 296x128 device for u8g2.
Of course this is not tested, because i do not own this device.

U8G2_IL3820_296X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	// WaveShare 2.9 inch eInk/ePaper Display

It would be great if someone code test this device with u8g2.

olikraus, 2017-08-14:
The lastest u8g2 lib is available here:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Please use the "add lib zip file" menu of the Arduino IDE for the installation.

olikraus, 2017-08-14:
hmmm.... i guess i need to order the display first for myself :-(

bobybobybob, 2017-08-16:
The 1.5 Waveshare is now here. Is testing ==>  So you would be able to test the 200x200 device? There is already a u8g2 constructor for the SSD1607. Will this constructor work?

Current actual or no need because failure with 2,9 waveshare?

bobybobybob, 2017-08-16:
Here is Video with 200x200  1.54 Waveshare Display. 

https://www.youtube.com/watch?v=IqKgGfVnwgU&feature=youtu.be

olikraus, 2017-08-16:
Great. Looks good. Thanks for testing.

bogus105, 2017-08-17:
so what is the problem with 2.9inch display? It uses different controller  (SSD1608 on 2.9inch and SSD1607 on 1.54inch) but what is the point use use different approach to the way data is organized on the screen? Does the problem lie in initial settings (type of RAM scanning: AM and ID bits) or what? I thought whese screens are only differ in resolution.

olikraus, 2017-08-17:
At the moment i have no idea what is wrong with the u8g2 device for the 2.9" display.
May plan is to order this display. However, i can not order before next week and then it will probably take another 2 to 4 weeks shipping time.

olikraus, 2017-08-18:
ok, i have created another test release with your suggestions

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

One difference: U8g2 usually renders small pages and sends them one by one.

bobybobybob, 2017-08-18:
Hello,
i have tested it just now. The 2.9 show same dot Wall, maybe a bit darker than the first one.

bobybobybob, 2017-08-18:
What does "enable 16 bit mode for this display!" mean?

olikraus, 2017-08-18:
Thanks for testing, i really need that e-paper device in my lab.

olikraus, 2017-08-18:
oh, i didn't notice that there is a ebay seller in berlin. This should deliver more quickly to south part of Germany :-)

bogus105, 2017-08-20:
Great to hear that Oliver. I'm so curious if You can manage this screen. I can't do it. To do it good programming skills are essential...

olikraus, 2017-08-26:
some sucess. At least graphics output seems to be ok now, however the LUT is probably wrong.
I have created a beta release if someone here wants to do some testing:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

olikraus, 2017-08-27:
beta release update (2.17.8)

bobybobybob, 2017-08-27:
where download for test?

olikraus, 2017-08-27:
Here:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Thanks for testing :-)

bobybobybob, 2017-08-27:
It is no happy for developer when nobody test the work if it works fine. That is my little part for good user experince and make developer also happy and user like me too. 

https://youtu.be/sO8TkhPyc6g
Congratulation, Epaper Waveshare 2.9 show now real und readable data  ^_^

(For Clock Demo i have make bigger Numbers so dont wonder why it is so big. )

Maybe for refresh there is need for optimize so the elements are all dark without little greydark is showing from past to new showing.

olikraus, 2017-08-27:
Thanks, great videos. I agree, somehow the dark/white programming waveform does not fit. Some grey remains.

olikraus, 2017-08-27:
Here is a picture of my own setup:

https://github.com/olikraus/u8g2/wiki/img/uno_waveshare29.jpg

ZinggJM, 2017-08-27:
The "Codemaker" has updated GxEPD. Some of your issues are known to me.

bobybobybob, 2017-08-27:
I can feel the mystic Waveshare that the Codemaker-Force is with you too. 

Great Job Codemakers

ZinggJM, 2017-08-28:
I feel the power of u8g2 coming to the e-paper world, and am very happy about it!

Some information:
Both the code of the Waveshare examples and of my GxEPD are based on demo code from Dalian Good Display. Dalian Good Display has recently made the actual demo code available for free download.
The actual demo code for 2.9inch has an updated wavetable (Look Up Table) for full and for partial refresh, I didn't check if Waveshare have updated their demo code.
The Address Mode bits do not work correctly on all 3 displays, as far as have tried. 2.9inch is the one that should work as far as I remember. This is a pending issue in my GxEPD also. My test code has a drawCornerTest for this reason.

I have mail contact with Dalian Good Display, and I know they work on refresh speed.
We should see more from them and Waveshare soon, the 4.2inch is already available.

olikraus, 2017-08-28:
Sounds good. Can you paste the latest LUT here?

ZinggJM, 2017-08-28:
#define xPixelsPar (GxGDEH029A1_X_PIXELS -1 )
#define yPixelsPar (GxGDEH029A1_Y_PIXELS -1 )

// old version
//const uint8_t LUTDefault_full[] =
//{
//  0x32,  // command
//  0x02, 0x02, 0x01, 0x11, 0x12, 0x12, 0x22, 0x22, 0x66, 0x69, 0x69, 0x59, 0x58, 0x99, 0x99,
//  0x88, 0x00, 0x00, 0x00, 0x00, 0xF8, 0xB4, 0x13, 0x51, 0x35, 0x51, 0x51, 0x19, 0x01, 0x00
//};

const uint8_t LUTDefault_full[31] = 
{
  0x32,  // command //C221 25C Full update waveform
  0x50, 0xAA, 0x55, 0xAA, 0x11, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

const uint8_t LUTDefault_part[] =
{
  0x32,  // command
  0x10, 0x18, 0x18, 0x08, 0x18, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x13, 0x14, 0x44, 0x12, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

uint8_t GDOControl[] = {0x01, yPixelsPar % 256, yPixelsPar / 256, 0x00}; //for 2.9inch
uint8_t softstart[] = {0x0c, 0xd7, 0xd6, 0x9d};
uint8_t VCOMVol[] = {0x2c, 0xa8};  // VCOM 7c
uint8_t DummyLine[] = {0x3a, 0x1a}; // 4 dummy line per gate
uint8_t Gatetime[] = {0x3b, 0x08};  // 2us per line
uint8_t RamDataEntryMode[] = {0x11, 0x01};  // Ram data entry mode

ZinggJM, 2017-08-28:
1.54inch

#define xPixelsPar (GxGDEP015OC1_X_PIXELS - 1)
#define yPixelsPar (GxGDEP015OC1_Y_PIXELS - 1)

// old version
//const uint8_t LUTDefault_part[31] =
//{
//  0x32,  // command
//  0x10, 0x18, 0x18, 0x08, 0x18, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
//  0x00, 0x00, 0x00, 0x00, 0x00, 0x13, 0x14, 0x44, 0x12, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
//};

const uint8_t LUTDefault_part[31] =
{
  0x32,	// command //C221 25C partial update waveform
  0x10, 0x18, 0x18, 0x08, 0x18, 0x18, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x13, 0x14, 0x44, 0x12, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

// old version
//const uint8_t LUTDefault_full[31] =
//{
//  0x32,  // command
//  0x02, 0x02, 0x01, 0x11, 0x12, 0x12, 0x22, 0x22, 0x66, 0x69, 0x69, 0x59, 0x58, 0x99, 0x99,
//  0x88, 0x00, 0x00, 0x00, 0x00, 0xF8, 0xB4, 0x13, 0x51, 0x35, 0x51, 0x51, 0x19, 0x01, 0x00
//};

const uint8_t LUTDefault_full[31] = {
  0x32,	// command //C221 25C Full update waveform
  0x50, 0xAA, 0x55, 0xAA, 0x11, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0xFF, 0xFF, 0x1F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

// old version
//uint8_t GDOControl[] = {0x01, 0xC7, 0x00, 0x00}; //for 1.54inch
//uint8_t softstart[] = {0x0c, 0xd7, 0xd6, 0x9d};
//uint8_t VCOMVol[] = {0x2c, 0xa8};  // VCOM 7c
//uint8_t DummyLine[] = {0x3a, 0x1a}; // 4 dummy line per gate
//uint8_t Gatetime[] = {0x3b, 0x08};  // 2us per line
//uint8_t RamDataEntryMode[] = {0x11, 0x01};  // Ram data entry mode

uint8_t GDOControl[] = {0x01, yPixelsPar % 256, yPixelsPar / 256, 0x00}; //for 1.54inch
uint8_t softstart[] = {0x0c, 0xd7, 0xd6, 0x9d};
uint8_t Rambypass[] = {0x21, 0x8f};   // Display update
uint8_t MAsequency[] = {0x22, 0xf0};  // clock
uint8_t GDVol[] = {0x03, 0x00}; // Gate voltage +15V/-15V
uint8_t SDVol[] = {0x04, 0x0a}; // Source voltage +15V/-15V
uint8_t VCOMVol[] = {0x2c, 0x9b}; // VCOM 7c
uint8_t BOOSTERFB[] = {0xf0, 0x1f}; // Source voltage +15V/-15V
uint8_t DummyLine[] = {0x3a, 0x1a}; // 4 dummy line per gate
uint8_t Gatetime[] = {0x3b, 0x08}; // 2us per line
uint8_t BorderWavefrom[] = {0x3c, 0x33}; // Border
uint8_t RamDataEntryMode[] = {0x11, 0x01}; // Ram data entry mode

2.13inch

#define xPixelsPar (GxGDE0213B1_X_PIXELS-1)
#define yPixelsPar (GxGDE0213B1_Y_PIXELS-1)

const uint8_t LUTDefault_full[] =
{
  0x32,  // command
  0x22, 0x55, 0xAA, 0x55, 0xAA, 0x55, 0xAA, 0x11, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x1E, 0x01, 0x00, 0x00, 0x00, 0x00
};

const uint8_t LUTDefault_part[] =
{
  0x32,  // command
  0x18, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0x0F, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

uint8_t GDOControl[] = {0x01, yPixelsPar % 256, yPixelsPar / 256, 0x00}; //for 2.13inch
uint8_t softstart[] = {0x0c, 0xd7, 0xd6, 0x9d};
uint8_t VCOMVol[] = {0x2c, 0xa8};  // VCOM 7c
uint8_t DummyLine[] = {0x3a, 0x1a}; // 4 dummy line per gate
uint8_t Gatetime[] = {0x3b, 0x08};  // 2us per line
uint8_t RamDataEntryMode[] = {0x11, 0x01};  // Ram data entry mode

olikraus, 2017-08-28:
the first LUT is for the 2.9" version?

ZinggJM, 2017-08-28:
Yes, I thought the name in the first two lines should be clear enough:

#define xPixelsPar (GxGDEH029A1_X_PIXELS -1 )
#define yPixelsPar (GxGDEH029A1_Y_PIXELS -1 )

I would appreciate if you took a look at my library; any comments?

[83 more replies]

closed 2017-09-10 by olikraus