olikraus/u8g2 #320: can you help me?

laterrps, 2017-07-28:
I would like to wirite in an oled ssd1306 something like
Variable=77
and below a graph in real time something  like PPG signal but i cant do it because i dont know how to update this signal without clearing all the screen. i would like to do something like in this video at 1:07 i mean how they graph the signal

olikraus, 2017-07-28:
The first/next page loop automatically clears the screen. The fullbuffer constructor includes also a screen clear function.

laterrps, 2017-08-03:
does your library have something like u8g2.clear(0,5) and this function clear these determined area from the screen? I do this question because sometimes we dont wanna to clear whole screen instead of this just clear determined area

olikraus, 2017-08-03:
You can use the box command.

closed 2017-08-12 by olikraus