olikraus/u8g2 #321: Mirror image and offset issue with ST7565P Zolen

SybyAbraham, 2017-07-29:
I recently received a **Zolentech ST7565P 128X64 SPI LCD** module I got off of eBay. I'm assuming it's not quite the same as the ST7565 controller. 

Using the harware SPI ST7565 Zolen constructor I was able to get the following result. 

[image]

**The image was mirrored, off by 4 pixels and the contrast was too high.** 

In the file **u8x8_d_st7565.c** under `static const uint8_t u8x8_d_st7565_zolen_128x64_init_seq[] = {`

- I commented out line 257 (v2.15.2)
` // U8X8_C(0x0a1),		                /* ADC set to reverse */`

- and as set the contrast to 
`U8X8_CA(0x081, 0x000), /* set contrast, contrast value, EA default: 0x016 */` 

[image]

After these changes, I was able to get a clean fully functioning display. I'm not entirely sure if these are the appropriate fixes, I'll leave that up to you. I tested the fix with `setFlipMode(1)` and the `U8G2_R0` `U8G2_R1` `U8G2_R2` `U8G2_R3` `U8G2_MIRROR` arguments in the constructor which all worked flawlessly. I tested it using both hardware and software SPI on the Arduino Mega 2560 and on an Arduino UNO R3.

**Thanks for  such an amazing library!**

olikraus, 2017-07-30:
Nice finding. Looks like there are different types of LCDs. 
However things do not really fit together to implement another constructor. Commenting line 257 is probably overwritten by setFlipMode(1). So i have a request: Can you sent me the result of the DOGM128 constructor?  I mean, will the DOGM128 constructor work better (ok contranst might be too high, but i am more interessted in  the mirror effect)

Thanks,
Oliver

SybyAbraham, 2017-07-30:
Hi Oliver,

I just tried the DOGM128 constructor. You can check out the results below. Like you said the contrast was high. No mirror image this time, just upside down and a strange rectangle at the corner. I believe I tried out every other constructor for the ST7565 and they all had some issue with this particular LCD. Earlier today I was testing out the display with **u8glib** and strangely enough, the **64128N** constructor from u8glib works flawlessly without any adjustment at all. 

Thanks,
Syby

[image]

olikraus, 2017-07-30:
Great. I have recently ported the 61128N constructor to u8g2. However it is not yet released (#292).
Can you download the latest developer release from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Then install the zip file in the Arduino IDE menu "add zip library".
After this you should be able to see the constructor in the examples:

U8G2_ST7565_64128N_1_4W_SW_SPI

Could you test this as well. Will this work?

SybyAbraham, 2017-07-30:
That works! The image is flipped but after setFlipMode(1) everything works great!

olikraus, 2017-07-31:
ok, great. So we can close this issue?

SybyAbraham, 2017-08-01:
Yup! Thanks for the help!

olikraus, 2017-08-01:
:-)

closed 2017-08-01 by olikraus