olikraus/u8g2 #322: ST7920 and ESP8266 HW SPI

pccr10001, 2017-07-29:
Hello.

I tried to connect ST7920 128x64 LCD to ESP8266.
When I tried SW SPI is working but slow.
After switch to HW SPI and nothing showed on LCD.

I found a solution that I changed spi_mode in u8x8_st7920_128x64_display_info(u8x8_d_st7920.c) to 0.
```C
#if defined(ESP8266)
  /* spi_mode = */ 0,
#else
  /* spi_mode = */ 3,
#endif
```

and everything worked.

olikraus, 2017-07-29:
I think this is a bug in the esp8266 lib.

closed 2017-08-27 by olikraus

microdimmer, 2018-03-06:
That bug still exists, pccr10001's solution worked for me

olikraus, 2018-03-06:
I posted the problem already:

https://github.com/esp8266/Arduino/issues/2416

Misiu, 2018-12-27:
@olikraus I know this is a bit old issue but does the problem still exists?
 I'm planning to use ST7920 with Wemos D1 and I don't know should I stick with this display or should I search for a different display.

olikraus, 2018-12-27:
I have not heard any news regarding this problem.

Misiu, 2018-12-27:
@olikraus I've already posted a comment in [2416](https://github.com/esp8266/Arduino/issues/2416) issue.
Hopefully this will be finally resolved.

llocani, 2020-01-11:
Hello, i did this change in the u8x8_d_st7920.c in src/clib/u8x8_d_st7920.c and recompile my project but the problem remains.
Do i have to do something else like tell arduino studio to recompile this file?
Thanks in advance.

olikraus, 2020-01-11:
Nothing needs to be modified, the problem should be fixed in the ESP software. My pull request was included: https://github.com/esp8266/Arduino/pull/5948

See also: https://github.com/olikraus/u8g2/issues/53

llocani, 2020-01-11:
Thanks for de quick response.
Should i update the u8g2 and spi libraries? a particular version?
Thanks in advance

olikraus, 2020-01-11:
You only need to update the ESP8266 environment (which includes SPI lib)

llocani, 2020-01-12:
Then i guess iv'e killed my display : (
Thank you anyway!