olikraus/u8g2 #323: callback function for the user interface procedures

olikraus, 2017-07-30:
https://forum.arduino.cc/index.php?topic=492110.new#new

maybe allow a callback function for the get key procedures. Could be done for cpp only... Arduino specific in the .cpp part.

enlinea777, 2018-08-16:
I was trying to implement U8g2, and I found the typical WDT error.
So, implement a few
yield ():
in calls to generate output buffers and others while and for
with what I got that the screens worked in the ESP8266 without it being blocked by such a long loop.

I hope you implement it.

-Note: I do not know how to collaborate but I made a fork of the project and it works very well with esp8266
[image]

olikraus, 2018-08-16:
yield() is called in the user interface functions, but nowhere else.

enlinea777, 2018-08-16:
yield ():
for those who are accustomed to arduino with their ATMEL, yes.
but for the ESP8266 it is 32 bits, here you can call it from any place because in the adaptation for arduino or Espressif SDK, it is contemplated, so it can be called from anywhere.
Only for the Esp8266, I do not know if it works also for ESP32 
[image]

olikraus, 2020-01-19:
closing this, not sure whether there is still a problem with yield()

closed 2020-01-19 by olikraus