olikraus/u8g2 #325: SSD1306 I2C u8g2 and u8x8

GeorgeEarslight, 2017-08-03:
Hello,
Thank you for the free library!
I've ported to an STML151RDT6 using the CubeMX libraries. I'm currently using the bit banged driver, to just draw something. 
I've managed to work with u8x8, so I'm guessing that my IO and delay functions are OK. 

`u8x8_Setup(&u8x8, u8x8_d_ssd1306_128x64_noname, u8x8_cad_ssd13xx_i2c, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_stm32);`
`u8x8_InitDisplay(&u8x8);`
`u8x8_ClearDisplay(&u8x8);`
`u8x8_SetPowerSave(&u8x8, 0);`
`u8x8_SetFont(&u8x8, u8x8_font_amstrad_cpc_extended_r);`
`u8x8_DrawString(&u8x8, 0,0, "Hello world!");`

When I try to use u8g2 I can't draw anything.

`u8g2_Setup_ssd1306_i2c_128x64_noname_2(&u8g2, U8G2_R0, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_stm32);`
`u8g2_InitDisplay(&u8g2);`
`u8g2_ClearDisplay(&u8g2);`
`u8g2_SetPowerSave(&u8g2, 0);`
`u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);`
`u8g2_DrawStr(&u8g2, 0, 0, "Hello world!");`

I've tried with noname_1, noname_2, and noname_f. I've also worked with the older library on hardware I2C.
If you could hint me any issues, I'd be happy to contribute with the port.

Best regards,
George

olikraus, 2017-08-03:
You mean STM32Lxxx?
I have ported u8g2 to STM32L0 once:

gpio procedure (I also only added software I2C):
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8g2_rtc/u8x8cb.c

Init code will look like this (https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8g2_rtc/main.c#L228):
```
  u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_stm32l0);
  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);
  u8g2_SetFont(&u8g2, u8g2_font_6x12_tf);
  u8g2_ClearBuffer(&u8g2);
  u8g2_DrawStr(&u8g2, 0,12, "STM32L031");
  u8g2_DrawStr(&u8g2, 0,24, u8x8_u8toa(SystemCoreClock/1000000, 2));
  u8g2_DrawStr(&u8g2, 20,24, "MHz");
  u8g2_SendBuffer(&u8g2);
```

Did you transfer the Buffer?

Oliver

GeorgeEarslight, 2017-08-03:
You're right, I was missing the SendBuffer command!
It seems I fell into an RTFM error, I'm terribly sorry for the trouble. 
I've tried the u8x8 lib first and thought you abolished the `NextPage` method from the older lib, but then decided to try it and worked. Then, I read your answer here.

My offer still stands in any case. When I'm done with the hardware driver I'll write up a guide and send it your way for the STM32 documentation.

Thanks again for your contribution and help!

olikraus, 2017-08-27:
ok, i think we can close this.

closed 2017-08-27 by olikraus