olikraus/u8g2 #326: U8G2 as class member

thasler, 2017-08-03:
When I use the library like in the tutorial and call u8x8.begin() in the setup() function, everything works fine.

I want to have a class "Display" which handles all my display stuff. Therefore I wanted to define u8g2 as a class member of "Display" and instantiate it when the constructor of the Display class is called. In the constructor I then call u8x8.begin(). However this does not work. The call to u8x8.begin() is not working...
How can I define and use the u8x8 class as a member of my Display class?

Kind regards,
Thomas

olikraus, 2017-08-03:
Difficult to say without any kind of code...

thasler, 2017-08-03:
I solved the problem on my own. Problem was that I called  u8x8.begin() in the constructor of my Display class, which is called before the Arduino setup() method...

I used an init method so that u8x8.begin() is called in the setup() and then it worked.

closed 2017-08-03 by thasler

olikraus, 2017-08-03:
:-)

treii28, 2018-04-28:
I'm trying to do something similar and similarly ran begin() within an init to solve the problem when starting. But now where I'm trying to draw the screen it is crashing the arduino. To make sure it wasn't something in my code, I pulled the drawLogo() function and used that code to create a method in my object to call in the do...while loop and it still crashes.

relevant code:

```
#ifndef Arduino_h
#include <Arduino.h>
#endif

#ifndef _U8G2LIB_HH
#include <U8g2lib.h>
#endif

class RPUISSD1306 {
private:
    // private variables
    U8G2 u8g2;

public:
    // constructors
    RPUISSD1306(void) {};

    void initDisplay()
    {
        U8G2_SSD1306_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, 3, 11, 10, 9, 8);
        u8g2.begin();
    };
    void drawLogo()
    {
       /* code from example sketch */
    };

    void drawScreen() {
        u8g2.firstPage();
        do {
            drawLogo();
        } while( u8g2.nextPage() );
    };
}

```

olikraus, 2018-04-29:
Probably I can not support here, but I wonder what the root cause is.

AlexLyo, 2024-02-26:
Hi.
Its worked if:
class gclcd{
    private:
const char *TAG = "gclcd";
private: 
uint16_t heigh=64;
uint16_t width=128;
U8G2 _u8g2;
 const bool isgraphic=true;
public: 
gclcd();
void init();
void printWelcom();
void printTestStr();

};
gclcd::gclcd()
{
   vTaskDelay(100/portTICK_PERIOD_MS);
  U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/PIN_CLK, /* data=*/PIN_MOSI, /* CS=*/PIN_CS, /* reset=*/PIN_RESET);
   vTaskDelay(100/portTICK_PERIOD_MS);
   _u8g2=u8g2;//<<<<<<<<<<THIS!!!!<<<<<<<<<<<<<<<<<<<<<<<<
}
void gclcd::init()
{
  _u8g2.begin();
  _u8g2.initDisplay();
  _u8g2.clearBuffer();
  _u8g2.clear();
  _u8g2.enableUTF8Print();
   vTaskDelay(100/portTICK_PERIOD_MS);
}

gclcd mylcd;
//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 18, /* data=*/ 23, /* CS=*/ 5, /* reset=*/ 22); 

void setup() {
  uint8_t u;
   Serial.begin(921600);
   mylcd.init();
   etc....