olikraus/u8g2 #328: setContrast Not Working

megamegacrazy, 2017-08-09:
setContrast function is not working with SSD1322 due to an issue in "ssd1322.c". The wrong address is used (0x081 is the address for SSD1325, C1 is for 1322):

#ifdef U8X8_WITH_SET_CONTRAST
    case U8X8_MSG_DISPLAY_SET_CONTRAST:
      u8x8_cad_StartTransfer(u8x8);
      u8x8_cad_SendCmd(u8x8, 0x081 ); <--------------
      u8x8_cad_SendArg(u8x8, arg_int );	/* ssd1325 has range from 0 to 255 */
      u8x8_cad_EndTransfer(u8x8);
      break;
#endif

This fixes it:

#ifdef U8X8_WITH_SET_CONTRAST
    case U8X8_MSG_DISPLAY_SET_CONTRAST:
      u8x8_cad_StartTransfer(u8x8);
      u8x8_cad_SendCmd(u8x8, 0xc1 ); <---------------
      u8x8_cad_SendArg(u8x8, arg_int );	/* ssd1322 has range from 0 to 255 */
      u8x8_cad_EndTransfer(u8x8);
      break;
#endif

olikraus, 2017-08-10:
Wow, nice finding, thanks

olikraus, 2017-08-11:
ok, seems to be fixed in 2.6.x

closed 2017-08-11 by olikraus