olikraus/u8g2 #329: u8g2 vs u8x8 for scrolling 2 lines 32 pix height on small OLED

blotfi, 2017-08-10:
Hi and thanks for your lib

I am trying scrolling routines for my webradio (it has to display radio name and title) on a very small OLED screen wih a nodeMCU (ESP8266).
I need to display only 2 big lines (32 pixel height per line) and scroll each one of them and roll.

My config is software I2C with SSD1306:
`U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ D4, /* data=*/ D3, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
`
It seems that to be readable, I have to go in:
u8x8.draw2x2String(0, 5, "Big");
are there more readable 8x8 than other when going in big 2x2 like u8x8_font_chroma48medium8_r

Are the 8x8 rendering for scrolling **faster** than u8g2
Are there u8x8 font native 2x2 ie big ones. I don't need graphics

Thanks for helping me to find the best solution

olikraus, 2017-08-10:
U8x8 interface is called U8x8 because it works with 8x8 pixel tiles only. As a consequence you have to live with several limitations:
- Font chars fit to a 8x8 tile
- Scrolling only is possible by moving tiles, that means you always have to scroll by 8 pixel at a time.
- draw2x2String is more like a add on. It scales up the 8x8 pixel glyph to a 16x16 box. Still i guess you need to have glyphs with a height of 32 pixel.

I suggest to try u8g2 instead. There are much more and especially bigger fonts. Also the ESP8266 should be fast enough to handle this. Moreover, scolling can be done on a pixel basis.

> are there more readable 8x8 than other when going in big 2x2 like u8x8_font_chroma48medium8_r

No, you need to use u8g2

> Are the 8x8 rendering for scrolling faster than u8g2

I guess the ESP8266 is to fast to notice the difference. Glyph rendering is more complicated for u8g2 compared to u8x8, so u8x8 glyph drawing is faster.

blotfi, 2017-08-10:
Yes I used the example, but it does only scroll small sentences and not long ones.

I tried 8x8, it is too ugly in 2x2, comparing to u8g2 of course, so I will use this latter
but this it is still slow at 1 pixel scroll / 10 ms (even with no delay...)
NodeMCU is not so much fast (80 MHz) it is not an ESP32 blasting runner

I am trying 
u8g2_font_logisoso32_tr
and
u8g2_font_logisoso30_tr
171 width

what does " this value must be lesser than 128 unless U8G2_16BIT is set" means exactly ?

olikraus, 2017-08-10:
> but this it is still slow at 1 pixel scroll / 10 ms (even with no delay...)
You could try 2, 3 or 4 pixel movements, I mean, use "offset-=2;" or some bigger value.

> what does " this value must be lesser than 128 unless U8G2_16BIT is set" means exactly ?

See here:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L153

blotfi, 2017-08-10:
Guess what, I boosted to 160 Mhz, it is better and still works
now the code for 2 lines is 👍 
`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

u8g2_uint_t offset1, offset2;			// current offset for the scrolling text
u8g2_uint_t width1, width2;			// pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
const unsigned int text1_y0=31, text2_y0=66;
const char *text1 = "Radio Chaine 3 ";  // scroll this text from right to left
const char *text2 = "Katy Perry - Miss You More ";  // scroll this text from right to left

/*
  U8glib Example Overview:
    Frame Buffer Examples: clearBuffer/sendBuffer. Fast, but may not work with all Arduino boards because of RAM consumption
    Page Buffer Examples: firstPage/nextPage. Less RAM usage, should work with all Arduino boards.
    U8x8 Text Only Example: No RAM usage, direct communication with display controller. No graphics, 8x8 Text only.
*/

// Please UNCOMMENT one of the contructor lines below
// U8g2 Contructor List (Frame Buffer)
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected

U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ D4, /* data=*/ D3, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display

void setup(void) {
  u8g2.begin();
//  u8g2.setFont(u8g2_font_ncenB14_tr);  // choose a suitable font
//  u8g2.setFont(u8g2_font_inb30_mr);  // set the target font to calculate the pixel width
  u8g2.setFont(u8g2_font_logisoso30_tr);  // set the target font to calculate the pixel width
  width1 = u8g2.getUTF8Width(text1);    // calculate the pixel width of the text
  width2 = u8g2.getUTF8Width(text2);    // calculate the pixel width of the text
  u8g2.setFontMode(0);		// enable transparent mode, which is faster
  offset1 = 0;              // start over again
  offset2 = 0;              // start over again
}

void loop(void) {
  u8g2_uint_t x;
  u8g2.setFont(u8g2_font_logisoso30_tr);    // set the target font
  u8g2.firstPage();
  do {
    // draw the scrolling text at current offset
    x = offset1;
    do {                // repeated drawing of the scrolling text...
      u8g2.drawUTF8(x, text1_y0, text1);      // draw the scolling text
      x += width1;           // add the pixel width of the scrolling text
    } while( x < u8g2.getDisplayWidth() );    // draw again until the complete display is filled
    
    x = offset2;
    do {                // repeated drawing of the scrolling text...
      u8g2.drawUTF8(x, text2_y0, text2);      // draw the scolling text
      x += width2;           // add the pixel width of the scrolling text
    } while( x < u8g2.getDisplayWidth() );    // draw again until the complete display is filled
    
//    u8g2.setFont(u8g2_font_inb16_mr);		// draw the current pixel width
//    u8g2.setCursor(0, 58);
//    u8g2.print(width);					// this value must be lesser than 128 unless U8G2_16BIT is set
    
  } while ( u8g2.nextPage() );
  
  offset1-=2;              // scroll by one pixel
  if ( (u8g2_uint_t)offset1 < (u8g2_uint_t)-width1 )   offset1 = 0;             // start over again
  offset2-=4;             // scroll by one pixel
  if ( (u8g2_uint_t)offset2 < (u8g2_uint_t)-width2 )   offset2 = 0;             // start over again
    
  //delay(1);							// do some small delay
}`

closed 2017-08-11 by olikraus

blotfi, 2017-08-12:
Just to add that I ported it on ESP32 (WEMOS Lolin32) for my Web Radio [https://github.com/Edzelf/ESP32-Radio/issues/33](url) 
and the ESP32 is very fast... he also has a H/W I2C so u8g2 transmit faster to the OLED
I will stop bying ESP8266 or Arduino... this is the best low cost microcontroller, it has BT BLE, Wifi, on board lipo charger... and we can programm it form Arduino IDE
There are different version, even with on board OLED
[https://www.aliexpress.com/item/4-MB-Flash-WEMOS-Lolin32-V1-0-0-WIFI-Bluetooth-Card-Based-ESP-32-ESP-WROOM/32823891656.html](url)
[https://www.aliexpress.com/wholesale?catId=0&initiative_id=AS_20170811183725&SearchText=wemos+lolin+esp32](url)

Hope this help

olikraus, 2017-08-12:
So, you can program the ESP32 from within the Arduino IDE?

underwoodblog, 2017-08-12:
Yes, it's now easy to integrate:

https://github.com/espressif/arduino-esp32

Check out:

https://www.youtube.com/channel/UCu7_D0o48KbfhpEohoP7YSQ/videos

He has lots of good infos about using the new ESP32 with Arduino. Not all features of the ESP32 are integrated yet in the Arduino IDE but it works well so far.

olikraus, 2017-08-12:
I will check, thanks for the information.

blotfi, 2017-08-12:
@olikraus 
Yes it is easy and straight forward, the only thing is that the library dir is not the same as the other boards (Arduino, nano, ESP8266)
[https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md](url)

olikraus, 2017-08-12:
Hmm... I followed the Arduino lib instructions. Specific name was not mentioned.

blotfi, 2017-08-13:
yes, not mentionned 
but you will notice that the libs will not install in the same dir than Arduino stuff

olikraus, 2017-08-13:
So what exactly should be changed?

blotfi, 2017-08-13:
if you want to install additional libraires like yours, you have to copy the .zip content of your lib to where are the ESP32 libraries
for me it is in:
c:\Users\myname\Documents\Arduino\hardware\espressif\esp32\libraries\
instead of traditionnal ones:
c:\Users\myname\Documents\Arduino\libraries\

olikraus, 2017-08-13:
Ok, but there is nothing what U8g2 can do here

blotfi, 2017-08-13:
yes of course, I am just giving you tips to set up correctly ESP32
I had to copy the U8g2 to this directory, because even if you install it with the library manager (traditional way) it will only work for Arduino or ESP8266 and not the ESP32

olikraus, 2017-08-13:
Ah, I see, Thanks