olikraus/u8g2 #330: test esp32 with u8g2

olikraus, 2017-08-12:
from #329 

> https://github.com/espressif/arduino-esp32
Check out:
https://www.youtube.com/channel/UCu7_D0o48KbfhpEohoP7YSQ/videos

olikraus, 2017-08-12:
still no boardfile for Arduino IDE

bobybobybob, 2017-08-18:
Call me "Mr. Tester" ;-D

Have a look at Video  https://youtu.be/Nd6Ffxltc04

1. ESP32 and E-Ink 1.54       Bugs
2. Pro Mini and E-Ink 1.54   OK
3. ESP32 and OLED 128x64 OK

olikraus, 2017-08-27:
Because of the missing boardfile, there is still no simple way to add support for ESP32 to Arduino IDE.
I will close this issue and reopen, once a boardfile is available.

closed 2017-08-27 by olikraus

bobybobybob, 2017-08-27:
But it works

olikraus, 2017-08-27:
Actually I tried everything to make it work, but failed. Maybe Linux is not supported. Anyhow, I will wait until a board package is available.

bobybobybob, 2017-08-27:
My Videos with ESP32 show it work with some device
I use IDE Windows