olikraus/u8g2 #331: Display pixels shifted to the left

smotesko, 2017-08-13:
I have an issue similar to #44 
I'm using an Arduino with 128x64 SSD1306 display (at least that was the controller chip advertised by the seller). 
Loading the Arduino example sketch [full buffer -> Hello World](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_full_buffer/HelloWorld/HelloWorld.ino) and uncommenting the constructor:
```
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
```
I've got displays in two sizes. Code works fine with 0.96" display, but on 1.3" the picture is shifted to the left and there's a bar of noise on the right edge. I've tried some different code examples, including page_buffer -> HelloWord and the behavior is always the same. I suspect there's something special about my 1.3" displays, but I don't know what it can be. Can you give me a hint?

0.96" display:
[image]

1.3" display:
[image]

olikraus, 2017-08-13:
This is not a SSD1306 controller, but probably a SH1106:
Try
U8G2_SH1106_128X64_NONAME_F_HW_I2C
instead.

smotesko, 2017-08-13:
Yes, `U8G2_SH1106_128X64_NONAME_F_HW_I2C` works as intended! I appreciate the help, thank you.

closed 2017-08-13 by smotesko

olikraus, 2017-08-13:
:-)

asokolsky, 2018-06-26:
Thank you!  Resolved my "problem" as well.

olikraus, 2018-06-27:
:)

djherceg, 2019-12-12:
Hello, I've had the exact same problem. Display shifted one pixel to the left and cropped there, and a random bar of pixels on the right edge. I changed the u8g2 object declaration to U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2 as suggested, but a huge question remains - how can I use **random I2C pins** on the ESP32 instead of default ones?

olikraus, 2019-12-12:
The HW I2C constructor supports two extra arguments to specify those "random" pins.
The square brackets denote optional arguments. If you do not need reset, place U8X8_PIN_NONE
U8G2_SH1106_128X64_NONAME_F_HW_I2C(rotation, [reset [, clock, data]])

from here:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1106-128x64_noname

In your case it will be:
```
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 25, 26);
```
also note this:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L43

djherceg, 2019-12-12:
Thank you, you're very kind!

gezzerlol, 2024-01-17:
solved my problem too, thanks!