olikraus/u8g2 #333: ERC240120 UC1608 240x120

floris-b, 2017-08-14:
Hello,

I want to use the ERC240120 from buydisplay.com and got it working with SW and HW SPI on a Teensy 3.2 with: U8G2_UC1608_240X128_F_4W_HW_SPI u8g2(U8G2_R0, 9, 8, 6);

A small problem I have is that the top 8 rows are not shown, it is a 240x120 pixel display
I tried changing the pixel_height but then the top 8 rows are still not visible and I lose the bottom 8 rows.
How can I solve this?

olikraus, 2017-08-14:
I have created a new device for this.
Please download u8g2 from here:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Please use the "add lib zip file" menu of the Arduino IDE for the installation.

Then use:
U8G2_UC1608_ERC240120_F_4W_HW_SPI u8g2(U8G2_R0, 9, 8, 6);

Note for 2.17 release: U8G2_UC1608_ERC240120 not yet added to ino constructor list

floris-b, 2017-08-14:
The update does not work. Maybe my explanation was not clear.
When I run: u8g2.drawFrame(0,0,240,120 )

U8G2_UC1608_240X128: bottom horizontal line is correct at the bottom of the display, top horizontal line does not show

U8G2_UC1608_ERC240120: bottom horizontal line is 8 pixels obove the botom of the display,  top horizontal line does not show ( after a while there is a shift from the top and is shown at the bottom 8 rows )

olikraus, 2017-08-14:
can you attach a picture?

olikraus, 2017-08-14:
Thanks.
I have created u8g2 2.17.2. It is available under the above link.
Will it work now?
Please also test u8x8/FlipMode example. Will there be a text in the upper left corner for both modes?

Thanks for testing!

floris-b, 2017-08-14:
At first it works but after some time there is an error, I am running the page buffer FPS example

[image]
[image]
[image]
[image]

floris-b, 2017-08-14:
Also the default contrast should be set to 80

olikraus, 2017-08-14:
ok, i have fixed the flipmode 1 and updated the contrast.
Will it work now?
The unexpected change after some time might be caused by glitches in your test environment.

floris-b, 2017-08-14:
The first time it flips upside down there are some random pixels and they stay there untill it flips the second time upside down
I also tried u8g2 flipmode example and that works perfect

[image]

olikraus, 2017-08-15:
Hmmm... Ok... Maybe some other issue, but I think your 240x120 display is now done :)
Thanks for testing.

floris-b, 2017-08-15:
Thanks for the quick response and the great library

closed 2017-08-15 by floris-b