olikraus/u8g2 #334: Minimum Prgogram memory

blotfi, 2017-08-17:
Hi

I have an arduino nano, with some big code
When I compile with 

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA);   // (rotation, [reset [, clock, data]])

I am 105% of memory
things works fine with 8x8, but I need to display a scrolling big msg on a small oled, so I need u8g2 graphic fonts
How can I reduce the lib/font footprint to reduce

> Sketch uses 31906 bytes (103%) of program storage space. Maximum is 30720 bytes.
> Global variables use 2164 bytes (105%) of dynamic memory, leaving -116 bytes for local variables. Maximum is 2048 bytes.

with option 1 instead of F, I can get into RAM but not in Flash size

> Sketch uses 31904 bytes (103%) of program storage space. Maximum is 30720 bytes.
> Global variables use 1268 bytes (61%) of dynamic memory, leaving 780 bytes for local variables. Maximum is 2048 bytes.

olikraus, 2017-08-17:
Maybe you can use a different font? Will it end in "r" (like reduced)?

olikraus, 2017-08-27:
no further feedback/question, so closing...

closed 2017-08-27 by olikraus

bogus105, 2018-07-02:
same issue here. 102%. Is there an option to reduce RAM usage? i'm using "u8g2_font_ncenB08_tr". Any idea how to free ca. 100 bytes of RAM?:/

olikraus, 2018-07-03:
You mean ROM?

bogus105, 2018-07-03:
Hi Oli, i mean i have run out of dynamic memory. I found F() macro. I've read it's good to use it with Serial.print calll like: 
Serial.print("blablabla"); 
it will when be: 
Serial.print(F("blablabla")); - it is said it will save SRAM. Here's the description:

https://learn.adafruit.com/memories-of-an-arduino/optimizing-sram

Will it work u8g2 as well?

olikraus, 2018-07-03:
> Will it work u8g2 as well?

Yes, U8g2 will use exactly the same print function:

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/PrintHelloWorld/PrintHelloWorld.ino#L232

or here:

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/PrintProgmem/PrintProgmem.ino#L246

bogus105, 2018-07-03:
wow, this is getting too complicated for me:(
still can't get the idea of 'firstPage/nextPage' construction... I have many places in my code where i want to put something on OLED screen. Simply do not know hot to make it and use 'page' approach. 
I tested F() now and it seems not to free any dynamic memory:/ So far i've put F() to all Serial.print calls. Haven't testes it with u8g2.putStr() yet though.

olikraus, 2018-07-03:
Using the picture loop approach will greatly reduce dynamic RAM. In fact it got introduce only to save RAM. So, if you want to save a good amount of RAM, try to use the picture loop (first/next page) approach. I have tried to explain this here: https://github.com/olikraus/u8glib/wiki/tpictureloop

It is NOT complicated.

bogus105, 2018-07-03:
Thank you Oli! 
so far F() did the trick but i've also tried your way to use F() trick wirh u8g2. However it is more complicated as i can't use:
u8g2.drawStr(x,y,"string");
instead i have to use set cursor method first and then print method:
u8g2.setCursor(x,y);
u8g2.print(F("string"));

now i can learn your explaination how to use page approach.

olikraus, 2018-07-03:
Yes, F() only works with print

bogus105, 2018-07-03:
one more thing: is there a way to save SRAM here using F() trick?:

u8g2.print((Encoder.getAccelerationEnabled()) ? "on" : "off");

olikraus, 2018-07-03:
Not sure, but maybe this works:
u8g2.print((Encoder.getAccelerationEnabled()) ? F("on") : F("off"));

bogus105, 2018-07-04:
update:
u8g2.print((Encoder.getAccelerationEnabled()) ? F("on") : F("off"));
checked, works fine!

Oli could you tell me why i get the following error during compilation?

class U8G2_SSD1306_128x32_UNIVISION_F_HW_I2C has no member named 'getMaxCharHeight'

i tried to use getMaxCharHeight to calculate cursor position to print string in the middle of the display.

olikraus, 2018-07-04:
It should be here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L301

I do not remember, when it got introduced, but it should be included in the latest release. If not, try the latest beta:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

bogus105, 2018-07-07:
Thanks Oli!