olikraus/u8g2 #336: Running on Teensy 3.6

Dealerpriest, 2017-08-17:
Hello. I'm trying to run a SSD1306 128x64 OLED on a Teensy3.6. The display only has I2C. It's one of those cheap ones from china. It doesn't have a reset pin. I've successfully interfaced the display with the adafruit libraries (Adafruit_GFX.h & Adafruit_SSD1306h) without any issues.
Wanted to try u8g2 because it seems to better align with my needs. 
Using the following constructor:
```
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, 19, 18);
```
and the following code:
```
// in setup
u8g2.begin();
// in loop
u8g2.setFont(u8x8_font_chroma48medium8_r);
u8g2.drawStr(0,20,"Hello World!");
```

Nothing happens. What could be the reason?

olikraus, 2017-08-18:
First point: Do not use u8x8-fonts with u8g2. It will not work, because the font structure is different.
u8x8_font_xxx ---> u8x8.setFont
u8g2_font_xxx ---> u8g2.setFont

Second point:

With the page buffer constructor

```
U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, 19, 18);
```
use:
```
  u8g2.firstPage();
  do {
     u8g2.setFont(u8g2_font_ncenB08_tr);
     u8g2.drawStr(0,20,"Hello World!");
  } while ( u8g2.nextPage() );
```

With the full size buffer 
```
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, 19, 18);
```
(note: _F_ instead of _1_), use:

```
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();		
```

Dealerpriest, 2017-08-21:
Alright. It works. Thx!

closed 2017-08-21 by Dealerpriest

olikraus, 2017-08-21:
:-)