olikraus/u8g2 #337: Unifont results are strange with Arduino IDE beyond version 1.6.6

edp1096, 2017-08-18:
Hi.

I'm testing 'Korean' in u8g2 examples with ssd1306 oled display and arduino uno r3.

I got a result well as expected with Arduino IDE 1.6.5-r5 but, Unicode characters (="안녕 세상") were disappeared and eat some ASCII characters with from Arduino IDE beyond 1.6.6 to latest 1.8.3.

How can I solve this problem?

olikraus, 2017-08-19:
Thanks for testing "Korean" fonts in u8g2. However I did not understand your problem. Is it a problem with the Arduino IDE? In such a case I can probably not help, because i am not related to Arduino IDE in any way.

edp1096, 2017-08-19:
I think you're mention is right.

I met problems not only korean but also latin, chinesse, greek.
I guess that those problems were caused from some compile settings in Arduino IDE's buillt in so I am try diffing and digging  both 1.6.5 and 1.6.6 of Arduinon IDEs.

But because I'm not well to computing knowledge, I don't know what is the best approach.
May I ask you how to do for solving?

Thank you for your kind answer!

olikraus, 2017-08-19:
Maybe ask in the Arduino Forum, for example here:
https://forum.arduino.cc/index.php?board=23.0

edp1096, 2017-08-20:
Ok, I will try it.

closed 2017-08-20 by edp1096

edp1096, 2017-08-20:
My question is related to [this thread](https://forum.arduino.cc/index.php?topic=409966.90).

Thank you again.