olikraus/u8g2 #339: EW50850FLWP - UC1611 driver -  8080 Mode (240W x 160H)

arnoldvillasanta, 2017-08-20:
Request to have this included in the supported LCD.  [Specs of LCD is here](http://www.datasheet39.com/PDF/921659/EW50850FLWP-datasheet.html)
I tried the following but I can't seem to make it work:

- //U8GLIB_UC1611_DOGM240 u8g(8, 9, 10, 11, 4, 5, 6, 7, 18, 3, 17, 16);   // 8Bit Com: D0..D7: 8,9,10,11,4,5,6,7 en=18, cs=3, di/a0=17,rw=16

- //U8GLIB_UC1611_DOGXL240 u8g(8, 9, 10, 11, 4, 5, 6, 7, 18, 3, 17, 16);   // 8Bit Com: D0..D7: 8,9,10,11,4,5,6,7 en=18, cs=3, di/a0=17,rw=16

olikraus, 2017-08-20:
I am confused. This is the U8g2 issue tracker. However you mention two old u8glib constructors. Did you test one of the U8g2 constructors?

arnoldvillasanta, 2017-08-21:
Sorry for the confusion.
I installed both new and old libraries.   The new library only has I2C construct for the UC1611 (no 8080) thus I used the old one.
I would rather prefer the new one but the LCD I mentioned doesn't have I2C.   Would you help me identify which constructor is good to test with?
I also wants to confirm the following pin connections if using 8080 mode:
en = 18 (WR0 of the LCD)
cs = 3 (CS1 of the LCD)
di/a0 = 17 (CD of the LCD)
rw = 16 (WR1 of the LCD)    //I'm not sure about this
BM0 of LCD is in GND to set to 8080 mode
RST of LCD is not connected

olikraus, 2017-08-21:
U8G2 also includes the parallel interface. They are just not mentioned in the examples. You need to consult the wiki pages.

Any uc1611 constructor will be fine as a starting point for a new device.

olikraus, 2017-08-21:
You are right, it is not included :-(
I will add the parallel interfaces.

olikraus, 2017-08-21:
The latest beta release now contains the missing constructors. You could for example use this constructor for further testing
U8G2_UC1611_EA_DOGM240_1_8080(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset])

the latest beta release is available here for download:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Add this zip library to Arduino IDE via the "add zip library" menu from within the Arduino IDE.

arnoldvillasanta, 2017-08-21:
This is great!!!  Thanks for all the efforts.  I will post here the result of my test.

arnoldvillasanta, 2017-08-21:
I would like to confirm about these 3 pins:
1.   enable
2.   cs
3.   dc

The LCD has this following pins:
1.   BMO = must be LOW for 8080 mode
2.   WR0 = write strobe (when in 8080 mode)
3.   WR1 = read strobe (when in 8080 mode)
4.   CD = register select input
5.   CS1 = chip enable
6.   RST = reset 

I tried the following connection but the LCD is blank (pageBuffer/HelloWorld example):
1.   enable  =  CS1
2.   cs = WR0
3.   dc = CD

and also this:
1.   enable  =  WR0
2.   cs = CS1
3.   dc = CD

Now I wonder if I had to declare pinMode of all/any of the pins or are they handled in the library? (have not read through the lib source code yet)
LCD [tech doc](http://www.datasheet39.com/PDF/921659/EW50850FLWP-datasheet.html).

arnoldvillasanta, 2017-08-21:
I added Serial write and it was not reached.  I think the execution stopped in the construct.
`U8G2_UC1611_EA_DOGM240_1_8080 u8g2(1, 8, 9, 10, 11, 4, 5, 6, 7, 18, 3, 17, 16);  //(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset])

// End of constructor list

void setup(void) {
  Serial.begin(115200);
  delay(100);
  Serial.println("Serial initialized.");
  //u8g2.begin();
}`

olikraus, 2017-08-21:
It should be like this (LCD vs. u8g2 argument)
WR0 = enable
WR1 = VCC (logic high) !!!
CS1 = cs
CD = dc

It is important that you connect the RD read strobe input of the LCD to high. I do not remmeber, whether there was a reset input of the display. If so, you need to connect the LCD reset also to a GPIO pin and name it on the constructor line.

> U8G2_UC1611_EA_DOGM240_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, 18, 3, 17, 16); //(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset])

So, to which GPIOs did you connect enable, cs and dc?

arnoldvillasanta, 2017-08-21:
There is a reset pin in LCD display.
I followed your instruction regarding WR0, WR1, CS1 and CD.
The GPIOs - enable:18, cs:3, dc:17, reset:16   in Arduino UNO.
There's no display yet on LCD after doing this.  I'll continue tracing my wiring to make sure.

olikraus, 2017-08-21:
> The GPIOs - enable:18, cs:3, dc:17, reset:16 in Arduino UNO.

ah yes, exactly. Somehow I didn't see that before.

Also if you use some of these prototyping boards... The contacts are not that reliable, so measuring the resistance between LCD pin to Arduino pin while power off might be also usefull.
Since you mention Uno, there is one more thing: Uno is a 5V device, while the UC1611 is a 3.3V device (if i remember correctly). I hope you have used a level converter for this.

arnoldvillasanta, 2017-08-21:
I used another protoboard with an Arduino Leonardo board but still no text/graphics in the LED no matter what sample sketch I use.

When I tried to do a Serial print to see where the problem is, I notice that when a delay is applied after the u8g2.begin(), there is a vertical bar that occupies 1/3 of the screen flushed at the left side.  Removing the delay will not show this vertical bar anymore.

void setup(void) {
  u8g2.begin();
  delay(100);
}

olikraus, 2017-08-21:
You added level shifters to your setup? Leonardo is also a 5v board?

arnoldvillasanta, 2017-08-22:
Oh, you're right...  I only have 1 8 channel level shifter... need to get 1 more to continue my test.

BTW, my end objective after making your library work on this LCD, is to **_'read'_** the whole display, parse a certain region, compare it to a pre-determined state, then send boolean to a web server.

My next question would be, is this possible in 8080 mode (since read strobe is hard-wired to logic high level)?  Is there a ready read function in the library to extract the whole display data (either formatted in page/column or not.)

olikraus, 2017-08-22:
The purpose of U8g2 is to render data on a display.. therefore only write is required. Read is not possible, however you could send the generated pixel bitmap to a different target.

olikraus, 2017-08-22:
Of course you can implement your own read function.

arnoldvillasanta, 2017-09-14:
Thanks for the response.  It will take time for me though to roll-out a read function since I'm clueless on the basics.  
I aim to execute a 'raw RAM read' to get the the 240 x 160 pixels and save it into a 2D array.   Then once transmitted to the web server, a backend process will convert it to a bitmap file (save and/or display to a web app).
Any lead/help or sample code will be highly appreciated.

olikraus, 2017-09-14:
> Any lead/help or sample code will be highly appreciated.

Maybe this is usefull: We have a new bitmap object, which finally allows rendering to .tga files:

https://github.com/olikraus/u8g2/tree/master/sys/bitmap

arnoldvillasanta, 2017-09-14:
Wow... a Screenshot function!!!

If I understand it right, the Hello World Sample (https://github.com/olikraus/u8g2/blob/master/sys/bitmap/hello_world/main.c)  will simply take a screenshot of the LCD?  And the parameters needed is just the dimension of the LCD:

>   u8g2_SetupBitmap(&u8g2, &u8g2_cb_r0, 240, 160);

I assume that this will work in 8080 mode, given that I do the connections correctly in a 3.3V GPIO board?

olikraus, 2017-09-14:
> I assume that this will work in 8080 mode, given that I do the connections correctly in a 3.3V GPIO board?

Well, it is more like an independent virtual display. But if you render the same content to your display AND the bitmap object, then you have a kind of a screencopy.

arnoldvillasanta, 2017-09-14:
Yeah, I re-read the code and I mis-understood the 'screencopy'.   So it's still a bitmap write to a buffer for saving the file in TGA format.  
I still need to work on getting the data from the display RAM and writing that to the buffer.

arnoldvillasanta, 2017-09-15:
Going back to the u8g2 library, I still have blank screens on all sample files.  I'm using Hologram Dash board ( https://hologram.io/store/) which has 3.3V GPIOs. I made sure all pins I'm using can drive HIGH at 3.3V as well.
I used OScope to see if there are write commands that are executed based on the UC1611 command table... but the fist thing I notice is that some wave form don't have 'holding period' at either positive or negative side, and there are some transitions that coincides/simultaneous with each other (enable vs data line D3) as shown below:
(blue line is enable (WR0) and red line is D3)
[image]

olikraus, 2017-09-15:
So you have a M0+ board?

The u8g2 procedure should be this:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_byte.c#L197

The nano delay ends up here:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L101

Maybe this code is wrong... in fact it is only an approximation, but still some delay should be visible.
Is it confirmed, that the Arduino specific function (like the delay functions) are correct?

arnoldvillasanta, 2017-09-15:
Yes, it has the M0+ board (not sure which + it is).   During code compilation, the CPU speed can be configured to run at either 48/100/120Mhz.

Delay() works fine but micros() differs from arduino in terms of 'resolution'.  I posted a question to the Hologram community to verify this but they have not get back to me yet. 
https://community.hologram.io/t/understanding-micros-in-dash/965

Thanks for putting me in the right direction (pointing me to the source code).  I'll have a look at the nano delay... maybe the timing issue is there.

[image]

arnoldvillasanta, 2017-09-20:
So far, I changed the nano delay from 0?0:1 to 0?0:1000: 
    ```
case U8X8_MSG_DELAY_NANO:
    delayMicroseconds(arg_int==0?0:1000);
    break;
```
and I'm seeing in the OScope that there's a 1 ms pulse per data already (no more cone-shaped wave).
However, there's no display yet on the LCD.

I noticed in the UC1611 AC Characteritics diagram that each WR0 RISING pulse coincides on the center of the DATA pulse (left side is the data setup time and the right side is the data hold time.).  Another thing I noticed is that there's a 1 WR0 : 1 Data pulse in the diagram while in the OScope, there's 1 WR0 : many Data ration.

![screen shot 2017-09-20 at 2 06 32 am](https://user-images.githubusercontent.com/31164013/30660477-792ea08a-9e73-11e7-832f-9cc22ffd5cc0.png)
![screen shot 2017-09-20 at 1 56 23 am](https://user-images.githubusercontent.com/31164013/30660476-79294284-9e73-11e7-84db-882d7cecb32f.png)

arnoldvillasanta, 2017-09-20:
In addition to my previous comment,  I noticed that the DATA PULSE (red) is changed/set exactly 1 clock cycle ahead of the ENABLE/WR0 PULSE (blue).   
But the UC1611 doc says it should be the ENABLE ahead by a small fraction to the DATA pulse so that it's RISING pulse is somewhere in the middle of the DATA pulse.

olikraus, 2017-09-20:
I bascially had tested the 8080 interface. Maybe there is some wiring problem?

arnoldvillasanta, 2017-09-21:
Which arduino board have you used?  I will try using that one to validate if M0 is the one who'se giving me a hard time.

I also noticed in: 

> uint8_t u8x8_byte_8bit_8080mode(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)

inside

> U8X8_MSG_BYTE_SEND

the call to write the DATA (0:7) is before the call to the Enable (WR0)'s data_setup_time_ns and write_pulse_width_ns

In the UC1611 document, I interpret that it should be data_setup_time_ns  > delay > data write >  write_pulse_width_ns ...  where data write and write_pulse_width_ns should be done asynchronously.

arnoldvillasanta, 2017-09-21:
BTW, I also placed some Serial.println() inside U8X8lib.cpp and verified that the fall back u8x8_byte_8bit_8080mode function was called (not the u8x8_byte_arduino_8bit_8080mode).

> 
> #else
>   /* fallback */
> extern "C" uint8_t u8x8_byte_arduino_8bit_8080mode(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
> {
>   Serial.println("Inside EXTERN C FALL BACK");
>   return u8x8_byte_8bit_8080mode(u8x8, msg,arg_int, arg_ptr);
> }

When I refactored the fallback code as below, I'm seeing that the waveform of ENABLE and DATA[0:7] are much closer to the UC1611 waveform, except that some DATA pulse is getting 'cone-shape' again.  
I don't know if I'm going further away to the solution or is getting closer to resolve the issue.

> case U8X8_MSG_BYTE_SEND:
> 	u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->data_setup_time_ns);
> 	u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_E, 0);
> 	delay(3);  //ARNOLD:  Added to delay data write
>       data = (uint8_t *)arg_ptr;
>       while( arg_int > 0 )
>       {
> 	b = *data;
> 	data++;
> 	arg_int--;
> 	for( i = U8X8_MSG_GPIO_D0; i <= U8X8_MSG_GPIO_D7; i++ )
> 	{
> 	  u8x8_gpio_call(u8x8, i, b&1);
> 	  b >>= 1;
> 	}    
> 	
>       }
> 	u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->write_pulse_width_ns);
> 	u8x8_gpio_call(u8x8, U8X8_MSG_GPIO_E, 1);
>       break;
>       

[image]

olikraus, 2017-09-21:
My own tests for the 8080 mode happend with an Arduino Uno and a SSD1306 OLED.

arnoldvillasanta, 2017-09-22:
I'll try Arduino Uno and UC1611 and will let you know about the test result.

On another note, the Chip Select of SSD1306 is active LOW while that of the uc1611 is active HIGH.   The below graph (BLACK LINE) is the CHIP SELECT which is low when data is written wich might be the reason why LCD is blank.   Where can I change to CS setting to make it HIGH so I can test it on my setup.

[image]

[image]

[image]

[image]

olikraus, 2017-09-22:
hmm... so you say, there is a active high chip select (CS1) and an active low chip select (CS0). Which of the two has to be controlled? I mean which of the two CS lines is available for wiring? 

If this polarity is an issue, it could be changed easily:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c#L157

arnoldvillasanta, 2017-09-22:
CS1 (activeHIGH) chip is available for wiring.  I'll check the code.   Thanks!!!   

[image]

arnoldvillasanta, 2017-09-22:
After the CS polarity was changed, the LCD shows up at the lower part a solid square on the left side and a gray rectangle adjacent to its right side (HelloWorld_U8G2.ino).   Do you have any clue moving forward?   (No other changes on code except the Nano delay delayMicroseconds(arg_int==0?0:3000);
For the meantime, I'll try Arduino Uno.

olikraus, 2017-09-22:
Initially you once said, you use U8G2_UC1611_EA_DOGM240_1_8080 (240x64)
However to me it looks the display is much higher but maybe smaller. What is the dimension of your display? To me it looks like that u8g2 erased the upper part but just skipped the lower part (because the constructor is for a display of 64 pixel height)

All in all I guess the parameters of your displays needs to be updated like offsets, dimensions, bias, etc...

I guess i need to create a new device in u8g2 for your display.

arnoldvillasanta, 2017-09-22:
Display of LCD is 240 x 160.   I saw the last parameter as the height so I adjusted it as well before I compiled the code.  You're maybe right when you said adjusting other LCD parameters as well...  (LCD:  http://www.datasheet39.com/download.php?id=921659)

arnoldvillasanta, 2017-09-22:
Also, where can I find the WRITE COMMANDS (control/data) codes of UC1611?    Does it make any sense for me to look at it right now?

olikraus, 2017-09-22:
Write commands are defined in the UC1611 datasheet. They are required for the init code (setup) and the writing of the actual data. At the moment there is no 240x160 device (one more reason to add one for you). The DOGXL240 is a 240x128 device. The init code for this is here:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c#L229

Other parameter are stored here (like cs polarity and size):
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c#L254

Writing to the display happens here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c#L79

But bitbanging and low level access (as discussed) is in a different procedure.

arnoldvillasanta, 2017-09-22:
Thanks for the links and the time to add 240x160 on your device list.    Bit-banging (Read RAM memory) will be next after I had the write commands working.  If you'll over-extend the help to RAM read then I'll try to have more than a beer to contribute to your efforts once I had it going.

olikraus, 2017-09-22:
RAM read unfortunately is not an option. U8g2 is a write only library and there is no infrastructure at all to read from a display.

olikraus, 2017-09-23:
I have created a new set of constructors:
U8G2_UC1611_EW50850_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,  /*enable=*/ 18, /*cs=*/ 3, /*dc=*/ 16, /*reset=*/ 16);

Download the beta release here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Use the "Add zip library" menu option to install this beta release in the Arduino IDE.

It includes modifications to support active high cs and the correct size (240x160).

Will this work better? Anything visible? 
It might help to execute the page-buffer / ContrastTest example to figure out the correct contrast value.

arnoldvillasanta, 2017-09-23:
I made a graphics representation of what's being displayed in the LCD...   I think we're very close.

[image]

With regards to the ContrastTest sample, the texts appears the same as with the Hello World and a good contrast ratio is 070 to 080

[image]

olikraus, 2017-09-23:
ok, great. But i guess i need more help from you.
Can you locate u8x8_d_uc1611.c on your computer?

The contrast can be modified here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c#L323

For the text mirror adjustment, i think we need to change the 0xc0 command:

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c#L329

Current value is this:
  U8X8_CA(0x0c0, 0x004),            	/* LCD Mapping */
you could try:
  U8X8_CA(0x0c0, 0x000),            	/* LCD Mapping */
or this:
  U8X8_CA(0x0c0, 0x006),            	/* LCD Mapping */

Which command will make "Hello World" appear correctly?

The other question is, why is there a large blank area on the display?

arnoldvillasanta, 2017-09-23:
Contrast works.
Text mirror did not work on 0x000 and 0x006.
Display is 1/3 enabled and the 2/3 is dead.  Is it related to display enable?
I saw this line of code is commented out.
I tried to do   U8X8_C(0x0af) but still the 2/3 is dead.
[image]

arnoldvillasanta, 2017-09-23:
Regarding LCD Mapping...  in the UC1611 document, it is a single byte command (0x0c_).
I wonder why in the code it's U8X8_CA(0x0c0, 0x004).

olikraus, 2017-09-23:
Maybe the leading 0 is confusing. It is the "(21) S ET LCD M APPING C ONTROL" command 0xc0.

olikraus, 2017-09-23:
Looks like i have a different datasheet: http://www.lcd-module.de/fileadmin/eng/pdf/zubehoer/uc1611s_v1_0.pdf

olikraus, 2017-09-23:
I have created a new beta here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Will the text be correct now?

olikraus, 2017-09-23:
ok, I added the 0xaf command. this should enable the missing display part.
I created another beta (2.18.8): https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Will this work?

arnoldvillasanta, 2017-09-24:
Full-screen display is now working.
Image is still mirrored in the x axis.
Can we try single byte code command for the LCD Mapping control as indicated in the [UC1611 datasheet](https://www.crystalfontz.com/controllers/UltraChip/UC1611/363/).

[image]

[image]

olikraus, 2017-09-24:
thanks for the datasheet link.
ahh, now I understand your comment on the LCD setting command. Yes, I will add the single byte command.

olikraus, 2017-09-24:
Beta release 2.18.9: https://github.com/olikraus/U8g2_Arduino/archive/master.zip

I fixed the LCD mapping command...

olikraus, 2017-09-24:
I did a review of the UC1611 datasheet. I am afraid, that the display memory structure is also different. This means, that even more effort is required to finish the driver for UC1611 :-(

But let us wait for your feedback on the latest u8g2 beta release.

arnoldvillasanta, 2017-09-24:
IT WORKS!!!  Hello World Indeed.
Let me know how I can help finish the driver for UC1611 (as tester maybe?)

[image]

A quick note:
I ran page_buffer/U82gLogo and the logo was mirror-displayed.  I comment-out flipMode(0) so that the 0x0c4 LCD Mapping Control value will not be over-ridden.
I just wonder why LMC is set to 0x0c4 rather than 0x0c0 (no rotation) so that flipMode(0) is for vertical flip and flipMode(1) is for horizontal flip.  I just conclude that the reason is deeply coded inside the core library.

[image]

olikraus, 2017-09-24:
> I just wonder why LMC is set to 0x0c4 rather than 0x0c0 (no rotation)

My mistake...

olikraus, 2017-09-24:
> Let me know how I can help finish the driver for UC1611 (as tester maybe?)

I think you already do this. :)
But the architecture change requires some more work :-p

olikraus, 2017-09-24:
Beta release 2.18.10: https://github.com/olikraus/U8g2_Arduino/archive/master.zip

- Draft version of the controller RAM architecture
- Once more, i fixed the LCM command (see above)

What will you see?

arnoldvillasanta, 2017-09-25:
The LCD displayed horizontal lines only for the HelloWorld.
I ran the page_buffer/GraphicsTest and I also got horizontal lines animating starting from the buttom row (so I suspect display is mirrored again).
I also ran the page_buffer/FlipMode and horizontal lines started to display from the bottom row up until the whole LCD display becomes 'black'.

This is one of the screens from GraphicsTest:
[image]

arnoldvillasanta, 2017-09-25:
This is the display for the HelloWorld:

[image]

olikraus, 2017-09-25:
ah, ok... thanks.
I fixed this file:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_uc1611.c
For further testing you could download this file directly and replace the same file in your local arduino library folder or you could wait until i am back to my lab for the creation of another beta (which will be in 6h or so).

[54 more replies]

closed 2017-09-28 by arnoldvillasanta