olikraus/u8g2 #340: Improve render-to-file (e.g. tga) support

matthijskooijman, 2017-08-23:
I was previously using u8glib with its pbm backend to generate screenshots from my application for a manual. With u8g2, the pbm backend was removed, but I did find a tga backend that could fulfill the same usecase.

However, the current tga backend in `sys/tga/common` seems to be very much tailored to the images generated for the u8g2 manual, where the 128x64 LCD screen is rendered at a hardcoded x3 size in the middle, with some room around it for descriptions. This is implemented using two separate u8g2 devices that both write to the same memory. It took me a while to figure out how these things fit together, and now I know I've found that this makes the current tga code unsuitable for what I need (create 128x64 and 128x(more than 64) screenshots without any annotation). I see a few ways to improve this, and I'm curious to your opinion:
 - I added a third device to `sys/tga/common` which implements just a plain tga renderer, without any scaling or other things I do not need. This seems to work, but introduces some duplicate code.
 - I just found that `tools/font/build` also has a tga rendering backend, very similar to the one I already added (except that it does not allow changing the size, but that should be easy to add).
 - I can imagine a cleaner way to unify these three tga backends would be to introduce a "dummy" device, which only renders to memory (essentially just making `DRAW_TILE` a noop). This memory can then be accessed directly by some TGA helper functions, which export the memory contents to a TGA file. To also support the u8g2-manual usecase, this TGA export would need simple compositing support, e.g. allow assembling the exported TGA from two dummy device memories: an LCD part and a (transparent) description part. This compositing needs to happen outside of the normal drawing routines (so we cannot draw the lcd in a smaller memory device and then draw the resulting LCD memory onto a bigger description device), since the resulting image should contain more than 2 colors.
  Writing this, it seems that this compositing stuff might become complicated quickly, so it might be better to use an existing library for doing that (or probably even for doing all the description drawing). A quick search comes up with [CImg](http://cimg.eu/), a single-file C++ library that can handle drawing and writing out images in various formats.

Does any of these approaches seem reasonable to you? For our own manual, I will probably explore the third option, since that would probably also give me options for highlighting parts of my screenshots.

olikraus, 2017-08-23:
I never though that someone else could be interessted in these tga export devices.
What is your purpose/application/product here?

It would not be too complicated to create a custom tga device for you only, if you specify your needs (variable size, bg/fg color selectable, etc)
On the other side I probably can not use another graphics lib. Especially CImg seems to be too complicated: My toolchain is c only, so it has to be moved to C++. Due to the support of png and jpg i may end up adding even more libs, which will be a huge effort.

matthijskooijman, 2017-08-23:
> I never though that someone else could be interessted in these tga export devices. What is your purpose/application/product here?

We use u8g2 in a product to render the menu interface of our machine, and we include screenshots of all available screens in the manual (by compiling our firmware on a regular PC, using some Arduino stubs, and generating the screens using the same code as in the actual machine). I need a fixed size screenshot (128x64), but also a variable size screenshot (128xsome length, to capture a full screen that is normally accessed by scrolling).

> It would not be too complicated to create a custom tga device for you only,
I already did that, which works, but duplicates quite some code from the exsting LCD and DESC tga devices.

> On the other side I probably can not use another graphics lib. Especially CImg seems to be too complicated: My toolchain is c only, so it has to be moved to C++.

It's a pity that it is C++, but it is certainly not really complicated (in particular it is only a single header file, which makes it easy to include). If you do what I suggested (let u8g2 render to memory and then let the application retrieve the rendered data and stuff it into CImg), only the application (so the manual-generating-application) needs to be converted, no need to convert the entire u8g2 library to C++.

A bonus feature for me would be to add highlighting in different colors to the generated images (e.g. to mark areas of the screenshot to refer to them from the text), which seems easiest using something like CImg. So i'll probably code up something along these lines and share it, then you can see if it would make sense for the u8g2 library and/or make sense to include it as an example in the library.

olikraus, 2017-08-23:
I could introduce a generic interface. I render the content to memory and finally do a call to a callback function (which in turn might call CImg) with the memory and the dimension as arguments.

So finally you can add whatever processing you want in that callback function.
Would this solve you problem?

matthijskooijman, 2017-08-23:
> I could introduce a generic interface. I render the content to memory and finally do a call to a callback function (which in turn might call CImg) with the memory and the dimension as arguments.

That might work, though it might be easier to pull the data out of u8g2 instead of getting it pushed through a callback. That would allow something like:

```
drawSomething(&u8g2);
CImg->drawBitmap(getMemory(&u8g2), getWidth(&u8g2), getHeight(&u8g2));
CImg->drawText("Some description", 0, 0)
```

(where `getMemory()` would be a new API, which might very well look completely different, `getWidth()` and `getHeight()` probably already exist by another name (but I was lazy) and the Cimg->drawXXX API is just hypothethical, but I hope you get the idea.

I actually suspect that one can already access the U8g2 memory by peeking into the datastructure, so essentially this would only need an U8x8 device that has a configurable (and ideally changeable) size, with a noop `DRAW_TILE`. A challenge is memory allocation (due to the configurable size), but I can imagine the memory would just be externally allocated by the application and passed in, or if nothing is passed in, it would be malloc'd.

olikraus, 2017-08-23:
By "passing the memory" I actually meant a pointer to that memory. And indeed I do not care whether this can be fetched by another getMemory(). It could be done like this. However, to be more generic, my plan is to build on top of DRAW_TILE which is so to say the main and most important abstraction layer between the display and the u8x8 and u8g2 world. 

I probably will introduce a new object/class, lets say "bitmap"
"Bitmap" objects will be allocated with size, defined at runtime.
DRAW_TILE will write to "bitmap" and of course there will be a getMemory(bitmap) function.
You would be able to create 
drawBitmapOnCImg(cimg, x, y, bitmap) 
function, which places the "bitmap" at a desired position on your target bitmap.

Reason for doing so, is because the complete bitmap may not exist at all in u8g2/u8x8 world.
So we need a new object which will represent the full size bitmap.

so my idea is to create a new device "u8x8_d_bitmap", which will take any DRAW_TILE messages and render these messages into an file local "bitmap" object.

On the other side there will be callback functions (maybe better say hook functions) which will be called, if a screen has been full rendered by u8x8 or u8g2.

olikraus, 2017-08-25:
Get a pixel from the internal bitmap:
``` 
uint8_t u8x8_GetBitmapPixel(uint16_t x, uint16_t y)
``` 

Save the internal bitmap to TGA file:
``` 
void u8x8_SaveBitmapTGA(const char *filename)
``` 

Init a u8x8 struct, which is then connected to a bitmap. This can be used as any other u8x8 object.
``` 
void u8x8_SetupBitmap(u8x8_t *u8x8, uint16_t tile_width, uint16_t tile_height)
``` 

Init a u8g2 struct, which is then connected to a bitmap. This can be used as any other u8g2 object.
``` 
void u8g2_SetupBitmap(u8g2_t *u8g2, const u8g2_cb_t *u8g2_cb, uint16_t tile_width, uint16_t tile_height)
``` 

... nothing tested as of now

matthijskooijman, 2017-08-25:
> Reason for doing so, is because the complete bitmap may not exist at all in u8g2/u8x8 world.

Ah, that makes sense (though I think you can configure `u8g2` in a way so the entire bitmap *does* exist (e.g. by not using any paging), I believe? In fact, your current code allocates the memory twice, so there is at least memory for the entire bitmap at the `u8g2` level. Or is there some other reason why it would not have the entire bitmap?

Anyway, thanks for creating the code for this, it looks good (haven't been able to test it yet). However, two remarks:
 - Currently, the size must be a multiple of 8x8 tiles. Would you think it would be hard to allow arbitrary sizes? (not really essential for me, but would be nice).
 - There are now some global variables used for this bitmap device, meaning you can only instantiate one. It would be elegant if you could actually instantiate multiple devices in parallel. One problem would be the `u8x8_bitmap` variable. Wouldn't it be possible to allocate that variable as part of the malloc, and store a pointer to it inside the `u8x8_t` struct? I'm not sure if there is already a generic, device-specific pointer in `u8x8_t` that could be used for this, though? Another problem could be `u8x8_bitmap_info`, which contains the device size. Is this size actually used anywhere? I suspect so, so solving this could mean to also allocate a copy of this struct in the malloc, and update the size values there? Again, this isn't essential for me, it would just seem to increase the flexibility of this feature.

Your mention of an SDL surface makes me think of a completely different option to approach this: The existing SDL device could perhaps the modified to allow creating a memory-only SDL_Surface to render onto, which can then be saved to a bitmap file by SDL, or further processed (copied onto yet anohter SDL_Surface to add annotation etc.). I'm not exactly sure how powerful SDL is in that regard, though.

olikraus, 2017-08-25:
> In fact, your current code allocates the memory twice

Yes, it is not so obvious, but i need the display buffer and the buffer for full page u8g2 mode. I put both memories into one memory block... little bit tricky... Of course finally it is not required, i mean u8g2 could directly write to the buffer, but this approach allows also some testing of the u8g2/u8x8 code.

> There are now some global variables

Yes, I know... It is still more a hack, than a full flexible feature. Will it be ok for you to have one instance only for now?

> which can then be saved to a bitmap file by SDL

Of course, but the new u8x8 bitmap code would be more flexible. It will work with any u8g2/u8x8 device in any mode.

matthijskooijman, 2017-08-25:
> Yes, I know... It is still more a hack, than a full flexible feature. Will it be ok for you to have one instance only for now?

Yeah, it should be fine!

I wonder if it would make sense to add a (for now unused) `u8x8*` parameter to `u8x8_GetBitmapPixel()` and `u8x8_SaveBitmapTGA()`, so if you ever allow multiple instances, these functions do not need to change in an incompatible manner?

> Of course, but the new u8x8 bitmap code would be more flexible. It will work with any u8g2/u8x8 device in any mode.

Good point, the current code could actually work on an Arduino along with an actual device as well, not just on a regular PC.

olikraus, 2017-08-25:
I have updated 
https://github.com/olikraus/u8g2/tree/master/sys/sdl/text_picture_loop
and
https://github.com/olikraus/u8g2/tree/master/sys/sdl/text_full_buffer
with a screenshot option: Pressing 't' will create a file "screenshot.tga".

I agree that this is not the best code, but is this what you need?
Probably you also need an example with a standalone bitmap, i mean, an example where i render to the bitmap only.

olikraus, 2017-08-25:
> I wonder if it would make sense to add a (for now unused) u8x8* parameter to u8x8_GetBitmapPixel() and u8x8_SaveBitmapTGA(), 

Agree, would make sense...

matthijskooijman, 2017-08-25:
I'll have a go at converting my code to use this (probably next monday or tuesday). If you want, I can strip out some things from that code to turn it into an example (that uses the new bitmap device to draw some  things and export a TGA from that). I might also expand this to use CImg and provide a separate example using that, though I'm not sure I really need that yet.

olikraus, 2017-08-25:
sure, let me know if you find some issues. Also any improvements are welcome :-)

matthijskooijman, 2017-08-30:
Some more questions:
 - Why does this code live under `sys/sdl`? It seems completely unrelated to SDL, since the bitmap stuff can be used with any other device, or by itself as well. Having this in the sdl folder might prevent people from finding it. Perhaps a new `sys/bitmap` or even include it in `csrc`?
 - Why are there widths and heights in the `u8x8_bitmap` struct? These values are already stored in `u8x8_bitmap_info`?

I've pushed some changes to [this branch](https://github.com/matthijskooijman/u8g2/commits/bitmap), which:
 - Fixes the u8g2 cppsrc without `U8X8_USE_PINS` (which I need to compile my Arduino code on a regular PC, using some Arduino stubs, to allow generating screenshots from my original application code).
 - Slightly cleans up the buffer allocation (not sure if you like it, but it seems clearer to me at the expense of one additional pointer).
 - Allows non-multiple-of-8 pixel dimensions. It seems the rest of the library already handles this, so this is a fairly trivial change.
 - Adds a `U8G2_BITMAP` class to allow using the bitmap stuff from a C++ context. Again, I need this to make my Arduino code run on a regular PC, since my Arduino code expects a `U8G2` object to work with. This new class lives in `cppsrc/u8g2.h`, so it might be slightly weird that it relies on stuff in `sys/sdl` (OTOH, the bitmap functions are declared under `csrc`, which is similarly weird).
- Adds an example for using the bitmap stuff by itself to generate a TGA file.

My next step is to fiddle with CImg to generate annotated screenshots, but I wanted to share my progress so far already.

olikraus, 2017-08-30:
Comment via mobile Access: 
1) agree, should be sys/bitmap once
2) info structure: i wanted to have something independent. But, yes, might be that this is redundant.
3) use pins: I wonder that something has to be done because of the use pins macro
4) buffer allocation... Sure, definitely more clean with two buffers
5) bitmap class... The cppsrc is a pure Arduino folder. Most content is automatically generated.
6) example... Yes... I only have limited access to the repo but I also wanted to create examples.

All in all it sounds good, but I have no time to look into your kids at the moment.

olikraus, 2017-08-30:
Forgot one point: none x8 dimensions: won't this break screenshot s from u8x8? Will it work? Did you change DRAW TILE msg?

matthijskooijman, 2017-09-04:
I won't respond to everything you said, since I think most of the things would be answered when you have time to look at my commits.

> bitmap class... The cppsrc is a pure Arduino folder. Most content is automatically generated.

My usecase is a bit mixed though: I'm compiling Arduino code on a regular PC so I do need the C++ wrappers. However, if adding one for the bitmap device doesn't work for you, I can just define the wrapper in my own code which will also just work.

> Forgot one point: none x8 dimensions: won't this break screenshot s from u8x8? Will it work? Did you change DRAW TILE msg?

Nope, I left DRAW_TILE unchanged. I did make sure that both buffers allocated do round up to whole 8x8 blocks, so I guess this essentially just means that all rendering happens as if the device was a bit bigger, except that getDeviceWidth() returns the actual, non-rounded size and the TGA generation also uses the unrounded version and simply ignores the "extra" parts of the memory buffer.

I also experimented a bit with using CImg for directly generating PNG images, as well as adding annotations around a screenshot. I'll prepare some code examples to show how that looks in a separate comment.

matthijskooijman, 2017-09-04:
Instead of just some code snippets, I ended up reducing my own code to an example that can be used stand-alone. I've pushed it to the bitmap branch, see https://github.com/matthijskooijman/u8g2/blob/eeb4a57efc278eef10df4d764b57db2542ca5953/sys/sdl/cimg_annotate_screenshot/main.cpp for the full example. The output generated by that example is this:

[image]

Here the black figures are rendered using u8g2, the grey border and the red annotation are rendered on top of that using CImg. I believe that this same approach could work for generating the u8g2 manual images as well, removing the need for the complicated two-layer TGA device that is used now. 

As another example, here's an image we've been generating for our actual application:
[image]

olikraus, 2017-09-08:
Amazing work. Looks definitly better than the u8g2 images. 
Nevertheless, this is a spare time project. U8g2 images are all done and i do not expect much more pictures for the manual in the future, so i guess the porting effort is not worth the benefit from the better pictures.

Most people anyway do not see/honor the effort in good docs... So I probably leave manual pics as they are...

matthijskooijman, 2017-09-08:
Thanks. It's probably a significant work to port things over, so I can imagine it's not worth the effort. Perhaps on some later overhaul of the documentation. I'll have one more look over my changes and create a PR from them, including both examples, so they'll be around for future reference.

> agree, should be sys/bitmap once

Should we make it `sys/bitmap` now? I can include a commit to move things in my PR if you want? To prevent confusion between `sys/bitmap` and `sys/tga`, it might be good to include a README in the latter, stating that `sys/bitmap` is now the recommended way to generate a screenshot file, and that `sys/tga` is mostly kept around for compatibility?

matthijskooijman, 2017-09-08:
I went ahead and created the PR, including the changes (rename to sys/bitmap) I suggested above. Let's continue discussion on those changes there? If you want anything different, just let me know.

matthijskooijman, 2017-09-08:
> Most people anyway do not see/honor the effort in good docs... So I probably leave manual pics as they are...

Personally, I think one of the advantages of u8g2 is its great docs, there's a really good reference and a lot of auxilliary documentation too. So thanks for that (and the library itself as well, which I think is excellent)!

olikraus, 2017-09-09:
Thanks for the pull request. 

Also thanks for your feedback on docs and lib :-)

olikraus, 2017-09-09:
I did some final test after the merge and all works as expected :)

Can I close this issue?

olikraus, 2017-09-10:
please create a new issue  if required

closed 2017-09-10 by olikraus

matthijskooijman, 2017-09-11:
Thanks for merging and also the changes you made yourself to support this. Everything looks good for me now, thanks!

olikraus, 2017-09-11:
:)