olikraus/u8g2 #341: Requires redraw every frame

Aidenir, 2017-08-24:
I'm trying to use this library on a couple of noname oleds, one 128x64 and one 128x32, using the constructors:
`U8G2_SSD1306_128X64_NONAME_F_SW_I2C`
and 
`U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C`
together with an ESP32 as the controller.

But I'm having trouble with keeping whatever I put in the buffer from being cleared automatically every time loop runs. The code (with some abstractions and stuff):
```
void Display::render(){
  bool bufferCleared = false;
  for(int i = 0; i < 4; ++i){
      if(lines[i]->render()){
        if(!bufferCleared){
          u8g2->clearBuffer();
          Serial.println("clear buffer");
          bufferCleared = true;
          u8g2->setFont(u8g2_font_courR14_tr);
        }
        Serial.println("draw");
        u8g2->drawStr(lines[i]->currentPositionX,lines[i]->y,lines[i]->str);
      }
  }
  if(bufferCleared){
    Serial.println("send buffer");
    u8g2->sendBuffer();
  }
}
```

To make it simple this function contains the only calls to u8g2, nowhere else (except begin) is the library called. And from what I can understand this code should call clearBuffer, drawStr and sendBuffer only when lines[i]->render returns true (which is does if anything has changed). And the serial prints support this belief, however the screen breifly shows the lines, and then its cleared, even though clearBuffer is not called from any of my code.

I've tried using buffered mode as well, with the same result, as soon as the library is not called the display stops showing anything.

Is this expected behaviour or is something off? Can't find any information which suggests either scenario...

olikraus, 2017-08-25:
U8g2 should work as described by you. However do do not completly understand your code.
Assume line[3]->render() returns true, then all the previous lines get cleared. Is this expected?
Shouldnn't you have two loops: In the first loop you check whether and line[i]->render() returns true, if yes, clear buffer and carry out the second loop (drawing all lines).

Aidenir, 2017-08-25:
Hmm that is true, my mistake. However I've modified the code and am getting the same result:

```
void Display::render(){
  bool bufferCleared = false;
  for(int i = 0; i < 4; ++i){
      if(lines[i] && lines[i]->render()){
          bufferCleared = true;
        }
      }
  if(bufferCleared){
    u8g2->clearBuffer();
    Serial.println("clear buffer");
    for(int i = 0; i < 4; ++i){
      if(lines[i]){
        Serial.println("Render line " + String(i) + " str: '" + lines[i]->str + "'");
        u8g2->setFont(u8g2_font_courR14_tr);
        u8g2->drawStr(lines[i]->currentPositionX,lines[i]->y,lines[i]->str);
        Serial.println("draw");
      }
    }
    u8g2->sendBuffer();
    Serial.println("send buffer");
  }
}
```
Leads to the output:
```
clear buffer
Render line 0 str: 'Line 1!!'
draw
Render line 1 str: 'This is a longer line that is animated'
draw
send buffer
```
Previously I tried a much simpler case by just doing clear, drawStr, sendBuffer and then delay for a couple of seconds, and the same thing happened there, the text flashes once and then the display goes dark.

olikraus, 2017-08-25:
Maybe place a for(;;); after Serial.println("send buffer"); to check whether the blank screen is caused by the above code.

Aidenir, 2017-08-25:
Tried the for(;;); as well but no dice, same thing still I'm afraid.

olikraus, 2017-08-25:
sounds odd. sendBuffer sends the buffer and nothing else

https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_buffer.c#L93

olikraus, 2017-09-23:
no further feedback, closing...

closed 2017-09-23 by olikraus