olikraus/u8g2 #342: how to draw multiple strings?

abonec, 2017-08-27:
Before I used u8x8 to just draw strings and it looks like:
```
void printGPSInfo()
{
    char buffer[16];
    resetLine();
    sprintf(buffer, "%02d:%02d:%02d", gps.time.hour(), gps.time.minute(), gps.time.second());
    addLine(buffer);
    sprintf(buffer, "Lat: %s", doubleToChar(gps.location.lat(), 4, 5));
    addLine(buffer);
    sprintf(buffer, "Lon: %s", doubleToChar(gps.location.lng(), 4, 5));
    addLine(buffer);
    sprintf(buffer, "Speed: %s", doubleToChar(gps.speed.kmph(), 4, 2));
    addLine(buffer);
    sprintf(buffer, "Course: %s", doubleToChar(gps.course.deg(), 4, 2));
    addLine(buffer);
    sprintf(buffer, "Satellites: %d", gps.satellites.value());
    addLine(buffer);
}
void addLine(char *val)
{
    if (line >= 8)
    {
        resetLine();
    }
    char result[16];
    sprintf(result, "%s", val);
    u8x8.drawString(0, line++, result);
}
```
It just ouput line after line by auto increment y position.
In u8g2 I can't understand how to manage with firstPage() and nextPage() in order to output few lines. It is always output only last line in the first line. In buffer mode it also output only last line. 

Buffer mode:
```
void printGPSInfo()
{
    char buffer[16];
    resetLine();
    clearBuffer();
    setFont();
    sprintf(buffer, "%02d:%02d:%02d", gps.time.hour(), gps.time.minute(), gps.time.second());
    addLine(buffer);
    sprintf(buffer, "Lat: %s", doubleToChar(gps.location.lat(), 4, 5));
    addLine(buffer);
    sprintf(buffer, "Lon: %s", doubleToChar(gps.location.lng(), 4, 5));
    addLine(buffer);
    sprintf(buffer, "Speed: %s", doubleToChar(gps.speed.kmph(), 4, 2));
    addLine(buffer);
    sprintf(buffer, "Course: %s", doubleToChar(gps.course.deg(), 4, 2));
    addLine(buffer);
    sprintf(buffer, "Satellites: %d", gps.satellites.value());
    addLine(buffer);
    sendBuffer();
}
void clearBuffer()
{
    u8g2.clearBuffer();
}
void sendBuffer()
{
    u8g2.sendBuffer();
}

void setFont()
{
    u8g2.setFont(u8g_font_courB08);
}
int font_size = 8;
void addLine(char *val)
{

    char result[16];
    sprintf(result, "%s", val);
    line += font_size;
    u8g2.drawStr(0, line, result);
}
```

In page mode I can't understand how to wrap `addLine` function to `do{} while(nextPage())` in order to save current logic. Should it wrap every `addLine` or all `addLine` in block? Both of variants for me doesnt work. May be it relates to autoincrement line in `addLine` but I just can't understand conception of `firstPage` and `nextPage`.

olikraus, 2017-08-27:
In full buffer mode, your code above should work, but ensure, that the constructor includes `_F_`. Maybe you can post a more complete code example to validate this.

In page mode, you need to remove the clearBuffer/sendBuffer calls from your printGPSInfo() procedure, then use:
```
firstPage()
do
   printGPSInfo()
while nextPage()
```

abonec, 2017-08-28:
Ok, I got it and it is worked somehow, but I don't understand a conception of pages.
Am I right that inside of `do while` block of code will performed many times until nextPage return false? And these iterations will be just void overhead and I should make all precalculations before `do ... while` and in this loop I should only call u8g2 api in order to make this code faster?

What the pros and cons of pages and buffers besides of memory usage? Am I right that if I have enough ram to work with full buffer I should choose this approach in order to make calculations of GUI more faster? Is there trade off between ram consumption and cpu usage? Maybe there is some better choice for update all the screen or just little part of it?

I'm just a newbie in MCU programming and graphics rendering at all.

olikraus, 2017-08-28:
I recently answered the same question here:
https://forum.arduino.cc/index.php?topic=489156.msg3391673
See msg #10 that thread.

abonec, 2017-08-28:
It is a good explanation, thank you. Am I right that in buffer mode we execute render code only once in order to calculate all the pixels? In page mode, we call the same draw as many times as buffer parts of the full buffer we have and after each draw call, we send one page to screen.

Are there some optimizations to send only changed part of the full buffer to the screen (in buffer mode) because we have a full buffer in memory? Or this is useless in terms of speed?

olikraus, 2017-08-28:
> It is a good explanation, thank you. Am I right that in buffer mode we execute render code only once in order to calculate all the pixels? In page mode, we call the same draw as many times as buffer parts of the full buffer we have and after each draw call, we send one page to screen.

Yes

> Are there some optimizations to send only changed part of the full buffer to the screen (in buffer mode) because we have a full buffer in memory? Or this is useless in terms of speed?

Thinkable, but not implemented.

abonec, 2017-09-02:
What should happens when lack of ram in full buffer mode? I migrated my project to full buffer mode and OLED is blank:
https://github.com/abonec/gps_speedometer/pull/1

I have Arduino pro mini with 328p and 2kb ram. Corresponding to your calculation for 128x64 should be 1kb of memory. Maybe other components of the project (such as tinygps) consume more than 1kb.

olikraus, 2017-09-02:
RAM consumption is shown in the log window of the Arduino IDE. If this value exceeds 80% of your controller, it will not work any more.

olikraus, 2017-09-30:
Actually there was a bug in u8g2 with the full buffer mode. The bug is fixed with 2.18.13 (issue #370).
Hope this will help.

closed 2017-09-30 by olikraus