olikraus/u8g2 #343: Support for Embedded Linux

chandrarockzz, 2017-08-29:
Hi,
I want to use this library for Embedded linux. Can you point out how to use it.

olikraus, 2017-08-29:
It needs to be ported first. There is a porting guide referenced on the wiki home page.

chandrarockzz, 2017-08-29:
can you point out the functions for writing to SPI.

olikraus, 2017-08-29:
Regarding Linux I don't know, regarding U8g2 please check  the porting guide.

chandrarockzz, 2017-08-29:
I have checked the porting guide and gone through the code and I have a clear picture now hoe to do it. Once I complete porting on Embedded Linux I will try to share the results and performace.

olikraus, 2017-08-29:
Sounds good, thanks

chandrarockzz, 2017-08-31:
I started porting your code. Wrote all the required routines and compiled it successfully. 
How do you manage the shifting of pixel. Like say if I want to print "Hai" on LCD, for each and every pixel one need to send the data then change to command mode then set to the next pixel, change to data mode send the pixel data and go on...
But what I observed is you are sending the pixel command only once that too first time. I am using IST3020 using HW SPI.
Please correct me if I am wrong.

olikraus, 2017-08-31:
U8g2 renders everything into an intermediate buffer an then sends the complete buffer.

olikraus, 2017-09-23:
no further feedback, closing...

closed 2017-09-23 by olikraus

huming2207, 2018-01-26:
Hi @olikraus , I suggest that you can have a try on the linux `spidev` interface as it should works with most of the 4-wire SPI screens.

olikraus, 2018-01-26:
Sure, i could start analysing how to use u8g2 with embedded linux. But along with SPI u8g2 also need GPIO access (DC line, CD line, reset line...). I simply do not have time to port u8g2 to other platforms.
From that perspective the Arduino Library serves as a nice hardware abstraction layer. So u8g2 will work with any "Arduino-like" hardware abstraction layer out of the box. If this is not there, then you can still port u8g2 to a random target system by following the porting guide (see the wiki).

huming2207, 2018-01-26:
Ah yea sorry, I forgot this issue. So we might need to write a wrapper for those Arduino functions. We can control those GPIO via `sysfs` interface, see here: https://www.kernel.org/doc/Documentation/gpio/sysfs.txt