olikraus/u8g2 #344: WaveShare 1.54" eInk

oceanapplications, 2017-08-30:
Using the U8G2_SSD1607_200X200_1_4W_SW_SPI constructor, it briefly draws the hello world text but then immediately refreshes.
```
void setup(void) {
  u8g2.begin();
   u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.firstPage();
  do {
    u8g2.setCursor(0, 20);
    u8g2.print(F("Hello World!"));
  } while ( u8g2.nextPage() );
}
```

Thanks

olikraus, 2017-08-30:
Maybe something in the main loop causes the flashing

oceanapplications, 2017-08-30:
I currently have nothing in the main loop but the same behavior is displayed.  It draws beautifully then very shortly after flashes the screen black then back to white leaving very faded text or nothing at all.

The u8x8 version has the same behavior.

Perhaps the WaveShare version of this display isn't identical to other versions of SSD1607?

olikraus, 2017-08-30:
Possible, I have not tested this waveshare display

jjjrmy, 2017-09-04:
Having the same issue on the 2.9inch e-Paper display from Waveshare

olikraus, 2017-09-09:
i did some optimization to the SSD1607 (#352). Maybe it works better...
But indeed, my e-paper is the embedded adventures e-paper not the waveshare e-paper with the same size.

olikraus, 2017-09-23:
as far as I remember, there once was a comment that the existing 200x200 u8g2 display works with the waveshare 1.54 display... not sure how to continue here... my own display works...

closed 2017-09-23 by olikraus