olikraus/u8g2 #345: 256x64 OLED based on SSD1322 + STM32F4 (4-wire SPI)

e-music, 2017-09-01:
The full frame buffer is a great improvement over the picture loop of the u8glib. However, I'm still facing certain challenges that need to be figured out somehow. 

1- How to write formatted text on the LCD. When I say formatted, I mean handling new lines, space between lines, navigation through text, etc.. I was wondering if this is what the print() function does in the Arduino implementation. 

2- In the old g8lib, I remember that I had a function to print integral/decimal values. In U8G2, I had to use sprintf() to convert the integer to a string then pass it to the DraStr() function. 

3- Finally, is there any way to support the hardware scrolling functions from within the U8G2? I have no problem scrolling text inside a picture_loop() in the former u8glib and the result was quite satisfying. Unfortunately, this was not the case for U8G2. The scrolling interval is kept very slow no matter how much is the increase of x_pos in each iteration, and even if there's no delay in between. 

Thank you again and I hope someone has an answer to my questions! 

Regards,

Zaher

olikraus, 2017-09-01:
1-print is supported, but no newline
2-itoa functions are still there, but not documented. Have a look into u8x8.h
3- speed should be the same, maybe you have accidentally selected the bit banging interface.

e-music, 2017-09-04:
Hi,

Thanks for your quick reply. As far as I know, I'm using the HW SPI. What follows is my byte function:

`void LCD_Init_U8G2()
{
	u8g2_Setup_ssd1322_nhd_256x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_gpio_and_delay_stm32);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);	
  u8g2_ClearBuffer(&u8g2);
	u8g2_SetContrast(&u8g2, 255);
}`

And here is the delay_gpio function:
`/* STM32F4 specific implementation, initializes SPI peripheral and 
	 provides low level functions which allows the u8g2 library to communicate with the display */
uint8_t u8x8_gpio_and_delay_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint16_t n;
	switch(msg)
		{
		//Initialize SPI peripheral 
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
			//initializeIO();
		break;
		
		//Function which implements a delay, arg_int contains the amount of ms
		case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);
		
		break;
		//Function which delays 10us
		case U8X8_MSG_DELAY_10MICRO:
			DWT_Delay_us(10);
		break;
		//Function which delays 100ns
		case U8X8_MSG_DELAY_100NANO:
			DWT_Delay_us(0.1);
		break;
		//Function to define the logic level of the clockline
		case U8X8_MSG_GPIO_SPI_CLOCK:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, GPIO_PIN_5, SET);
			else HAL_GPIO_WritePin(GPIOA, GPIO_PIN_5, RESET);
		break;
		//Function to define the logic level of the data line to the display
		case U8X8_MSG_GPIO_SPI_DATA:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, GPIO_PIN_7, SET);
			else HAL_GPIO_WritePin(GPIOA, GPIO_PIN_7, RESET);
		break; 
		// Function to define the logic level of the CS line  
		case U8X8_MSG_GPIO_CS:
		if (arg_int) HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, SET);
			else HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, RESET);
		break;
		//Function to define the logic level of the Data/ Command line
		case U8X8_MSG_GPIO_DC:
			if (arg_int) HAL_GPIO_WritePin(LCD_RS_GPIO_Port, LCD_RS_Pin, SET);
			else HAL_GPIO_WritePin(LCD_RS_GPIO_Port, LCD_RS_Pin, RESET);
		break;
		//Function to define the logic level of the RESET line
		case U8X8_MSG_GPIO_RESET:
			if (arg_int) HAL_GPIO_WritePin(LCD_RESET_GPIO_Port, LCD_RESET_Pin, SET);
			else HAL_GPIO_WritePin(LCD_RESET_GPIO_Port, LCD_RESET_Pin, RESET);		
		break; 
		
		default:
		return 0; //A message was received which is not implemented, return 0 to indicate an error
	}
	
	return 1; // command processed successfully.

}`

Let me know if I have missed something there. 

Thanks again,

Zaher

e-music, 2017-09-04:
Sorry, this is the modified byte_4wire_sw_spi():

`uint8_t u8x8_byte_4wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t i, b;
  uint8_t *data;
  uint8_t takeover_edge = u8x8_GetSPIClockPhase(u8x8);
  uint8_t not_takeover_edge = 1 - takeover_edge;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;
      while( arg_int > 0 )
      {
				b = *data;
				HAL_SPI_Transmit(&hspi1, &b, 1, 100);
				data++;
				arg_int--;
		for( i = 0; i < 8; i++ )
				{
					if ( b & 128 )
					u8x8_gpio_SetSPIData(u8x8, 1);
					else
					u8x8_gpio_SetSPIData(u8x8, 0);
					b <<= 1;
	  
					u8x8_gpio_SetSPIClock(u8x8, not_takeover_edge);
					u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->sda_setup_time_ns);
					u8x8_gpio_SetSPIClock(u8x8, takeover_edge);
					u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->sck_pulse_width_ns);
				}    
      }
      break;
      
    case U8X8_MSG_BYTE_INIT:
      /* disable chipselect */
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      /* no wait required here */
      
      /* for SPI: setup correct level of the clock signal */
      u8x8_gpio_SetSPIClock(u8x8, u8x8_GetSPIClockPhase(u8x8));
      break;
    case U8X8_MSG_BYTE_SET_DC:
      u8x8_gpio_SetDC(u8x8, arg_int);
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);  
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
    default:
      return 0;
  }
  return 1;
}
`

olikraus, 2017-09-04:
Still it looks like slow SW emulated SPI bitbanging.

e-music, 2017-09-05:
Yes, is there any way to make it faster? How about the parallel 8080 interface? Is it any faster?

olikraus, 2017-09-05:
I am sure the STM32 has a hardware SPI interface, which should be much faster.

e-music, 2017-09-05:
I'm not sure if we understand each other here. Do you mean my byte/communication function did not utilize the hardware SPI of the STM32F4?

olikraus, 2017-09-06:
Yes, correct: the bit loop is still there in your code from above.

e-music, 2017-09-06:
Do you mind elaborating more on that? And by the way, what about scrolling? Is there any way to do it with U8G2?

e-music, 2017-09-06:
I would really appreciate your help if you can show me exactly where I need to modify the code for the HW SPI?

olikraus, 2017-09-06:
> Do you mind elaborating more on that? And by the way, what about scrolling? Is there any way to do it with U8G2?

Sure it is possible

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ScrollingText/ScrollingText.ino

> I would really appreciate your help if you can show me exactly where I need to modify the code for the HW SPI?

Unfortunately I have no idea how to use HW SPI with the STM32, but to me it looks like you need to remove the loop and its body:

for( i = 0; i < 8; i++ )
{
   ...

e-music, 2017-09-06:
Thank you for the prompt follow up. No problem if you do not know how to do it on STM32. Any example for any uC is highly appreciated. So, you suggest removing that loop completely from the code or substitute it with something?

e-music, 2017-09-06:
Looks like you were 100% correct. I have just commented the loop inside the byte file and the code runs much faster than ever. I'm going to try to scrolling and let you know. Thank you again!

olikraus, 2017-09-06:
No problem, example for HW SPI is here (Arduino HAL interface):

https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L349

e-music, 2017-09-06:
Just tried it with my own scrolling implementation, updating x_pos, and everything looks perfect now! Scrolling runs smoothly without any issue! THank you again for your help!

e-music, 2017-09-06:
You may close the issue now!

olikraus, 2017-09-06:
ok..

closed 2017-09-06 by olikraus