olikraus/u8g2 #350: Use of the u8g2.begin(); statement

Tulsa2008, 2017-09-07:
I have an application where I am using two GLCD displays driven by an Arduino Mega 2560 processor.  The application performs a large number of tasks and for the most part works very well.  The application I have developed can display ten (10) different screens on each of the two displays.

The constructors I am using are as follows:
"U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2 (U8G2_R0, 3, 2, 4);" & "U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2A (U8G2_R0, 3, 2, 4);"

The problem I am having involves only one of the two displays.  The problematic display will randomly (maybe once or twice a day) get lost and start displaying the data sent to it in a corrupted manner.    Once the display becomes corrupted, changing to a different screen will not correct the problem.  To date, the only remedy I have is to reboot the entire system and this is not acceptable.     

As a possible patch, I have written a small procedure that will call and run "u8g2.begin();" and "  u8g2A.begin();".  This will force a re-initialization of the display software.  

The code presented below is a copy code I have developed:
<
void ScreenReset(){
        boolean SwitchValue = HIGH; 

  if (digitalRead(9) == LOW) {  //digital input (switch)
     SwitchValue = LOW;
     delay(100);
     if (digitalRead(9) == SwitchValue) {
        u8g2.begin(); 
        u8g2A.begin(); 
        delay (100);
       }
    // Screen initial (Welcome) message
    u8g2.clearBuffer(); u8g2.sendBuffer(); u8g2.setFont(u8g2_font_pxplusibmvga9_tr);
    u8g2.drawStr(8,28,strcpy_P(ScreenLabel, (char*)pgm_read_word(&(TextHdrTbl[0]))));
    u8g2.drawStr(41,40,strcpy_P(ScreenLabel, (char*)pgm_read_word(&(TextHdrTbl[1]))));
    u8g2.sendBuffer();
    delay(1000);
    Display(); // This will rewrite the currently selected screens to the two displays 
   }
}
> 

My concern is that doing this may cause memory problems.  

What are your thoughts on doing this??

olikraus, 2017-09-07:
You really have both displays at the same GPIO ports?
If so, then 
u8g2.begin();
u8g2A.begin();
is useless.
In fact in such a case you need only one constructor.

Tulsa2008, 2017-09-08:
Sorry about the confusion.  The correct constructors are as follows:
  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2(U8G2_R0, 47, 48, 49);  
  U8G2_ST7565_LM6059_F_4W_HW_SPI u8g2A (U8G2_R0, 46, 48,49);

When I wrote the first post I cut an pasted the same constructor and did not change the GPIO pins, My bad.

Ken

olikraus, 2017-09-08:
ok, understand. 
But yes, it may always happen, that an OLED "forgets" internal values. Calling "begin" is a typical practise to refresh the display content. 

To avoid wrong data also ensure to have short wires between board and OLED.

Tulsa2008, 2017-09-08:
Thanks for your input.  
As for the wire length, understood.  I have the two displays mounted end to end with the level shifter board mounted above the displays. The cables (wires) to each display are around 4 cm long and are setup such that every other wire is grounded.    The grounded wires are grounded only on the level shifter board to eliminate ground loop problems.

Again, Thanks for your input
Ken

olikraus, 2017-09-09:
Well, this sounds shourt enough.... 
I do not know what else i can do from my side. I personally think it is a hardware problem.

Tulsa2008, 2017-09-09:
I suspect you are correct (Hardware problem).
I was just wondering if by rerunning "u8g2.begin();" if I was was risking memory problems and you have answered that.

You have answered my question.  I think we can close this issue.

Thanks for your input and a GREAT lib.

Ken

olikraus, 2017-09-09:
:)

closed 2017-09-09 by olikraus