olikraus/u8g2 #354: Global variables used extensively more with U8G2 than U8G lib.

slaiuno, 2017-09-11:
Ported my project from u8g to u8g2, did no other change.
Compiling with u8g2 produces warning about global variables using 93% of dynamic memory.
With old library the same reading was 39%.
Any obvious reason for this?

olikraus, 2017-09-11:
Which constructor?

slaiuno, 2017-09-11:
Used:
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
But I'm still trying to figure out if that's the correct one to replace:
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE|U8G_I2C_OPT_DEV_0);	// I2C / TWI 
At least it works.

olikraus, 2017-09-11:
There are three constructors:
U8G2_SSD1306_128X64_NONAME_1_SW_I2C
U8G2_SSD1306_128X64_NONAME_2_SW_I2C
U8G2_SSD1306_128X64_NONAME_F_SW_I2C

They do occupy different amount of memory:
U8G2_SSD1306_128X64_NONAME_1_SW_I2C  128 Bytes
U8G2_SSD1306_128X64_NONAME_2_SW_I2C  256 Bytes
U8G2_SSD1306_128X64_NONAME_F_SW_I2C 1024 Bytes

The exact replacement for U8GLIB_SSD1306_128X64 is U8G2_SSD1306_128X64_NONAME_1_SW_I2C

This information is also provided here:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_noname

If you use the `_1_` constructor, your program should consume the same amount of dynamic memory.

slaiuno, 2017-09-11:
Thanks! 
That helped, now back at 42% with the dyn memory.

olikraus, 2017-09-11:
:-)

closed 2017-09-11 by olikraus