olikraus/u8g2 #356: Use U8g2 with STM32

sboy888, 2017-09-12:
Hi,
U8g2 is the great library for LCD and i want to use it with my stm32f4 board. I was use LCD with ST7920 chip. I done the setup for my stm with #179 and u8g2setupc. But the LCD keeping show some strange characters. 
[image]
I'm new to stm and u8g2. So many thanks for helps.
Here my config

PA1 - CLK - E
PA2 - SPI data - R/W
PA3 - RST - RST
PA4 - CS - PSB
PA5 - CD - RS

>u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_stm32);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);

>void draw(u8g2_t *u8g2)
{
    u8g2_SetFontMode(u8g2, 1);
    u8g2_SetFontDirection(u8g2, 0);
    u8g2_SetFont(u8g2, u8g2_font_6x12_tr);
    u8g2_DrawStr(u8g2, 0, 20, "Hello World");
}

>u8g2_FirstPage(&u8g2);
    do
    {
      draw(&u8g2);
    } while( u8g2_NextPage(&u8g2) );

>uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg,U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
	switch(msg){
		//Initialize SPI peripheral
		case U8X8_MSG_GPIO_AND_DELAY_INIT:			
		break;
		//Function which implements a delay, arg_int contains the amount of ms
		case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);
		break;
		//Function to define the logic level of the clockline
		case U8X8_MSG_GPIO_SPI_CLOCK:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, GPIO_PIN_1, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOA, GPIO_PIN_1, GPIO_PIN_RESET);
		break;
		//Function to define the logic level of the data line to the display
		case U8X8_MSG_GPIO_SPI_DATA:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, GPIO_PIN_2, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOA, GPIO_PIN_2, GPIO_PIN_RESET);
		break;
		// Function to define the logic level of the CS line
		case U8X8_MSG_GPIO_CS:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, CS_LCD_Pin, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOA, CS_LCD_Pin, GPIO_PIN_RESET);
		break;
		//Function to define the logic level of the Data/ Command line
		case U8X8_MSG_GPIO_DC:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, CD_LCD_Pin, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOA, CD_LCD_Pin, GPIO_PIN_RESET);
		break;
		//Function to define the logic level of the RESET line
		case U8X8_MSG_GPIO_RESET:
			if (arg_int) HAL_GPIO_WritePin(GPIOA, RST_LCD_Pin, GPIO_PIN_SET);
			else HAL_GPIO_WritePin(GPIOA, RST_LCD_Pin, GPIO_PIN_RESET);
		break;
		default:
			return 0; //A message was received which is not implemented, return 0 to indicate an error
	}
	return 1; // command processed successfully.
}

olikraus, 2017-09-12:
I think you have to implement the other delay functions also:

https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

sboy888, 2017-09-12:
Many thanks to you **olikraus**
I just already checked the wires and added some delay functions (us, ns) . 
Follows: 
[https://github.com/olikraus/u8g2/issues/179](url)
[https://github.com/olikraus/u8g2/wiki/setup_tutorial](url)
[https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform](url).
So my LCD works perfectly. 
[image]
I think people who new to stm32 and u8g2 like me can follow three reference materials above. So you can close this issue. Thankyou! ^^

olikraus, 2017-09-12:
:)

closed 2017-09-12 by olikraus

qiuwe, 2017-10-10:
I use OLED(SSD1306 12864), can you send your whole project to me for reference? thank you！

rokk0, 2017-11-05:
Is there any detailed tutorial how to use ST7920 and STM32 with u8g2? I was trying to use these "issue-guides", but when I call: u8g2_Setup_st7920_s_128x64_1(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_stm32); - I'm out of RAM. Old version (u8glib) works fine, but I need UTF-8 fonts from new version.

Without calling setup function:
Program Size: Code=9844 RO-data=228 RW-data=28 ZI-data=2020

With it:
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching u8g2_d_memory.o(.bss).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching startup_stm32f030x8.o(STACK).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching tim.o(.bss).
projectl.axf: Error: L6406E: No space in execution regions with .ANY selector matching spi.o(.bss).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching adc.o(.bss).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching main.o(.bss).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching i2c.o(.bss).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching main.o(.data).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching u8x8_cad.o(.bss).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching system_stm32f0xx.o(.data).
project.axf: Error: L6406E: No space in execution regions with .ANY selector matching stm32f0xx_hal.o(.data).
project.axf: Error: L6407E: Sections of aggregate size **0x17ef8 bytes** could not fit into .ANY selector(s).
Not enough information to list image symbols.
Not enough information to list the image map.

olikraus, 2017-11-05:
maybe you did not activate the gcc linker garbage collector.

rokk0, 2017-11-05:
I was trying to add --gc-sections in Linker parameters, but got this error:
project.axf: error: L3900U: Unrecognized option '--gc-sections'.

olikraus, 2017-11-05:
Usually options are:
compiler: -ffunction-sections -fdata-sections
linker: -Wl,--gc-sections

rokk0, 2017-11-05:
Looks like I cannot enable garbage collection in Keil5 linker...

rokk0, 2017-11-05:
Keil uses ARM Compiler Toolchain and it has option --remove, that works the same as --gc-sections, but it changed nothing :(

12.118 --remove, --no_remove

Enables or disables the removal of unused input sections from the image.
Usage
An input section is considered used if it contains an entry point, or if it is referred to from a used section.
By default, unused section elimination is disabled when building dynamically linked libraries (DLLs) or shared objects, Use --remove to re-enable unused section elimination.
Use --no_remove when debugging to retain all input sections in the final image even if they are unused.
Use --remove with the --keep option to retain specific sections in a normal build.

rokk0, 2017-11-06:
Main problem were static uint8_t buf[x] variables in u8g2_d_memory.c. I commented all of them and created a global one in that file. Here is step by step guide, how to run ST7920 with STM32 in Keil 5 using HAL:

1. Add all files from u8g2/csrc folder to your project
2. Include u8g2.h in main.c and create global variable "u8g2_t u8g2;"
3. Add function for HAL HW SPI usage:
```
uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t byte;
  uint8_t *data;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;
      while( arg_int > 0 )
      {
        byte = *data;
        data++;
        arg_int--;
        HAL_SPI_Transmit(&hspi1, &byte, 1, 100);
      }
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_SET);
      __nop(); // 21 ns
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      __nop(); // 21 ns
      HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_RESET);
      break;
    default:
      return 0;
  }
  return 1;
}
```

4. Add gpio_and_delay function (we can leave it "empty" because all required calls to it are inside u8x8_byte_4wire_hw_sp function):
```
uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg,U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
  return 1; // command processed successfully.
}
```

5. Add draw function:
```
void draw(u8g2_t *u8g2)
{
  u8g2_ClearBuffer(u8g2);

  u8g2_SetFontMode(u8g2, 1);
  u8g2_SetFontDirection(u8g2, 0);
  u8g2_SetFont(u8g2, u8g2_font_6x10);

  sprintf(string_1, "LCD refresh: %d ms", lcd_refresh_time);
  u8g2_DrawStr(u8g2, 2, 10, string_1);

  u8g2_SendBuffer(u8g2);
}
```

6. Init u8g2 in main function:
```
u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8g2_gpio_and_delay_stm32);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
```

7. Call draw function in main loop: draw(&u8g2);

LuizRoque, 2017-11-24:
Hi rokk0, I've try to follow the steps you've described, but still i'm not be able to write nothing!, i'm using STM32F070-nucleo board,

here is my code

/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "u8g2.h"
#include "spi.h"

// LCD 128x64
u8g2_t u8g2;

uint8_t count = 0;

GPIO_InitTypeDef  GPIO_InitStruct;

uint8_t lcd_refresh_time = 1;
char string_1[30];

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void Error_Handler(void);
uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr);
uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);

/* Private functions ---------------------------------------------------------*/

/**
  * @brief  Main program.
  * @param  None
  * @retval None
  */
int main(void)
{

  HAL_Init();

  /* Configure LED2 */
  BSP_LED_Init(LED2);
  /* Configure the system clock to 48 MHz */
  SystemClock_Config();

  	  __HAL_RCC_GPIOB_CLK_ENABLE();
	/* SPI SCK, MOSI GPIO pin configuration  */
	GPIO_InitStruct.Pin = GPIO_PIN_13 | GPIO_PIN_15;
	GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
	GPIO_InitStruct.Pull = GPIO_NOPULL;
	GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
	HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

	//RST_LCD_GPIO_CLK_ENABLE();
	GPIO_InitStruct.Pin = GPIO_PIN_2;
	GPIO_InitStruct.Pull = GPIO_NOPULL;
	GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
	HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

	//CS_LCD_GPIO_CLK_ENABLE();
	GPIO_InitStruct.Pin = GPIO_PIN_1;
	GPIO_InitStruct.Pull = GPIO_NOPULL;
	GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
	HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

	GPIO_InitStruct.Pin = GPIO_PIN_11 | GPIO_PIN_12;
	GPIO_InitStruct.Pull = GPIO_NOPULL;
	GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
	HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);

	MX_SPI2_Init();

  u8g2_Setup_st7920_128x64_f(&u8g2, U8G2_R0,
[... 178 lines cut ...]
  /* No HSE Oscillator on Nucleo, Activate PLL with HSI as source */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_NONE;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PREDIV = RCC_PREDIV_DIV2;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL12;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct)!= HAL_OK)
  {
    /* Initialization Error */
    while(1); 
  }

  /* Select PLL as system clock source and configure the HCLK, PCLK1 clocks dividers */
  RCC_ClkInitStruct.ClockType = (RCC_CLOCKTYPE_SYSCLK | RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_PCLK1);
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1)!= HAL_OK)
  {
    /* Initialization Error */
    while(1); 
  }
}

Can you give a little help?

rokk0, 2017-11-24:
I think you need to call u8g2_Setup_st7920**_s_**128x64_f, not u8g2_Setup_st7920_128x64_f.

breezer15, 2017-12-05:
Hello,

i want to use u8g2 with a STM32L432. I used the instruction from rokk0 to implement the code, but the Display shows nothing.

Here are the important function calls:

uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t byte;
  uint8_t *data;

  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;
      while( arg_int > 0 )
      {
        byte = *data;
        data++;
        arg_int--;
        HAL_SPI_Transmit(&hspi1, &byte, 1, 100);
      }
      break;
    case U8X8_MSG_BYTE_SET_DC:
    	if(arg_int == 0){HAL_GPIO_WritePin(
    			DC_GPIO_Port, DC_Pin, GPIO_PIN_RESET);
    	}
    	if(arg_int == 1){HAL_GPIO_WritePin(
    			DC_GPIO_Port, DC_Pin, GPIO_PIN_SET);
    	}
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_SET);
      //__nop(); // 21 ns+
      asm("NOP");
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      //__nop(); // 21 ns
      asm("NOP");
      HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_RESET);
      break;
    default:
      return 0;
  }
  return 1;
}

uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg,U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
  return 1; // command processed successfully.
}

void draw(u8g2_t *u8g2)
{
  u8g2_ClearBuffer(u8g2);

  u8g2_SetFontMode(u8g2, 1);
  u8g2_SetFontDirection(u8g2, 0);
  u8g2_SetFont(u8g2, u8x8_font_chroma48medium8_r);

  sprintf(string_1, "LCD refresh: %d ms", lcd_refresh_time);
  u8g2_DrawStr(u8g2, 2, 10, string_1);

  u8g2_SendBuffer(u8g2);
}

LuizRoque, 2017-12-07:
Hi, breezer15 

All your code seen ok to me, my display has the ks0108 controller, but using his setup didn't work. It just worked using the 

u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0,
		  u8x8_byte_4wire_hw_spi, u8g2_gpio_and_delay_stm32);

Now is all working!

thihakyawjob, 2018-11-24:
> Main problem were static uint8_t buf[x] variables in u8g2_d_memory.c. I commented out all of them and created a global one in that file. Here is step by step guide how to run ST7920 with STM32 in Keil 5 using HAL:
> 
> 1. Add all files from u8g2/csrc folder to your project
> 2. Include u8g2.h in main.c and create global variable "u8g2_t u8g2;"
> 3. Add function for HAL HW SPI usage:
> 
> ```
> uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
> {
>   uint8_t byte;
>   uint8_t *data;
>  
>   switch(msg)
>   {
>     case U8X8_MSG_BYTE_SEND:
>       data = (uint8_t *)arg_ptr;
>       while( arg_int > 0 )
>       {
>         byte = *data;
>         data++;
>         arg_int--;
>         HAL_SPI_Transmit(&hspi1, &byte, 1, 100);
>       }
>       break;
>     case U8X8_MSG_BYTE_START_TRANSFER:
>       HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_SET);
>       __nop(); // 21 ns
>       break;
>     case U8X8_MSG_BYTE_END_TRANSFER:
>       __nop(); // 21 ns
>       HAL_GPIO_WritePin(SPI1_CS_GPIO_Port, SPI1_CS_Pin, GPIO_PIN_RESET);
>       break;
>     default:
>       return 0;
>   }
>   return 1;
> }
> ```
> 1. Add gpio_and_delay function (we can leave it "empty" because all required calls to it are from u8x8_byte_4wire_**sw**_spi function, which one we don't use):
> 
> ```
> uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg,U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
> {
>   return 1; // command processed successfully.
> }
> ```
> 1. Add draw function:
> 
> ```
> void draw(u8g2_t *u8g2)
> {
>   u8g2_ClearBuffer(u8g2);
> 
>   u8g2_SetFontMode(u8g2, 1);
>   u8g2_SetFontDirection(u8g2, 0);
>   u8g2_SetFont(u8g2, u8g2_font_6x10);
> 
>   sprintf(string_1, "LCD refresh: %d ms", lcd_refresh_time);
>   u8g2_DrawStr(u8g2, 2, 10, string_1);
> 
>   u8g2_SendBuffer(u8g2);
> }
> ```
> 1. Init u8g2 in main function:
> 
> ```
> u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_hw_spi, u8g2_gpio_and_delay_stm32);
> u8g2_InitDisplay(&u8g2);
> u8g2_SetPowerSave(&u8g2, 0);
> ```
> 1. Call draw function in main loop: draw(&u8g2);

thihakyawjob, 2018-11-24:
> Main problem were static uint8_t buf[x] variables in u8g2_d_memory.c. I commented out all of them and created a global one in that file. Here is step by step guide how to run ST7920 with STM32 in [...]

Hi, rokko,

I follow your instruction using Nucleo-F746ZG.  My setup is as follow.
PA5 - SCK to ST7920 Pin - E
PA7 - MOSI to ST7920 - RW
PD14 - CS

I would like to know the following information from you.
1) What is the SPI configuration? SPI Clock, CPOL and CPHA setting? SPI_Datasize? 8Bit or 16Bits.
2) I attached my file for your review.

Please help me check. I don't know what wrong with it.

Thanks.

Regards,
Thiha 
[main.txt](https://github.com/olikraus/u8g2/files/2612417/main.txt)

olikraus, 2018-11-24:
Probably I can not support here as library author. I do not what HAL_SPI_Transmit does.

thihakyawjob, 2018-11-25:
I think the issue is due to Signal Level. STM32F logic level is 3.3 V and
ST7920 requires 5V x 0.7=3.7V. Let me verify it with level shifter.

Thanks

thihakyawjob, 2018-12-10:
Dear Olikraus,

ST7920 LCD doesn't show properly. I don't know what is wrong. Please give me some suggestion.

[image]

/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "stm32f7xx_hal.h"

/* USER CODE BEGIN Includes */
#include "u8g2.h"
/* USER CODE END Includes */

/* Private variables ---------------------------------------------------------*/

SPI_HandleTypeDef hspi1;

/* USER CODE BEGIN PV */
/* Private variables ---------------------------------------------------------*/
u8g2_t u8g2;
/* USER CODE END PV */

/* Private function prototypes -----------------------------------------------*/
void SystemClock_Config(void);
static void MX_GPIO_Init(void);
static void MX_SPI1_Init(void);

/* USER CODE BEGIN PFP */
/* Private function prototypes -----------------------------------------------*/
uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg,U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr);
uint8_t u8x8_byte_4wire_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
void draw(u8g2_t *u8g2);
/* USER CODE END PFP */

/* USER CODE BEGIN 0 */

/* USER CODE END 0 */

/**
  * @brief  The application entry point.
  *
  * @retval None
  */
int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MCU Configuration----------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_SPI1_Init();
  /* USER CODE BEGIN 2 */
[... 237 lines cut ...]
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t* file, uint32_t line)
{ 
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     tex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

/**
  * @}
  */

/**
  * @}
  */

/************************ (C) COPYRIGHT STMicroelectronics *****END OF FILE****/

olikraus, 2018-12-10:
I do not know the STM32 specific API nor did I wrote the code (additionally you didn't use proper code tags here). Moreover I do not have time to debug this. If you did not wrote the glue code between u8g2 and STM32, then maybe try to find the person who wrote this code.

I can only check the hardware with you (wiring, signal levels) with you. Did you add proper signal level converters here?

iamsaber, 2019-07-01:
> Main problem were static uint8_t buf[x] variables in u8g2_d_memory.c. I commented out all of them and created a global one in that file. 

Hi rokko,
Can you explain this part a little more please? how and in which file you create a global variable instead ? can you put  here this part of your code ? and also tell us if you change "u8g2_Setup_st7920_s_128x64_f" function declaration in the library also or not?

Thanks alot